MarlinFirmware/Marlin #23275: [BUG] Bed heating pid is broken

ErsaFap, 2021-12-07:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Bed heating is off by 10 degress causing thermal protection cutoff.
Bang mode works correctly.

### Bug Timeline

Old

### Expected behavior

Bed heats to 100C

### Actual behavior

Bed heats to 90C

### Steps to Reproduce

1 Open repetier host
2 Turn on bed heating to any temperature 
3 The bed heats up to  (tset - 10deg) and stabilizes 
4 Thermal fault after 20 sec

### Version of Marlin Firmware

2.0.x, 2.0.x bugfix, 2.0.9

### Printer model

Flyingbear P902

### Electronics

Bigtreetech SKR 1.3

### Add-ons

RepRap discount smart controller

### Host Software

Repetier Host

### Additional information & file uploads

[image]

ellensp, 2021-12-07:
Do you have #define BED_MAXTEMP  90

ErsaFap, 2021-12-07:
> Do you have #define BED_MAXTEMP 90

I'm sad to read this. I am not imbecile. 
[config.zip](https://github.com/MarlinFirmware/Marlin/files/7671229/config.zip)

Here is my config, I hope it will be useful

thisiskeithb, 2021-12-07:
> I'm sad to read this. I am not imbecile.

No one said that.

You didn't include your configs in the original report like the issue template requested, so we can only guess as to what the issue could be.

NezSez, 2021-12-07:
> I'm sad to read this. I am not imbecile.

He didn't mean to imply that, it's just normal good tshooting practice to rule out obvious/simple fixes first especially when going on minimal information.

ErsaFap, 2021-12-07:
> > I'm sad to read this. I am not imbecile.
> 
> He didn't mean to imply that, it's just normal good tshooting practice to rule out obvious/simple fixes first especially when going on minimal information.

its ok. Don't worry about it .
I found this problem has been around for a long time. Possible since version 2.0.6, Im not sure.

thisiskeithb, 2021-12-07:
Duplicate of #21799

closed 2021-12-07 by thisiskeithb

ErsaFap, 2021-12-07:
Its not duplicate (I think), the regulator is doing great but target temperature is incorrect. 
It heats up to 90 degrees in 6 minutes. Power is enough. But it should heat up to 100.