MarlinFirmware/Marlin #23277: Acknowledgement of Logo creator

ahmetcemturan, 2021-12-08:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Hi, I would like to ask You to put an acknowledgement on the creator of the logo and the name (me). I noticed that Alessandro from slic3r did that and think that it would be nice to get some recognition. You can confirm that with Erik.  
I could also give You insights why the Name and the shapes of the Logo(s)

Best Regards.

### Bug Timeline

old

### Version of Marlin Firmware

ALL

Roxy-3D, 2021-12-11:
How many hours of work did it take to create the original logo?
How many iterations of the logo were there?

sjasonsmith, 2021-12-11:
FYI, I found acknowledgement in an old commit title:

[Added marlin logos. (Designed by Ahmet Cem TURAN)](https://github.com/MarlinFirmware/Marlin/commit/8f9067faa698d2bc59d7b21fa387ac536eeeeca6) (8f9067faa698d2bc59d7b21fa387ac536eeeeca6)

Roxy-3D, 2021-12-11:
@ahmetcemturan   It would seem to me the acknowledgement has always been there.    Is that sufficient for your purposes?

closed 2021-12-11 by Roxy-3D

ahmetcemturan, 2021-12-14:
Oh thanks.

ahmetcemturan, 2022-01-06:
Sorry to reply this late.  

A few hours together with the small variant.

The backgound of the design process went along with the Name itself actually.

I did a lot of testing for Erik at that timeand he was theonly one working on true acceleration and all the features that Marlin had brought over Sprinter (not to mention tonokip fw etc..).

 

So the firmware needed a Name.  It had to reflect the character of the firmware.  What do we have:

-The creator is Dutch

-It is fast

-It gets around curves easily

-It works a miracle in sharp corners like infill (no more skipping steps)

 

So the Name neede to sound dutch.  ....Marien, Marleen....  

Well there’s that superfast fish called Marlin...   İt is fast, manouvrable.....    There it is the name...

 

Now the logo....  Well a Marlin of course but just the fish?  We are not a fishermans supply brand..

The frame itself has a sharp and a round corner as perimeter (as Marlin does both great) and also marlin does infill/solid infill great...

There we go for the background.  

 

So that’s the story...

 

And the Pronterface logo for Kliment... 

Well we were on IRC the whole time.  No emojis at that time...  

And there was face in the name.   So that’s it...  

 

I love and still use Marlin btw.

It gets better every day but I think the config and adv.config files need to be split in more parts.  The adv has become sooo long...

 

Keep up the great work and always happy to help with graphics.

 

 

Best Regards,

Ahmet Cem TURAN

www.miafilament.com <http://www.miafilament.com>

thinkyhead, 2022-01-07:
How's it going, Ahmet? To keep your author credit as closely tied to the SVG file(s) as possible it would make sense to insert an author tag and a description that indicates its license. Are you versed enough in the GitHub Way to take care of that and submit a PR for us?

ahmetcemturan, 2022-01-07:
Thanks, it’s going well.   

I think I can do the pull request.

Did You like the Marlin story?

 

Best Regards,

Ahmet TURAN

ahmetcemturan, 2022-01-07:
Hi Scott,

Thanks it’s going fine,  

I think the files dont exist in the main repo anymore.  

So I should update the ones in the https://github.com/MarlinFirmware/MarlinDocumentation

Correct me if I am wrong pls.

 

Best Regards,

Ahmet TURAN

thinkyhead, 2022-04-21:
@ahmetcemturan — Marlin still contains `marlin.svg` and `marlin-old.svg` which are possibly derived from your original logo files. Please examine those files in the folder [`buildroot/share/pixmaps/logo`](https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x/buildroot/share/pixmaps/logo) to confirm that they are the correct images.