MarlinFirmware/Marlin #23279: [BUG] Marlin/src/module/endstops.cpp:567:35: error: macro "BOTH" requires 2 arguments, but only 1 given

salfter, 2021-12-09:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

I pulled in an update for one of my printers for the first time in a while, and got the error in the title at compile time.  Looking at the same code in the 2.0.x branch, it appears that 

```#if BOTH(PROBE_ACTIVATION_SWITCH)```

should instead be

```#if BOTH(MARLIN_DEV_MODE, PROBE_ACTIVATION_SWITCH)```

### Bug Timeline

It's the first I've seen of it...last merge I built was 470+ commits ago.

### Expected behavior

Compile successfully.

### Actual behavior

Failed to compile.

### Version of Marlin Firmware

bugfix-2.0.x

### Printer model

homebuilt Hypercube 300

### Electronics

SKR Pro 1.2

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Prusa Slicer

### Additional information & file uploads

[config.zip](https://github.com/MarlinFirmware/Marlin/files/7682626/config.zip)

ellensp, 2021-12-09:
This error was just introduced 3 hours ago.. https://github.com/MarlinFirmware/Marlin/commit/4d45fdf0eb605629960d06abb86704cfbd62db49

ellensp, 2021-12-09:
It would seem the intent is to not require MARLIN_DEV_MODE
So perhaps it should be  #if ENABLED(PROBE_ACTIVATION_SWITCH)

ellensp, 2021-12-09:
merged

closed 2021-12-09 by ellensp