MarlinFirmware/Marlin #23281: Seek any specific active endstop in an axis [FR] (feature summary)

sanjaymorphle, 2021-12-09:
### Describe the feature you want

It would be very helpful to have a function to support seeking any particular endstop on an axis. There is already a similar logic- 'homing' that does it. When homing is requested, the motor seeks the homing endstop limit in one direction until it finds it. A similar function for any endstops related to lets say 'X' axis woul help the motor to reach exactly the same point everytime even if there are steps lost during motion. This would be helpful if you have optical endstops at a position on the rail where you want to stop at apart from the homing limits.

Roxy-3D, 2021-12-11:
This is an interesting feature idea.    Can you explain how this would be used to enhance your prints?

sanjaymorphle, 2021-12-13:
> This is an interesting feature idea. Can you explain how this would be used to enhance your prints?
> 
> Does it make sense to allow an arbitrary number of position sensors on a given axis? Would more than 1 extra sensor have value?

Actually i am using the marlin firmware in a pick&place robot. So i use optical encoders to compensate for any errors after reaching the destination. But these are not absolute encoders, so in case the encoders stop working or lose data, i can seek to the nearest endstop (which might not be the homing limit), and update the encoder value to the current position of the motor,  since the position of a fixed hardware endstop is likely to change and will be the most accurate reference to precisely reach a point. It simply acts like a zeroing point for my encoders which might have accumulated errors in a long run.

sanjaymorphle, 2021-12-15:
I have worked on this feature, and have a working code for seeking endstops. Tried it with one of the axis, now just need to update the code for general use case with other axes.