MarlinFirmware/Marlin #23286: Anet A8 - Probing outside of the bed

HuguesDug, 2021-12-11:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

I upgraded from Marlin 2.0.6.1 to Marlin 2.0.9.3 (bugfix branch).

I updated the parameters of the config file (downloaded as from github official repo) to reintroduce the specific values of my config (ANET A8 template) (file compare with notepad++)

In the particular case :
#define NOZZLE_TO_PROBE_OFFSET { 25, 55, -0.45 }
and
#define PROBING_MARGIN 20
and finally
#define AUTO_BED_LEVELING_BILINEAR

With these parameters in 2.0.6.1, the probe was probing in 9 points of the bed, including corners. Exactly where and how it should be.

With 2.0.9.3, sending G29 command, the probe does not take measures in the right spots (mostly on the Y axis), and finally, at the rear of Y axis, try to get the measurement outside of the bed. It ends up with a collision between the nozzle and the build plate.

After G28, If I send the G28 command, and then G0 X0 Y0, the nozzle will be exactly in the right place. So, the definitions of my geometry are correct. I also measured the exact distance nozzle/probe. I confirm the values in mm.

I wonder if it does not relate to these
#define X_MIN_POS -31
#define Y_MIN_POS -8

These offsets between "end stop" and "actual geometry" does not seems to be taken into account in the calculation of the points coordinate for probing.

### Bug Timeline

Worked perfectly in 2.0.6.1

### Expected behavior

Behave as it should be:
Probe 9 points at 20mm from the bed frame and in center of each axis.

### Actual behavior

Probe outside of the bed, points not being aligned with actual bed physical position.

### Steps to Reproduce

See description of the bug

### Version of Marlin Firmware

2.0.9.3

### Printer model

Anet A8

### Electronics

Stock with probe (from ANET)

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Slic3r

### Host Software

OctoPrint

### Additional information & file uploads

[Configuration-ANET-A8.zip](https://github.com/MarlinFirmware/Marlin/files/7696555/Configuration-ANET-A8.zip)

ellensp, 2021-12-11:
Configuration issue. Not a Marlin bug. 

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- 3D Printing Discord server. Join link: https://discord.gg/qNnSzsug
- Marlin specific Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2021-12-11 by ellensp

HuguesDug, 2021-12-11:
That was a quick closing.

I consider it as a bug (or at least a regression), as it was working perfect in previous release.

HuguesDug, 2021-12-20:
Finally installed stable branch 2.0.9.2
Exactly the same config.

It works fine,

Are you real sure of the bugfix branch ?