MarlinFirmware/Marlin #23287: [BUG] ANET A8 - ZONESTAR_LCD - Buttons non working

HuguesDug, 2021-12-11:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

I upgraded Marlin from 2.0.6.1 to latest 2.9.0.3 (bugfix branch).

After the upgrade, the buttons of the LCD screen do not produce any effect (you can press all buttons, nothing happen).

Firmware source and config files from this Repo.

#define ZONESTAR_LCD defined.

### Bug Timeline

Worked perfectly in 2.0.6.1

### Expected behavior

Menu should react when pressing buttons

### Actual behavior

Buttons are not operating

### Steps to Reproduce

Upload firmware, reboot.

Then, screen display perfectly but buttons are not doing anything.

### Version of Marlin Firmware

2.0.9.3

### Printer model

Anet A8

### Electronics

Stock with probe (from ANET)

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Slic3r

### Host Software

OctoPrint

### Additional information & file uploads

[Configuration-ANET-A8.zip](https://github.com/MarlinFirmware/Marlin/files/7696563/Configuration-ANET-A8.zip)

HuguesDug, 2021-12-20:
Rolled back to 2.0.9.2
All is fine.

Something strange with bugfix branch

ellensp, 2021-12-30:
Can you test PR https://github.com/MarlinFirmware/Marlin/pull/23388
This should fix this

HuguesDug, 2021-12-30:
The problem is that the "bugfix" branch also  brought me also a problem with UBL. It was probing outside of the bed. It was working fine with 2.0.6.1 the 2.0.9.2 stable releases. I opened a issue... that was closed in 5mn "working OK".

So, I can do the test, but I feel I will lose the UBL...

I will come back with findings.

ellensp, 2021-12-31:
I have check on a ramps with a ZONESTAR_LCD, I don't see any reason why a BOARD_ANET_10 wouldn't work.

closed 2021-12-31 by ellensp