MarlinFirmware/Marlin #23289: [BUG] Cannot connect MMU2 DIP board to SKR2 REVB F429  (marlin 2.0.9.3)

binarylefmart, 2021-12-11:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Hi,
I have an issue with my new SKR2,
Impossible to communicate with my mmu2-DIP on tft pin.

### Bug Timeline

new

### Expected behavior

Have a communication between SKR2 and MMU2-DIP.

### Actual behavior

No communication between two boards.
Rx and TX are inverted, and i setup the MM2_Serial_PORT 1 (the tft uart is uart 1)
Tried MMU2_Serial Serial, Serial1, MSerial, MSerial1, to quote the first line or tested too on marlin 2.0.9.2 without success.

### Steps to Reproduce

wiring mmu2-dip to skr on tft port.
flash the skr with the latest marlin and muu2 configuration activated.

### Version of Marlin Firmware

2.0.9.3

### Printer model

Sovol Sv01

### Electronics

Bltouch, tmc 2209

### Your Slicer

Prusa Slicer

### Host Software

SD Card (headless)

### Additional information & file uploads

[](https://pastebin.com/KfB8xEsC) Configuration.h

[](https://pastebin.com/8Xx6JhRk) Configuration_adv.h

ellensp, 2021-12-11:
Config files are not attached, not links

thisiskeithb, 2021-12-12:
Someone with a MMU2 and SKR2 just pinged me about that hardware combo not working with the latest bugfix, so this was likely broken recently.

eightheads, 2021-12-12:
I'm running into the same with the latest bugfix from today. SKR2 Rev A and MMU2-DIP.  The mmu2 was also tested on a SKR v1.4 running an older Marlin 2.0 and it communicates fine, but does not seem to connect at all with the SKR2. I also went ahead and tested the TFT port on the SKR2 with a TFT24 and it connects fine with the same current bugfix firmware as using to test the MMU2(swapped the mmu for the TFT, same pins). This seems to be specific to the MMU2 communication.

ellensp, 2021-12-12:
[Configuration files.zip](https://github.com/MarlinFirmware/Marlin/files/7698065/Configuration.files.zip)
Files they upload to discord 
also its a f429 version

eightheads, 2021-12-12:
> [Configuration files.zip](https://github.com/MarlinFirmware/Marlin/files/7698065/Configuration.files.zip) Files they upload to discord also its a f429 version

Mine is a RevA, so it does not seem to be specific to the F429

ellensp, 2021-12-12:
Provided config configs has  
#define MMU2_SERIAL_PORT 1    and 
#define SERIAL_PORT_2 1
which is in conflict

eightheads, 2021-12-12:
> 

I'm running serial -1
And have tried mmu2 serial 1 and 2 without luck

ellensp, 2021-12-12:
TFT port is UART 1  PA9 is USART1_TX and PA10 is USART1_RX

ellensp, 2021-12-12:
The original poster was saying they couldn't test previous versions as they don't support  f429, which is only in latest bugfix.
But adding f429 support is relatively easy.
Just edit the 3 file and add a the file marlin_STM32F429VGT6.json (copy it from bugfix if you want) 
details are here https://github.com/MarlinFirmware/Marlin/pull/23177/files

ellensp, 2021-12-13:
please enable mmu2 debugging  see if its anything like documented here https://revilor.github.io/MMU2-Marlin/debugging.html

eightheads, 2021-12-16:
> please enable mmu2 debugging see if its anything like documented here https://revilor.github.io/MMU2-Marlin/debugging.html or if it is talking at all

The only response I get related to the MMU2 with debug is "RECV: MMU not responding - DISABLED"

ellensp, 2021-12-16:
so it is not talking at all.

eightheads, 2021-12-16:
> so it is not talking at all.

I reset the printer and connected immediately and grabbed it from the start
"
MMU <= reset
Printer is now online.
MMU not responding - DISABLED
"

eightheads, 2021-12-16:
Here is a copy of my current testing configs. These are very minimal. I only have the LCD and MMU2 connected.
[configs.zip](https://github.com/MarlinFirmware/Marlin/files/7723610/configs.zip)

ellensp, 2021-12-16:
can you include picture of how you have it wired, to confirm RX and TX are correct

eightheads, 2021-12-16:
It's connected to the TFT connector. It works fine on my SKR V1.4 with it's TFT connector and according to BTT pinots they're the same.

ellensp, 2021-12-16:
The pinout are not the same, they are reversed and on a different uart. 
skr 1.4 rst rx0 tx0 gnd npwr
skr 2  +5v gnd tx1 rx1 rst

eightheads, 2021-12-16:
> The pinout are not the same, they are reversed and on a different uart. skr 1.4 rst rx0 tx0 gnd npwr skr 2 +5v gnd tx1 rx1 rst

No, one is reversed 180deg compared to the other, but the layout is the same. Different UART channels, yes, but if you look at my config I have it set to channel 1.

ellensp, 2021-12-16:
"the same" != "the same, but I flipped the cable around 180 degrees and didn't tell anyone"

eightheads, 2021-12-16:
> "the same" != "the same, but I flipped the cable around 180 degrees and didn't tell anyone"

Sorry if there was any confusion, but the connector is keyed, so it "fits the same". One board is just rotated 180deg compared to the other.

eightheads, 2021-12-18:
@ellensp I did some further testing this afternoon with my SKR v2 RevA.
**2.0.9.2 Release** - working as expected
**Current Bugfix** - loss of MMU2 connection/control.

I've tried back and forth a couple times with minimal config, just enough to run a LCD and the MMU2 with the same results. Something appears to be broken related to the MMU2 control with the latest bugfix.

ellensp, 2021-12-18:
@eightheads could you try a git bisect?
eg https://git-scm.com/docs/git-bisect
It does a binary search from now back to when it works to find the actual commit that breaks it

J-charles-C, 2021-12-29:
I,

Same problem with an SKR Pro 1.1. Impossible to connect this motherboard to MMU CONTROL Board. I have test lastest bug fix, 2.0.7.2, 2.0.9.2. I don't have Lcd only 12864.

best

thinkyhead, 2022-01-05:
Would you be able to test some `bugfix-2.0.x` commits to narrow this down?

The strategy is to find a commit from some point in the "distant" past where the feature works. Then, test a commit from halfway between that date and today…. And then you keep going to the commit half-way in between your "known to work" commit and your "known to be broken" commit until you find the exact day where it broke.

If you started from a point 256 commits in the past, it would take no more than 8 tests to find the exact commit that broke it.

J-charles-C, 2022-02-05:
> I,
> 
> Same problem with an SKR Pro 1.1. Impossible to connect this motherboard to MMU CONTROL Board. I have test lastest bug fix, 2.0.7.2, 2.0.9.2. I don't have Lcd only 12864.
> 
> best

I, 

some news for this bug ?

Best

ellensp, 2022-03-27:
Just got a MMU2-DIP and and playing catch up
Can Someone confirm the wiring to a SKR2 that used to work?

ellensp, 2022-03-27:
Also those that are having issues.
If you do a MMU reset from the LCD, the device should turn a red LED (D8 near center of board) ) on momentarily and reboot the MMU2-DIP
Does yours do this?

ellensp, 2022-03-30:
Further digging: 

There are two reset modes

A hardware reset mode. Where a IO pin is pulled low then high.  
A software reset mode where "X0" command is sent to the MMU2 controller.

My MMU2-DIP does not seem to support hardware reset on the P1 reset pin

In Marlin hardware reset is selected when you #define MMU2_RST_PIN in Configuration_adv.h

ellensp, 2022-03-30:
It seems to be working fine on current bugfix on a SKR2 REVB F429

Result form sending reset MMU on LCD

> MMU <= reset
> MMU => 'start'
> MMU <= 'S1'
> MMU => 106
> MMU <= 'S2'
> MMU => 130
> MMU <= 'P0'
> MMU => 1
> MMU - ENABLED

Here are the Configs I used for testing.

[bugfix example configs.zip](https://github.com/MarlinFirmware/Marlin/files/8376617/bugfix.example.configs.zip)

closed 2026-10-08 by thinkyhead