MarlinFirmware/Marlin #23290: MKS SGEN_L + MKS Mini 12864 No image[BUG] (bug summary)

ffnull, 2021-12-11:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

While using MKS SGEN_L + MKS Mini 12864 and uncomment #define MKS_MINI_12864 - LCD doesnt works. Encoder and SD slot works, Backlight lights. If using MKS GEN - works without issue

### Version of Marlin Firmware

Latest

### Printer model

Custom

### Electronics

MKS SGEN_L + MKS Mini 12864

Sonicboom247, 2021-12-13:
may be related to #23244

ellensp, 2021-12-17:
No configs. Please provide configs.
what version of MKS SGEN_L?
what version of MKS Mini 12864?

thisiskeithb, 2021-12-17:
Don’t forget to reset your EEPROM with `M502` followed by `M500` after flashing to use the correct contrast values for the LCD.

Contrast is likely initializing/stuck at 0 which is unreadable.

Sonicboom247, 2021-12-17:
how do you reset the eeprom with no LCD and no connection to the printer? i just tested this on my Artillery Ruby with the Fystec Mini 12864 v1 and last nights changes to bugfix 2.0 and still same issue.  If i go back to 2.0.9.1 it works.

ellensp, 2021-12-17:
get a connection to the printer working. Pronterface for eg.

Sonicboom247, 2021-12-17:
just tried with octoprint, can't connect to it, will try pronterface next, oh and @ellensp guess what changed again and can build without it? default_envs = Artillery_Ruby, last time it was default_envs = marlin_Artillery_Ruby...lol

Sonicboom247, 2021-12-17:
so pronterface can't connect either, tested with STM32 programmer and it connected, not sure if these screen grabs help but under USB i now only see STM32 Bootloader and not sure if pid vid helps
[image]
[image]

closed 2022-01-05 by thinkyhead