MarlinFirmware/Marlin #23291: Marlin 2.0.9.2  -   BTT SKR2 fails to drive all TMC2209 Drivers

StogerHaus, 2021-12-11:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

I am using Marlin 2.0.9.2 with a custom delta config based on the example code.

After uploading to the SKR2 (REV_B) board with TMC2209 drivers and doing a test print, the X stepper motor vibrated (moved back and forth very quickly) instead of moving smoothly. Y and Z worked as expected.

I switched the stepper drivers and the X still vibrated. I then switched the motor and X was still vibrating, while Y and Z worked fine. NOT a wiring issue or motor/driver hardware issue. The mA sent to each motor via UART in config_adv. is all the same, set at 800.

I then replaced the X stepper driver with an A4988 and changed the firmware to suit. The X motor then worked fine, however now the Y was vibrating??

I have now changed all the drivers to A4988 and the motors turn as expected, no issues, however I need the TMC2209 to work.

Any help would be much appreciated!

### Bug Timeline

New bug

### Expected behavior

Stepper Motor to move smoothly

### Actual behavior

Stepper Motor Vibrated

### Steps to Reproduce

1. Wire a SKR 2 (REV_B) board with five TMC2209 stepper drivers.
2. Upload Marlin 2.0.9.2 firmware files provided
3. Stepper motors will not all act as expected.

### Version of Marlin Firmware

2.0.9.2

### Printer model

Custom Delta

### Electronics

SKR2 (REV_B) Board with TMC2209 Drivers

### Your Slicer

Other (explain below)

### Host Software

SD Card (headless)

### Additional information & file uploads

[Config_Files.zip](https://github.com/MarlinFirmware/Marlin/files/7697887/Config_Files.zip)

ellensp, 2021-12-11:
Set to TMC2209 and check their status with M122

StogerHaus, 2021-12-12:
>>> M122 S1
SENDING:M122 S1
Testing X connection... OK
Testing Y connection... OK
Testing Z connection... OK
Testing E connection... OK
Testing E1 connection... OK
>>> M122 I
SENDING:M122 I
Testing X connection... OK
Testing Y connection... OK
Testing Z connection... OK
Testing E connection... OK
Testing E1 connection... OK

StogerHaus, 2021-12-12:
> Set to TMC2209 and check their status with M122

>>> M122 S1
SENDING:M122 S1
Testing X connection... OK
Testing Y connection... OK
Testing Z connection... OK
Testing E connection... OK
Testing E1 connection... OK
>>> M122 I
SENDING:M122 I
Testing X connection... OK
Testing Y connection... OK
Testing Z connection... OK
Testing E connection... OK
Testing E1 connection... OK

StogerHaus, 2021-12-12:
Also to note: While interfacing via pronterface, sending movement commands to the motors works with TMC2209. They run smoothly.
However when gcode (which works with A4988) is run on the machine, the X motor seizes up when auto-homing

ellensp, 2021-12-12:
M122 on it own, gives all settings, current limits, mircostepping, everything

StogerHaus, 2021-12-12:
>>> m122
SENDING:M122
		X	Y	Z	E	E1
Address		0	0	0	0	0
Enabled		false	false	false	false	false
Set current	800	800	800	800	800
RMS current	795	795	795	795	795
MAX current	1121	1121	1121	1121	1121
Run current	25/31	25/31	25/31	25/31	25/31
Hold current	12/31	12/31	12/31	12/31	12/31
CS actual	25/31	12/31	12/31	12/31	12/31
PWM scale
vsense		1=.18	1=.18	1=.18	1=.18	1=.18
stealthChop	true	true	true	true	true
msteps		8	16	16	16	16
interp		true	true	true	true	true
tstep		max	max	max	max	max
PWM thresh.
[mm/s]
OT prewarn	false	false	false	false	false
pwm scale sum	29	14	14	14	14
pwm scale auto	0	0	0	0	0
pwm offset auto	36	36	36	36	36
pwm grad auto	14	14	14	14	14
off time	3	3	3	3	3
blank time	24	24	24	24	24
hysteresis
 -end		-1	-1	-1	-1	-1
 -start		1	1	1	1	1
Stallguard thrs	0	0	0	0	0
uStep count	16	8	8	8	8
DRVSTATUS	X	Y	Z	E	E1
sg_result	0	0	0	0	0
stst
olb
ola
s2gb
s2ga
otpw
ot
157C
150C
143C
120C
s2vsa
s2vsb
Driver registers:
		X	0xC0:18:00:00
		Y	0xC0:0C:00:00
		Z	0xC0:0C:00:00
		E	0xC0:0C:00:00
		E1	0xC0:0C:00:00
Testing X connection... OK
Testing Y connection... OK
Testing Z connection... OK
Testing E connection... OK
Testing E1 connection... OK

> M122 on it own, gives all settings, current limits, mircostepping, everything

>>> m122
SENDING:M122
		X	Y	Z	E	E1
Address		0	0	0	0	0
Enabled		false	false	false	false	false
Set current	800	800	800	800	800
RMS current	795	795	795	795	795
MAX current	1121	1121	1121	1121	1121
Run current	25/31	25/31	25/31	25/31	25/31
Hold current	12/31	12/31	12/31	12/31	12/31
CS actual	25/31	12/31	12/31	12/31	12/31
PWM scale
vsense		1=.18	1=.18	1=.18	1=.18	1=.18
stealthChop	true	true	true	true	true
msteps		8	16	16	16	16
interp		true	true	true	true	true
tstep		max	max	max	max	max
PWM thresh.
[mm/s]
OT prewarn	false	false	false	false	false
pwm scale sum	29	14	14	14	14
pwm scale auto	0	0	0	0	0
pwm offset auto	36	36	36	36	36
pwm grad auto	14	14	14	14	14
off time	3	3	3	3	3
blank time	24	24	24	24	24
hysteresis
 -end		-1	-1	-1	-1	-1
 -start		1	1	1	1	1
Stallguard thrs	0	0	0	0	0
uStep count	16	8	8	8	8
DRVSTATUS	X	Y	Z	E	E1
sg_result	0	0	0	0	0
stst
olb
ola
s2gb
s2ga
otpw
ot
157C
150C
143C
120C
s2vsa
s2vsb
Driver registers:
		X	0xC0:18:00:00
		Y	0xC0:0C:00:00
		Z	0xC0:0C:00:00
		E	0xC0:0C:00:00
		E1	0xC0:0C:00:00
Testing X connection... OK
Testing Y connection... OK
Testing Z connection... OK
Testing E connection... OK
Testing E1 connection... OK

StogerHaus, 2021-12-12:
> M122 on it own, gives all settings, current limits, mircostepping, everything

>>> m122
SENDING:M122
		X	Y	Z	E	E1
Address		0	0	0	0	0
Enabled		false	false	false	false	false
Set current	800	800	800	800	800
RMS current	795	795	795	795	795
MAX current	1121	1121	1121	1121	1121
Run current	25/31	25/31	25/31	25/31	25/31
Hold current	12/31	12/31	12/31	12/31	12/31
CS actual	25/31	12/31	12/31	12/31	12/31
PWM scale
vsense		1=.18	1=.18	1=.18	1=.18	1=.18
stealthChop	true	true	true	true	true
msteps		8	16	16	16	16
interp		true	true	true	true	true
tstep		max	max	max	max	max
PWM thresh.
[mm/s]
OT prewarn	false	false	false	false	false
pwm scale sum	29	14	14	14	14
pwm scale auto	0	0	0	0	0
pwm offset auto	36	36	36	36	36
pwm grad auto	14	14	14	14	14
off time	3	3	3	3	3
blank time	24	24	24	24	24
hysteresis
 -end		-1	-1	-1	-1	-1
 -start		1	1	1	1	1
Stallguard thrs	0	0	0	0	0
uStep count	16	8	8	8	8
DRVSTATUS	X	Y	Z	E	E1
sg_result	0	0	0	0	0
stst
olb
ola
s2gb
s2ga
otpw
ot
157C
150C
143C
120C
s2vsa
s2vsb
Driver registers:
		X	0xC0:18:00:00
		Y	0xC0:0C:00:00
		Z	0xC0:0C:00:00
		E	0xC0:0C:00:00
		E1	0xC0:0C:00:00
Testing X connection... OK
Testing Y connection... OK
Testing Z connection... OK
Testing E connection... OK
Testing E1 connection... OK

ellensp, 2021-12-12:
"msteps 8 16 16 16 16"
Ie X is set to 1/8th while all the rest are set to 1/16th

StogerHaus, 2021-12-12:
After sending M122 this sets it to 16, however the pwmscale is different for x than the others. While homing all the motors are making very loud noises. Side note: do I need to enable EEPROM for these settings to save?

>>> M122 I
SENDING:M122 I
		X	Y	Z	E	E1
Address		0	0	0	0	0
Enabled		false	false	false	false	false
Set current	800	800	800	800	800
RMS current	795	795	795	795	795
MAX current	1121	1121	1121	1121	1121
Run current	25/31	25/31	25/31	25/31	25/31
Hold current	12/31	12/31	12/31	12/31	12/31
CS actual	23/31	12/31	12/31	12/31	12/31
PWM scale
vsense		1=.18	1=.18	1=.18	1=.18	1=.18
stealthChop	true	true	true	true	true
msteps		16	16	16	16	16
interp		true	true	true	true	true
tstep		max	max	max	max	max
PWM thresh.
[mm/s]
OT prewarn	false	false	false	false	false
pwm scale sum	25	14	14	14	14
pwm scale auto	0	0	0	0	0
pwm offset auto	36	36	36	36	36
pwm grad auto	14	14	14	14	14
off time	3	3	3	3	3
blank time	24	24	24	24	24
hysteresis
 -end		-1	-1	-1	-1	-1
 -start		1	1	1	1	1
Stallguard thrs	0	0	0	0	0
uStep count	8	8	8	8	8
DRVSTATUS	X	Y	Z	E	E1
sg_result	0	0	0	0	0
stst
olb
ola
s2gb
s2ga
otpw
ot
157C
150C
143C
120C
s2vsa
s2vsb
Driver registers:
		X	0xC0:12:00:00
		Y	0xC0:0C:00:00
		Z	0xC0:0C:00:00
		E	0xC0:0C:00:00
		E1	0xC0:0C:00:00
Testing X connection... OK
Testing Y connection... OK
Testing Z connection... OK
Testing E connection... OK
Testing E1 connection... OK

StogerHaus, 2021-12-12:
Ok I had to enable TMC_DEBUG in config_adv and send M122 I to the board in order to reset the drivers, all values looking normal now and motors working (very loudly, definitely not in stealthchop) but thats another issue. Thanks for the help ellensp!

closed 2021-12-12 by StogerHaus