MarlinFirmware/Marlin #23294: [FR] Power loss recovery by measuring xyz position

anotherboringuser, 2021-12-12:
### Describe the feature you want

I don’t know if this is possible but it may be useful. It’s certainly not urgent. 
The idea is that after a power outage, the printer heats up the heated bed and nozzle and then moves up a bit. Now it homes the machine. While homing, it counts how many step signals are send to the stepper drivers and calculates the position of where the tool head was when the power went out. Now it searches the gcode for the closest coordinates to the measured position (it would also need to take into account that the print might have stopped between the points A and B but only coordinates for A and B are written in the gcode. Implementing this feature just based on the Z height is probably much easier. ). Now the user could resume the print, no power loss detection or prevention hardware needed and without wearing out an SD card. 
Caveats: 
-	The printer needs to retain its position, especially in z direction, when the power is cut. 
-	The printer needs to be able to home the Z axis safely. It would be useful to have the option to move the X and Y axis around after they are homed but before z is homed. 
The option to start the recovery routine could be hidden somewhere deep in the menus. The printer has no clue what happened before the outage, so the user needs to start the recovery routine.

ellensp, 2021-12-12:
It would take a approximately the same time it takes to print to locate the correct location. Every gcode line would need to be read and processed, to add all the relative offsets together to work out the current loction.
Ie if it stops after 2 hours, it will take 2 hours to resume!

ellensp, 2021-12-12:
The other issue being, what gcode file on the sdcard/usb stick was being processed?   If you need to save that information, you may as well save the position periodically, ie what we have now.

ellensp, 2021-12-12:
The printer does not visit each coordinate uniquely.
Any move the the head moves over the bed to get to where it wants to start printing.
How do you know if was printing or just moving to a location when it stopped?
There is no way to know. So no way to work out the resume point.

closed 2021-12-12 by ellensp