MarlinFirmware/Marlin #23300: [BUG] Unable to compile Marlin-bugfix-2.0.x

TheSnowButcher, 2021-12-15:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Under .vscode>c_cpp_properties.json I get the below error that prevents me from compiling the firmware.
[{
	"resource": "/c:/Marlin Firmware/Marlin-bugfix-2.0.x/.vscode/c_cpp_properties.json",
	"owner": "c:\\Marlin Firmware\\Marlin-bugfix-2.0.x",
	"severity": 4,
	"message": "Cannot find \"C:\\Users\\USER\\.platformio\\packages\\framework-arduinoststm32@4.10900.200819\\variants\\MARLIN_F103Rx\".",
	"startLineNumber": 39,
	"startColumn": 17,
	"endLineNumber": 39,
	"endColumn": 116
}]

### Bug Timeline

First experienced 12/14/2021

### Expected behavior

I expected the firmware to compile

### Actual behavior

It failed out because it is trying to point to a folder that doesn't exist.

### Steps to Reproduce

1). Download Marlin 2.0.9.1 and unzip it.
2) Download the configuration files from the Configurations-bugfix-2.0.x\config\examples\Creality\Ender-5 Pro\CrealityV422 folder
3) Install those files in the Marlin directory
4) Make changes per https://3dprintscape.com/marlin-firmware-on-creality-board-complete-guide/
6) Verified that changes in bug ticket https://github.com/MarlinFirmware/Marlin/issues/22606 were made in current build
5) Attempt to compile

### Version of Marlin Firmware

Bugfix-2.0.x

### Printer model

Creality Ender 5 Pro

### Add-ons

CR Touch

### Bed Leveling

ABL Bilinear mesh

ellensp, 2021-12-15:
Not a marlin issue, computer issue 
for reasons unknown your python is not running buildroot/share/PlatformIO/scripts/generic_create_variant.py
which copies buildroot/share/PlatformIO/variants/MARLIN_F103Rx into C:\Users\USER\.platformio\packages\framework-arduinoststm32@4.10900.200819\variants\MARLIN_F103Rx

ellensp, 2021-12-15:
Issue #22606 seems completely unrelated
Its for a Anycubic Predator using a TRIGORILLA_PRO controller.

ellensp, 2021-12-15:
```
1) Download Marlin 2.0.9.1 and unzip it.
2) Download the configuration files from the Configurations-bugfix-2.0.x\config\examples\Creality\Ender-5 Pro\CrealityV422 folder

```
You are miss matching versions and config files
you cannot use bugfix configs on release 2.0.9.1 code
you need release config files from https://github.com/MarlinFirmware/Configurations/archive/refs/heads/release-2.0.9.1.zip

or use bugfix code

closed 2021-12-15 by ellensp