MarlinFirmware/Marlin #23302: [FR] GCode to manually start and end "Print Mode"

mAd-DaWg, 2021-12-15:
### Is your feature request related to a problem? Please describe.

Marlin configured for cnc routing(see attached configurations and current gcode), meaning no extruders and no hotends. Homing is also done manually before hand to find the workspace zero co-ordinates.

When starting a "print"/cut, marlin does not recognize and enter print mode. This is both an annoyance and a safety hazard, as there is no straight forward way to pause, stop, adjust feedrate, etc that is usually available in print mode. The only way around this is to run to a nearby computer and frantically lookup and send commands manually, or hit the emergency off switch, which could cause the spindle to slam into the work piece(in my case), ruining it, or potentially causing other dangerous or expensive outcomes. Interfaces that link up to the controller are also unable to stop "prints", as marlin does not recognize that it is in the middle of a "print" and therefore does nothing with stop print commands.

****

### Describe the feature you want

I propose 2 GCode commands be added that can directly tell marlin a print is starting and ending(to be appended and prepended to gcode files). This allows for more flexibility for new machines and use cases, as new heuristics would not need to be added before the machine can be used as intended

### Additional context

[configuration.zip](https://github.com/MarlinFirmware/Marlin/files/7719432/configuration.zip)
[example_gcode.zip](https://github.com/MarlinFirmware/Marlin/files/7719435/example_gcode.zip)
)

ellensp, 2021-12-15:
* Marlin knows a print job is running when:
 *  1. Running a print job from media started with M24.
 *  2. The Print Job Timer has been started with M75.
 *  3. The heaters were turned on and PRINTJOB_TIMER_AUTOSTART is enabled.
 
Ie add these to your gcode.
https://marlinfw.org/docs/gcode/M075.html
https://marlinfw.org/docs/gcode/M077.html

ellensp, 2021-12-15:
So in my view this is already implemented.

closed 2021-12-15 by ellensp

mAd-DaWg, 2021-12-15:
M75 does seem to work, thanks! Please update the documentation to reflect this behavior.