MarlinFirmware/Marlin #23304: [BUG] Ender 3 V2 - 4.2.2 Broken and unusable UI after flashing new firmware (Marlin 2.0.9.2)

Balltimore, 2021-12-15:
### Did you test the latest `bugfix-2.0.x` code?

No, but I will test it now!

### Bug Description

After building and flashing new Marlin 2.0.9.2 with correct default configuration files for Ender 3 v2 (board 4.2.2) UI on stock printer display is completely broken and unusable. Bootscreen is just blank logo and after initialization UI don't have any graphical elements and most menus are displayed improperly.

During building process that are no errors.

I tried reflashing DWIN_SET multiple times but same UI behaviour occurs.

This bechaviour exist on Ender-3 v2 OEM display(#DWIN_CREALITY_LCD) and on MarlinUI for Creality's DWIN (#DWIN_MARLINUI_PORTRAIT/LANDSCAPE)

I was unable to test Ender-3 v2 OEM display with enhancements by Jacob Myers because of occurring build errors.

This problem was also introduced to my modified version of marlin after updating it to 2.0.9.2 (This is why on some pictures there is text saying Ender 3 V2.1)

Before using this version UI and printer worked correctly.

### Bug Timeline

First time after updating to current release

### Steps to Reproduce

Clone current 2.0.x branch
Copy correct configuration files for marlin 2.0.9.2 from marlin config repo
Build and flash Ender 3 v2

### Version of Marlin Firmware

Marlin 2.0.9.2

### Printer model

Ender 3 V2

### Electronics

Stock mainboard 4.2.2

### Add-ons

BLTouch

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Cura

### Host Software

OctoPrint

### Additional information & file uploads

Stock Config - [Config.zip](https://github.com/MarlinFirmware/Marlin/files/7721933/Config.zip)

My Modified Marlin- [MarlinModified.zip](https://github.com/MarlinFirmware/Marlin/files/7721973/MarlinModified.zip)
``` I think my changes are irrelevant because this bug occured only after updating to 2.0.9.2 ```

Images of broken UI
[image]
[image]
[image]
[image]
[image]

The-EG, 2021-12-15:
Verify that the sdcard you are using to flash the LCD is formated to fat32 with 4096 allocation size. It *absolutely, positively MUST* be 4096 allocation size. It will not work otherwise, and if you haven't formatted the card yourself it's probably not in that format.

As noted in the examples readme, the screen will change colors during flashing, from blue to orange. If the screen doesn't stay blue for more than a second or so during this process, the flashing isn't working properly; see the above.

Balltimore, 2021-12-15:
Thank you for helpful tips but even without reflashing DWIN_SET problem is gone just by rolling back to marlin 2.0.9. UI is working properly now... at least I have usable printer back. I hope this problem will be solved in next update

The-EG, 2021-12-15:
The DWIN_SET assets were changed between 2.0.9 and 2.0.9.2. You are missing icons because the LCD hasn't been updated with the latest version of the assets.

This isn't a bug with that version, so you'll eventually have to flash the DWIN properly if you want to use a version newer than 2.0.9 because all versions after that will require the new assets.

Balltimore, 2021-12-15:
I 100% agree with argument that this issue is mostly created by incorrect DWIN assets but I tried reflashing DWIN_SET multiple times using DWIN_SET folder from Marlin 2.0.9.2 configuration repository.  As in description ^^

https://github.com/MarlinFirmware/Configurations/tree/release-2.0.9.2/config/examples/Creality/Ender-3%20V2/DWIN_SET

This is why I opened this issue because to my understating I'm using correct DWIN_SET for that version.

thisiskeithb, 2021-12-15:
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- [Marlin Documentation](https://marlinfw.org) hosted on MarlinFW.org
- [Marlin Forum](https://forums.reprap.org/list.php?415) hosted on RepRap.org
- [Marlin Firmware](https://www.facebook.com/groups/1049718498464482/) Facebook Group 
- [Marlin Firmware for 3D Printers](https://www.facebook.com/groups/3Dtechtalk/) Facebook Group
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- [Marlin Firmware](https://discord.gg/n5NJ59y) Discord server

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2021-12-15 by thisiskeithb

RipperDrone, 2021-12-30:
@Balltimore - I'd like to leave you my solution here, even though @thisiskeithb pointed you to the correct channels for support: I had exactly the same issue, and the simple answer was: use another SD card... check this out:

Nothing had worked for me, going through 2 different SD cards of sizes 4GB/32GB - properly formatted to FAT32/4K... but then...

I took the pain of finding a THIRD microSD, and I only had 64GB ones - found an instruction how to format them down to <32GB via Windows data storage manager (create simple volume, enter smaller size, ...). I purposedly made it a 2GB drive only this time - and guess what: Update worked now. Just 1 minute ago, I am super happy. Blue, then orange screen, firmware successfully flashed. Seems the LCD SD card drive is super sensitive to micro SD card types/sizes/timings? So: Stick in, find yourself some more SD cards to try ;-)