MarlinFirmware/Marlin #23308: [BUG] BL Touch probe becomes inoperative when RGB LEDs are turned on

rcdaley, 2021-12-16:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

With RGB LEDs and BL Touch enabled and configured, BL Touch stops working (fails to deploy or stow) whenever the RGB LEDs are turned on.  If the RGB LEDs are cycled on then off, the touch probe still fails to work.  The touch probe will remain inoperative until the board is reset.  

### Bug Timeline

Probably new to the STM32 F4

### Expected behavior

I expect the touch probe to work independent of the RGB LED state.

### Actual behavior

Touch probe is inoperative, z-axis crashes during homing.

### Steps to Reproduce

1) I would start with a STM32 F4 platform. 
2) enable BL Touch and RGB LED functions
3) See if touch probe deploy/stow functions work properly.

### Version of Marlin Firmware

2.0.9.2

### Printer model

Custom/STMEVAL-3DP001

### Electronics

STMEVAL-3DP001

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Cura

### Host Software

OctoPrint

### Additional information & file uploads

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/7724675/Configuration.zip)

sjasonsmith, 2021-12-16:
Your pin 47 (PD_15 from variant.h) uses timer 4 for PWM output, which is the default timer used by that board for the Servo PWM used by the BLTouch.

You'll need to choose another value for TIMER_SERVO in `buildroot\share\PlatformIO\variants\MARLIN_STEVAL_F401VE\variant.h`.

You probably have several options for the timer, but since you don't have `SPEAKER` enabled I'd suggest just using `TIM5` currently assigned to `TIMER_TONE`.

rcdaley, 2021-12-17:
Thanks for the quick response.  I suspected it was a resource conflict but had no idea where to look. I changed TIMER_SERVO to use TIM5 as you suggested and everything works perfectly.  Thanks again.

closed 2021-12-17 by rcdaley