MarlinFirmware/Marlin #23309: [BUG] (Auto Delta calibration no longer works on Anycubic Linear Plus)

OneOfEleven, 2021-12-16:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Using latest v2.0.9.2, firstly it does not compile using Arduino IDE 1.8.16 (see below).
To get it to compile we had to temporarily comment out line 148 in "language.h".

Anyway, auto delta calibration does not work.

Machine is an Anycubic Kossel Linear Plus.

Upload newly compiled firmware.
M502 M500
Reboot printer firmware.
G28 to home it.
M111 S247 for enabling logging.
G33 to do an auto delta calibration.

At this point it starts OK, does a single center probe followed by 3 outter probes close around the center, at which point the probing stops with the following message ..

"Correct delta settings with M665 and M666"

The full log is attached along with the config files.

Bed levelling (G29) works OK, delta height (G33 P1) works OK, just no auto delta calibration (G33),

Marlin v1.1.19 works without fail. There are no hardware problems (limit switches, probe switch etc).

========================
Compilation problem temporary fix is to comment out line 148 in "language.h"

In file included from C:\Projects\3D Printer\AnyCubic Kossel Linear Plus\Marlin-2.0.9.2\Marlin\src\lcd\extui\ftdi_eve_touch_ui\language\language.h:23:0,
                 from C:\Projects\3D Printer\AnyCubic Kossel Linear Plus\Marlin-2.0.9.2\Marlin\src\lcd\extui\ftdi_eve_touch_ui\language\language.cpp:25:
C:\Projects\3D Printer\AnyCubic Kossel Linear Plus\Marlin-2.0.9.2\Marlin\src\lcd\extui\ftdi_eve_touch_ui\language\language_en.h:148:24: error: redefinition of 'const char Language_en::MSG_MESH_VIEW []'
   PROGMEM Language_Str MSG_MESH_VIEW                = u8"View Mesh";
                        ^~~~~~~~~~~~~
In file included from C:\Projects\3D Printer\AnyCubic Kossel Linear Plus\Marlin-2.0.9.2\Marlin\src\core\language.h:563:0,
                 from C:\Projects\3D Printer\AnyCubic Kossel Linear Plus\Marlin-2.0.9.2\Marlin\src\inc\MarlinConfig.h:53,
                 from C:\Projects\3D Printer\AnyCubic Kossel Linear Plus\Marlin-2.0.9.2\Marlin\src\MarlinCore.h:24,
                 from C:\Projects\3D Printer\AnyCubic Kossel Linear Plus\Marlin-2.0.9.2\Marlin\src\lcd\extui\ftdi_eve_touch_ui\language\language.cpp:23:
C:\Projects\3D Printer\AnyCubic Kossel Linear Plus\Marlin-2.0.9.2\Marlin\src\lcd\language\language_en.h:160:8: note: 'const char Language_en::MSG_MESH_VIEW [10]' previously defined here
   LSTR MSG_MESH_VIEW                      = _UxGT("View Mesh");
        ^~~~~~~~~~~~~

===============

### Bug Timeline

No idea

### Expected behavior

We expect it to complete the auto delta calibration routine much like v1.1.19 does.

### Steps to Reproduce

1) Upload newly compiled firmware
2) "M502"
3) "M500"
4) Reboot printer firmware
5) "G28" to home
6) "M111 S247" to enable full logging.
7) "G33" to start an auto delta calibration.

### Version of Marlin Firmware

v2.0.9.2 latest source

### Printer model

Anycubic Kossel Linear Plus

### Electronics

Stock/original board

### Add-ons

None

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Cura

### Host Software

Pronterface

### Additional information & file uploads

[config_log_files.zip](https://github.com/MarlinFirmware/Marlin/files/7727489/config_log_files.zip)

Using latest v2.0.9.2, firstly it does not compile using Arduino IDE 1.8.16 (see below).
To get it to compile we had to temporarily comment out line 148 in "language.h".

Anyway, auto delta calibration does not work.
It works fine using Marlin v1.1.19.
G29 bed levelling works, just not auto delta calibration (G33). 
There are no hardware faults, limit switches are all fine, Z-probe switch is fine.

It starts to probe OK, does a single center probe followed by 3 outter probes close around the center, at which point the probing stops with the following message ..

"Correct delta settings with M665 and M666"

========================
Compilation problem temporary fix is to comment out line 148 in "language.h"

In file included from C:\Projects\3D Printer\AnyCubic Kossel Linear Plus\Marlin-2.0.9.2\Marlin\src\lcd\extui\ftdi_eve_touch_ui\language\language.h:23:0,
                 from C:\Projects\3D Printer\AnyCubic Kossel Linear Plus\Marlin-2.0.9.2\Marlin\src\lcd\extui\ftdi_eve_touch_ui\language\language.cpp:25:
C:\Projects\3D Printer\AnyCubic Kossel Linear Plus\Marlin-2.0.9.2\Marlin\src\lcd\extui\ftdi_eve_touch_ui\language\language_en.h:148:24: error: redefinition of 'const char Language_en::MSG_MESH_VIEW []'
   PROGMEM Language_Str MSG_MESH_VIEW                = u8"View Mesh";
                        ^~~~~~~~~~~~~
In file included from C:\Projects\3D Printer\AnyCubic Kossel Linear Plus\Marlin-2.0.9.2\Marlin\src\core\language.h:563:0,
                 from C:\Projects\3D Printer\AnyCubic Kossel Linear Plus\Marlin-2.0.9.2\Marlin\src\inc\MarlinConfig.h:53,
                 from C:\Projects\3D Printer\AnyCubic Kossel Linear Plus\Marlin-2.0.9.2\Marlin\src\MarlinCore.h:24,
                 from C:\Projects\3D Printer\AnyCubic Kossel Linear Plus\Marlin-2.0.9.2\Marlin\src\lcd\extui\ftdi_eve_touch_ui\language\language.cpp:23:
C:\Projects\3D Printer\AnyCubic Kossel Linear Plus\Marlin-2.0.9.2\Marlin\src\lcd\language\language_en.h:160:8: note: 'const char Language_en::MSG_MESH_VIEW [10]' previously defined here
   LSTR MSG_MESH_VIEW                      = _UxGT("View Mesh");
        ^~~~~~~~~~~~~

========================

sjasonsmith, 2021-12-16:
This has already been fixed in bugfix-2.0.x and verified by multiple people (myself included).

I’m closing this issue, but please reply if you still encounter problems with the bugfix branch.

closed 2021-12-16 by sjasonsmith

OneOfEleven, 2021-12-16:
> This has already been fixed in bugfix-2.0.x and verified by multiple people (myself included).
> 
> I’m closing this issue, but please reply if you still encounter problems with the bugfix branch.

Yes by all means close it @sjasonsmith, sorry about that, used the latest 2.0.9.2 instead of bugfix ver.

I'm running the bugfix one now and all seems good!

My bad.