MarlinFirmware/Marlin #23313: [FR] Artillery Genius Pro support

gudvinr, 2021-12-17:
### Are you looking for hardware support?

Brand: Artillery 3D
Model: Genius Pro
Board: Artillery Ruby custom board (STM32F401RC)

### Describe the feature you want

Upstream board/printer support

### Additional context

Genius Pro uses same Artillery Ruby board as Sidewinder X2, which support was added recently in #23029  
There's no configuration for Genius Pro, however. Which is expected because there were no issues about that particular printer.

Artillery has Marlin source code for 2.0.9.1 in their repo: https://github.com/artillery3d/genius-pro-firmware

Sonicboom247, 2021-12-17:
pretty sure thats the same board on the Hornet as well. hopefully the display is not the same Fysetc 12864 v1

ellensp, 2021-12-17:
Its a touch fully colour display, so no not a  Fysetc 12864, but and extreamly limited in functionally touch interface

gudvinr, 2021-12-17:
There's display firmware too, albeit only binaries available.

https://github.com/artillery3d/genius-pro-tft-firmware

thisiskeithb, 2021-12-27:
See https://github.com/MarlinFirmware/Configurations/pull/641.

closed 2021-12-27 by thisiskeithb

Dviros, 2021-12-30:
Hi @thisiskeithb,
Can confirm that flashing latest 2.0.9.3 with your default configs results in my card not booting, requiring to boot to DFU (using the sad warranty voiding boot0+3.3v short circuit) and revert to Artillery's 2.0.9.1 :\
Using digant's TFT28 firmware as well.

I've compared it all - artillery's configs, project ini's and did not find anything meaningful.

Using Genius Pro and can share my configs if required.

Many thanks and Happy New Year!

thisiskeithb, 2021-12-30:
> Can confirm that flashing the latest 2.0.9.3 with your artillery configs results in my ST32 Ruby card not booting

Download 2.0.9.3 again. There was a bug that prevented boards from booting so it was fixed and re-tagged.

Dviros, 2021-12-30:
OMG. Will do, thanks a lot!

Dviros, 2021-12-30:
@thisiskeithb
Can confirm that I've tested the latest 2.0.9.3 tag and the commits in the repo, both are not booting :(
Any way to double check this please?

thisiskeithb, 2021-12-30:
@Dviros Please open a new [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose) so this can be properly tracked.

Sonicboom247, 2021-12-30:
sad, was hoping this would also address the marlin crash on boot according to @ellensp on the Hornet with the same board.

Dviros, 2021-12-30:
https://github.com/MarlinFirmware/Marlin/issues/23393