MarlinFirmware/Marlin #23317: [BUG ]Neopixel LED issue

DVSVIDEO, 2021-12-19:
### Did you test the latest `bugfix-2.0.x` code?

No, but I will test it now!

### Bug Description

My WS2811 type 12V Neopixel LED strip worked perfectly with the previous version of marlin (2.0.6). Since unufied bed leveling did not work properly, I replaced it with the latest marlin (2.9.0.2) CONFIGURATION_H_VERSION 02000902)). The UBL now works perfectly, but the Neopixel tape does not work properly. It does not perform the test at startup and only the first of the 4 segments is willing to turn on, but only the blue color.
I have reloaded the old firmware (CONFIGURATION_H_VERSION 020006) and it works fine again. The settings in the configuratuion.h file are exactly the same, so this clearly looks like a BUG.
However, UBL does not work properly here (does not finish leveling)

### Bug Timeline

marlin 2.0.6 ok, marlin 2.0.9.2 issue

### Expected behavior

The neopixel LED strip for normal operation

### Actual behavior

The WS2811 LED strip is incorrectly controlled. Only the first segment lights up, but here too only the color blue. The blue color on the first segment can only be turned on and off, nothing else works.

### Steps to Reproduce

1. Old marlin 2.0.6 version, WS2811 Neopixel LED is ok
2. latest marlin 2.0.9.2 version LED is not working
3. reload old 2.0.6 marlin LED is work normally again

### Version of Marlin Firmware

2.0.9.2

### Printer model

DIY cartesian core XZ printer

### Electronics

SKR 1.4 turbo board, TFT35 touch screen

### Add-ons

none

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Simplify3D

### Host Software

SD Card (headless)

### Additional information & file uploads

/**
 * RGB LED / LED Strip Control
 *
 * Enable support for an RGB LED connected to 5V digital pins, or
 * an RGB Strip connected to MOSFETs controlled by digital pins.
 *
 * Adds the M150 command to set the LED (or LED strip) color.
 * If pins are PWM capable (e.g., 4, 5, 6, 11) then a range of
 * luminance values can be set from 0 to 255.
 * For NeoPixel LED an overall brightness parameter is also available.
 *
 * *** CAUTION ***
 *  LED Strips require a MOSFET Chip between PWM lines and LEDs,
 *  as the Arduino cannot handle the current the LEDs will require.
 *  Failure to follow this precaution can destroy your Arduino!
 *  NOTE: A separate 5V power supply is required! The NeoPixel LED needs
 *  more current than the Arduino 5V linear regulator can produce.
 * *** CAUTION ***
 *
 * LED Type. Enable only one of the following two options.
 */
//#define RGB_LED
//#define RGBW_LED

#if EITHER(RGB_LED, RGBW_LED)
  //#define RGB_LED_R_PIN 34
  //#define RGB_LED_G_PIN 43
  //#define RGB_LED_B_PIN 35
  //#define RGB_LED_W_PIN -1
#endif

// Support for Adafruit NeoPixel LED driver
#define NEOPIXEL_LED
#if ENABLED(NEOPIXEL_LED)
  #define NEOPIXEL_TYPE  NEO_BRG // NEO_GRBW / NEO_GRB - four/three channel driver type (defined in Adafruit_NeoPixel.h)
  #define NEOPIXEL_PIN   P1_24    // LED driving pin
  //#define NEOPIXEL2_TYPE NEOPIXEL_TYPE
  //#define NEOPIXEL2_PIN    5
  #define NEOPIXEL_PIXELS 4       // Number of LEDs in the strip. (Longest strip when NEOPIXEL2_SEPARATE is disabled.)
  #define NEOPIXEL_IS_SEQUENTIAL   // Sequential display for temperature change - LED by LED. Disable to change all LEDs at once.
  #define NEOPIXEL_BRIGHTNESS 255  // Initial brightness (0-255)
  #define NEOPIXEL_STARTUP_TEST  // Cycle through colors at startup

  // Support for second Adafruit NeoPixel LED driver controlled with M150 S1 ...
  //#define NEOPIXEL2_SEPARATE
  #if ENABLED(NEOPIXEL2_SEPARATE)
    #define NEOPIXEL2_PIXELS      15  // Number of LEDs in the second strip
    #define NEOPIXEL2_BRIGHTNESS 127  // Initial brightness (0-255)
    #define NEOPIXEL2_STARTUP_TEST    // Cycle through colors at startup
  #else
    //#define NEOPIXEL2_INSERIES      // Default behavior is NeoPixel 2 in parallel
  #endif

  // Use some of the NeoPixel LEDs for static (background) lighting
  //#define NEOPIXEL_BKGD_INDEX_FIRST  0              // Index of the first background LED
  //#define NEOPIXEL_BKGD_INDEX_LAST   5              // Index of the last background LED
  //#define NEOPIXEL_BKGD_COLOR { 255, 255, 255, 0 }  // R, G, B, W
  //#define NEOPIXEL_BKGD_ALWAYS_ON                   // Keep the backlight on when other NeoPixels are off
#endif

DVSVIDEO, 2021-12-19:
Sorry: The version in which the Neopixel LED works well: 2.0.6.1 (Aug 28, 2020)
I've tried version 2.0.7 and it doesn't work well anymore.

DVSVIDEO, 2021-12-19:
I experimented all day and realized something:
The version 2.0.6.1 in which my 12V Neopixel tape works is a version called: Marlin-2.0.x-SKR-V1.4-Turbo
I've been downloading it for a long time and I didn't think it was set up specifically for this motherboard.
I downloaded a plain version 2.0.6.1 and copied the configuration files. When I compile everything is fine, but the size of the compiled firmware.bin is slightly different and with that the LED strip doesn't work.
Same firmware, but something different.
The wording of the error is therefore:
The 12V WS2811 (Neopixel) LED strip does not work properly with any firmware except the Marlin-2.0.x-SKR-V1.4-Turbo version.
The problem is that UBL is not working properly in this.
My request is to improve the control of the marlin LED so that I can use the latest version as well.
Thanks

ellensp, 2021-12-19:
Real marlin 2.0.6.1 and the BTT version from https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/tree/master/BTT%20SKR%20V1.4/Firmware/Marlin-2.0.x-SKR-V1.4-Turbo  has no significant code changes
Apart from the expected Configuration files.

ellensp, 2021-12-20:
All differences outside of Config files

```DIFF
diff --git a/Marlin/src/module/temperature.cpp b/Marlin/src/module/temperature.cpp
index b162173bc6..079d945260 100644
--- a/Marlin/src/module/temperature.cpp
+++ b/Marlin/src/module/temperature.cpp
@@ -45,12 +45,12 @@
 #endif
 
 #if ENABLED(MAX6675_IS_MAX31865)
-  #include <Adafruit_MAX31865.h>
+  #include "Adafruit_MAX31865.h"
   #ifndef MAX31865_CS_PIN
-    #define MAX31865_CS_PIN     MAX6675_SS_PIN  // HW:49   SW:65    for example
+    #define MAX31865_CS_PIN     PC9 //MAX6675_SS_PIN  // HW:49   SW:65    for example
   #endif
   #ifndef MAX31865_MOSI_PIN
-    #define MAX31865_MOSI_PIN   MOSI_PIN        //            63
+    #define MAX31865_MOSI_PIN   PC5 //MOSI_PIN        //            63
   #endif
   #ifndef MAX31865_MISO_PIN
     #define MAX31865_MISO_PIN   MAX6675_DO_PIN  //            42
diff --git a/Marlin/src/pins/stm32f4/pins_BTT_SKR_PRO_V1_1.h b/Marlin/src/pins/stm32f4/pins_BTT_SKR_PRO_V1_1.h
index fb4b17b6fb..de831c07c2 100644
--- a/Marlin/src/pins/stm32f4/pins_BTT_SKR_PRO_V1_1.h
+++ b/Marlin/src/pins/stm32f4/pins_BTT_SKR_PRO_V1_1.h
@@ -28,3 +28,19 @@
 #define BOARD_INFO_NAME "BTT SKR Pro V1.1"
 
 #include "pins_BTT_SKR_PRO_common.h"
+
+// SPI for Max6675 or Max31855 Thermocouple
+// Uses a separate SPI bus
+// If you have a two-way thermocouple, you can customize two THERMO_CSx_PIN pins (x:1~2)
+#if 0
+#define THERMO_SCK_PIN                      PF10   // SCK
+#define THERMO_DO_PIN                       PF9   // MISO
+#define THERMO_CS1_PIN                      PF8   // CS1
+#define THERMO_CS2_PIN                      PC9   // CS2
+
+#define MAX6675_SS_PIN            THERMO_CS1_PIN
+#define MAX6675_SS2_PIN           THERMO_CS2_PIN
+#define MAX6675_SCK_PIN           THERMO_SCK_PIN
+#define MAX6675_DO_PIN             THERMO_DO_PIN
+
+#endif
\ No newline at end of file
diff --git a/Marlin/src/pins/stm32f4/pins_BTT_SKR_PRO_common.h b/Marlin/src/pins/stm32f4/pins_BTT_SKR_PRO_common.h
index 56c509562f..dae3823166 100644
--- a/Marlin/src/pins/stm32f4/pins_BTT_SKR_PRO_common.h
+++ b/Marlin/src/pins/stm32f4/pins_BTT_SKR_PRO_common.h
@@ -218,7 +218,7 @@
 //
 // Temperature Sensors
 //
-#define TEMP_0_PIN                          PF4   // T1 <-> E0
+#define TEMP_0_PIN                          PF7//PF4   // T1 <-> E0
 #define TEMP_1_PIN                          PF5   // T2 <-> E1
 #define TEMP_2_PIN                          PF6   // T3 <-> E2
 #define TEMP_BED_PIN                        PF3   // T0 <-> Bed
diff --git a/Marlin/src/pins/stm32f4/pins_RUMBA32_common.h b/Marlin/src/pins/stm32f4/pins_RUMBA32_common.h
index 3b513b31ca..dbc1eb462b 100644
--- a/Marlin/src/pins/stm32f4/pins_RUMBA32_common.h
+++ b/Marlin/src/pins/stm32f4/pins_RUMBA32_common.h
@@ -138,6 +138,7 @@
 #define BTN_PIN                             PC10
 #define PS_ON_PIN                           PE11
 #define KILL_PIN                            PC5
+#define NEOPIXEL_PIN                        PB4
 
 #define SDSS                                PA2
 #define SD_DETECT_PIN                       PB0
diff --git a/platformio.ini b/platformio.ini
index e3ff6c1526..df360cf0cc 100644
--- a/platformio.ini
+++ b/platformio.ini
@@ -18,7 +18,7 @@
 [platformio]
 src_dir      = Marlin
 boards_dir   = buildroot/share/PlatformIO/boards
-default_envs = mega2560
+default_envs = LPC1769
 include_dir  = Marlin
 
 #
@@ -1140,6 +1140,7 @@ board             = BigTree_SKR_Pro
 build_flags       = ${common_stm32.build_flags}
   -DUSB_PRODUCT=\"STM32F407ZG\"
   -DTARGET_STM32F4 -DSTM32F407_5ZX -DVECT_TAB_OFFSET=0x8000
+  -DUSBD_USE_CDC_COMPOSITE -DUSE_USB_FS
 extra_scripts     = ${common.extra_scripts}
   pre:buildroot/share/PlatformIO/scripts/generic_create_variant.py
 #upload_protocol   = stlink

```

DVSVIDEO, 2021-12-20:
> All differences outside of Config files
>
> ```diff
> diff --git a/Marlin/src/module/temperature.cpp b/Marlin/src/module/temperature.cpp
> index b162173bc6..079d945260 100644
> --- [...]

This means that without modification, marlin is not compatible with the BTT SKR 1.4 Turbo Card. (at least the LED control is not).
Unfortunately, it is beyond my knowledge to intervene outside of configuration files.
Then it stays with 2.0.6.1 with the faulty UBL (Bilinear leveling works, I use it).
Thanks for the reply!

DVSVIDEO, 2021-12-20:
This means that without modification, marlin is not compatible with the BTT SKR 1.4 Turbo Card. (at least the LED control is not).
Unfortunately, it is beyond my knowledge to intervene outside of configuration files.
Then it stays with 2.0.6.1 with the faulty UBL (Bilinear leveling works, I use it).
Thanks for the reply!

DVSVIDEO, 2021-12-23:
And something is different in the marlin at this link, https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/tree/master/BTT%20SKR%20V1.4/Firmware/Marlin-2.0.x-SKR-V1 .4-Turbo because it works, the original https://github.com/MarlinFirmware/Marlin/releases/tag/2.0.6.1
and no (already like Neopixel control)

DVSVIDEO, 2021-12-29:
I'm not a programmer, so I ask for help:
What is the difference between the original marlin 2.0.6.1 and the marlin 2.0.6.1 dedicated to the BTT SKR 1.4 Turbo motherboard?
Especially since in the former the Neopixel control is faulty and in the latter the UBL leveling does not go through. (I used the same configuration files in both projects - I copied them).
Thanks in advance.

zonalimitatore, 2021-12-29:
I've the same board,with 8 (1 bar) of ws2812.
It work well.
Why do you use 12v? Max input for that led it's 7v, and the board (on dedicated port), give 5v

DVSVIDEO, 2021-12-29:
Also available in 12V version. I use this because it was left at home from a previous project.
Yes the Neopixel connector on the motherboard gives 5V, but all I have to do is take the power supply from the 12V connector, not this one. The control signal is from the control connector.
Slap is simple and also works with the firmware I downloaded from BTT for approx. Half year ago. When I want a fresher marlin, the led doesn't work properly. I didn’t touch on the physical part, the problem is in the software.
[image]

zonalimitatore, 2021-12-30:
I see that you have define only 4leds.
How many LEDs have?

ellensp, 2021-12-31:
was looking at these 12V WS2812 Neopixel (having not seen then before), since they are still 5v chip/leds they run them in sets of 3
ie for every 1 led you turn on 3 leds comes on
So with the 12? leds on the left 4 would be the correct setting.

zonalimitatore, 2021-12-31:
> was looking at these 12V WS2812 Neopixel (having not seen then before), since they are still 5v chip/leds they run them in sets of 3 ie for every 1 led you turn on 3 leds comes on So with the 12? leds on the left 4 would be the correct setting.

Mmm....no. In my 2812 I've set 8. Because it's 1 strip with 8 LEDs. And i can control it individually by index

ellensp, 2021-12-31:
@zonalimitatore you have a standard 5v bar of individually controlled leds not the 12v variety

ellensp, 2021-12-31:
5v
[image]
12v
[image]

From https://electronics.stackexchange.com/questions/427463/5v-vs-12v-addressable-led-efficiency

ellensp, 2021-12-31:
no actual steps to reproduce, 
are you just using M150 gcodes?  
Are you just heating up? (since I see NEOPIXEL_IS_SEQUENTIAL is enabled)
Are you just wishing on a star and it turns green?

ellensp, 2021-12-31:
I have just setup a skr 1.4T with 4 standard 5v neopixels leds running current marlin bugfix 2.0.x

On power on all LEDS flash Red then Green and Blue then turn off.
On sending M150 R30 all 4 leds turn Red
On sending M150 B30 all 4 leds turn Blue
On sending M150 U30 all 4 leds turn Green

My settings are
```
#define NEOPIXEL_TYPE   NEO_GRB
#define NEOPIXEL_PIXELS 4       // Number of LEDs in the strip. (Longest strip when NEOPIXEL2_SEPARATE is disabled.)
#define NEOPIXEL_IS_SEQUENTIAL   // Sequential display for temperature change - LED by LED. Disable to change all LEDs at once.
#define NEOPIXEL_BRIGHTNESS 25  // Initial brightness (0-255)
#define NEOPIXEL_STARTUP_TEST  // Cycle through colors at startup
#define PRINTER_EVENT_LEDS
```

DVSVIDEO, 2022-01-02:
> I see that you have define only 4leds. How many LEDs have?

Yes, I have 4 groups of 3 LEDs, a total of 12 LEDs.
This is not a relevant question because I described at the beginning: With the BTT SKR 1.4 turbo firmware (2.0.6.1) all 4 group LEDs work perfectly. When the display changes to Marlin mode, the segments change in color when printing starts.
The problem is that the LEDs do not work with any of the original marlin firmware. Only the first segment is willing to light up, only in blue.
Physically, nothing changes, a firmware replacement changes it. The configuration files (configuration.h and configuration_adv.h) were the same for both firmware, this could not be the difference.
Logically, the algorithms that control LEDs can be different.

DVSVIDEO, 2022-01-02:
> was looking at these 12V WS2812 Neopixel (having not seen them before), since they are still 5v chip/leds they run them in sets of 3 ie for every 1 led you turn on 3 leds comes on So with the 12? leds on the left 4 would be the correct setting.

See my comment above.

DVSVIDEO, 2022-01-02:
> 2\. https://github.com/MarlinFirmware/Marlin/releases/tag/2.0.6.1

Yes, that's how it works with this firmware: https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/tree/master/BTT%20SKR%20V1.4/Firmware/Marlin-2.0.x -SKR-V1.4-Turbo
But it no longer works properly: https://github.com/MarlinFirmware/Marlin/releases/tag/2.0.6.1

DVSVIDEO, 2022-01-02:
> I have just setup a skr 1.4T with 4 standard 5v neopixels leds running current marlin bugfix 2.0.x
> 
> On power on all 4 LEDS flash Red then Green and Blue then turn off. On sending M150 R30 all 4 LEDS turn Red On sending M150 B30 all 4 LEDS turn Blue On sending M150 U30 all 4 LEDS turn Green
> 
> My settings are
> 
> ```c++
> #define NEOPIXEL_TYPE   NEO_GRB
> #define NEOPIXEL_PIXELS 4       // Number of LEDs in the strip. (Longest strip when NEOPIXEL2_SEPARATE is disabled.)
> #define NEOPIXEL_IS_SEQUENTIAL   // Sequential display for temperature change - LED by LED. Disable to change all LEDs at once.
> #define NEOPIXEL_BRIGHTNESS 25  // Initial brightness (0-255)
> #define NEOPIXEL_STARTUP_TEST  // Cycle through colors at startup
> #define PRINTER_EVENT_LEDS
> ```

Yes, that's how it works with this firmware: https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/tree/master/BTT%20SKR%20V1.4/Firmware/Marlin-2.0.x -SKR-V1.4-Turbo
But it no longer works properly: https://github.com/MarlinFirmware/Marlin/releases/tag/2.0.6.1

DVSVIDEO, 2022-01-02:
> no actual steps to reproduce, are you just using M150 gcodes? Are you just heating up? (since I see NEOPIXEL_IS_SEQUENTIAL is enabled) Are you just powering on and it turns green?

When turned on, the LEDs perform a test: Each color flashes once (this is enabled in the config). After flashing marlin 2.0.6.1, there is no self-test and it cannot be turned on. Only when green is turned on is 1 segment blue.

thinkyhead, 2022-01-02:
Please test the [`bugfix-2.0.x` branch](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to see where it stands. If the problem has been resolved then we can close this issue. If the issue isn't resolved yet, then we should investigate further.

DVSVIDEO, 2022-01-02:
> Please test the [`bugfix-2.0.x` branch](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to see where it stands. If the problem has been resolved then we can close this issue. If the issue isn't resolved yet, then we should investigate further.

All right, I'll do it tomorrow and I'll document it too.

DVSVIDEO, 2022-01-07:
[image]

1. LED strip working fine with BTT SKR Turbo firmware, HERE THE VIDEO:
https://user-images.githubusercontent.com/96373457/148562709-dc44eef7-ce85-46e3-9a19-2ca80a851f88.mp4
[image]

2. LEDs not working properly with original latest marlin bugfix, HERE THE VIDEO:
https://user-images.githubusercontent.com/96373457/148562732-d2cee9a9-36f8-460c-9007-f2fb0323ad3e.mp4
[image]

DVSVIDEO, 2022-01-07:
> Please test the [`bugfix-2.0.x` branch](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to see where it stands. If the problem has been resolved then we can close this issue. If the issue isn't resolved yet, then we should investigate further.

I finally had time. I also took pictures and videos.
Unfortunately, the error still exists.

thinkyhead, 2022-04-21:
There have been a number of changes in recent weeks, especially related to timers, catching pin conflicts, etc. Please test the [`bugfix-2.0.x` branch](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to see where it stands. If the problem has been resolved then we can close this issue. If the issue isn't resolved yet, then we should investigate further.

closed 2022-07-01 by github-actions[bot]

DVSVIDEO, 2022-07-03:
Today I tried the new 2.1 marlin and finally my NEOPIXEL LED strip works well! 
Somebody somewhere did something and now it is correct. Thanks to the developers!