MarlinFirmware/Marlin #23318: Dual_X_Carriage: Change in the Menu "Move" not available

mawi1986, 2021-12-19:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

I have a printer that has Marlin version 2.0.5.3 installed by default. The printer has two extruder heads that can be moved independently of each other, i.e. Dual_X_Carriage.
The board is an MKS RobinPro and the display is a RobinTFT35 which runs over LVGL.

Now I wanted to install a BLTouch sensor and have adapted the Marlin version 2.0.7.1 so that this version runs on two other printers from the same manufacturer without any problems. Unfortunately, neither with Marlin 2.0.7.1 nor with the current Nightly version from December 18, 2021, the menu for the distance between the two X Carriages is displayed, nor can I switch between the X1 and X2 in the Move menu. I have these two options with the originally installed marlin 2.0.5.3.

Can anyone help me? I think it's because of the LVGL or the menu of it.

### Version of Marlin Firmware

2.x from 18.12.2021

### Electronics

MKS RobinPro with TFT35

### Additional information & file uploads

[image]
[image]
[image]

ellensp, 2021-12-19:
Which tft UI is that? TFT_LVGL_UI I would guess
Please provide bugfix config files

mawi1986, 2021-12-19:
Yes, its TFT_LVGL_U

My Config-Files are here...
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/7740701/Marlin.zip)
I