MarlinFirmware/Marlin #23320: [FR] (feature summary)

Mordi1984, 2021-12-19:
### Is your feature request related to a problem? Please describe.

Hi. I nee an option to calibrate the expected 
 temperature lenght coeffizient.
Because if the bed is cold my probe (induction) feels the bed is about 0.65mm away. But because the nozzle to bed temperature is cold it is absolut 1,1mm away.

### Are you looking for hardware support?

ANET AM8 with Stock Board.
V6 Bolzano Nozzle.
6mm Aluminium Plate

### Describe the feature you want

I nees to calibrate it for my Material (ASA) to 120 Bed and 250°C Nozzle temp. But if i print PLA or an other Material, my temperature nee to be lower and the Nozzle is now 0.8mm away.
Is there an compensation table for nozzle heat cold to hot? 
I know Chance the nozzle material also Results in lenght Change. 

Maybe you can add an point to the console in LCD Screen. 

Maybe it is there and i did not seen it, or it nee to get programmed to become more flexibel.

Many prints bekomme bad because of that missi g Feature

ManuelMcLure, 2021-12-19:
This is something that is better done in the slicer. Because I want to print PETG at around 0.08mm higher than PLA, I adjusted my Z offset for PLA and then have the following in my slicer gcode settings for PETG:
```
G28
G0 Z2
Z92 Z1.92
```
This makes Marlin think the printhead is 0.08mm lower than it really is, so the print will be 0.08mm higher than the Z offset.

thinkyhead, 2022-01-05:
Probe Temperature Compensation is set using the `PTC_PROBE`, `PTC_BED`, and `PTC_HOTEND` options…
```cpp
/**
 * Thermal Probe Compensation
 *
 * Adjust probe measurements to compensate for distortion associated with the temperature
 * of the probe, bed, and/or hotend.
 * Use G76 to automatically calibrate this feature for probe and bed temperatures.
 * (Extruder temperature/offset values must be calibrated manually.)
 * Use M871 to set temperature/offset values manually.
 * For more details see https://marlinfw.org/docs/features/probe_temp_compensation.html
 */
//#define PTC_PROBE    // Compensate based on probe temperature
//#define PTC_BED      // Compensate based on bed temperature
//#define PTC_HOTEND   // Compensate based on hotend temperature
```

closed 2022-01-05 by thinkyhead