MarlinFirmware/Marlin #23324: [BUG] LCD 12864 stops working when SD card is plugged in

Klabbob, 2021-12-20:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

I got an MKS Monster8 for my CR-20 Pro and are trying to set it up, but running into a problem with the original 12864 display from Creality without SD card reader. I can get it running fine, but as soon as you plug in the micro SD into the onboard reader of the board, the screen becomes unresponsive. The reason for this seems to be that the SD is hijacking the SPI Bus. I have attached the Configuration.h, the Configuration_adv.h and pins_MKS_MONSTER8.h

### Bug Timeline

20.12.2021

### Expected behavior

LCD and SD card can be used at the same time.

### Actual behavior

Only LCD or SD card can be used at a time. Therefore printing from the SD card can not be done via the display

### Steps to Reproduce

The LCD is wired to EXP1 and EXP2 of the Monster8 and pins PA5 (SPI1 SCK), PA4 (SPI1 CS), PA7 (SPI1 MOSI) are used to communicate. This should be SPI1 from what I understand. The onboard SD card reader is connected with PC10-12 which is SPI3.

I am running `#define MKS_MINI_12864` in Configuration.h and have changed the following in pins_MKS_MONSTER8.h:
` #if ENABLED(MKS_MINI_12864)
    //#define LCD_BACKLIGHT_PIN             -1
    //#define LCD_RESET_PIN                 -1
    #define DOGLCD_A0                EXP1_04_PIN   // PE15
    #define DOGLCD_CS                EXP1_05_PIN   // PD8
    #define DOGLCD_SCK              EXP2_09_PIN   // PA5
    #define DOGLCD_MOSI            EXP2_05_PIN  // PA7`

Running M43 returns the following when **no** SD card is inserted:
`PIN: PC10        M42 P42          <unused/unknown>                       Input  = 1
PIN: PC11        M42 P43          <unused/unknown>                       Input  = 1
PIN: PC12        M42 P44          <unused/unknown>                       Input  = 1
PIN: PD8         M42 P56          DOGLCD_CS                              Output = 1
PIN: PE15        M42 P79          DOGLCD_A0                              Output = 1`

Running M43 with the SD card inserted changes this to:
`PIN: PC10        M42 P42          <unused/unknown>                       Alt Function: 6  - SPI3
PIN: PC11        M42 P43          <unused/unknown>                       Alt Function: 6  - SPI3
PIN: PC12        M42 P44          <unused/unknown>                       Alt Function: 6  - SPI3
PIN: PD8         M42 P56          DOGLCD_CS                              Output = 1
PIN: PE15        M42 P79          DOGLCD_A0                              Output = 1`

The SD card can be accessed via Pronterface and is therefore working correctly. While experimenting I managed to get PC10 and PC12 to be returned as DOGLCD_SCK and DOGLCD_MOSI, even though PA5 and PA7 should be having this function (and seemingly do as the display is connected to them and works).

To me it seems like the LCD and the SD card are running on the same SPI and the SD card gains priority if inserted. This seems unclear to me as they are connected physically different pins that correspond to SPI1 and SPI3 respectively (which should be independent, right?). https://www.st.com/resource/en/datasheet/dm00037051.pdf page 50.

M43 indicates to me that DOGLCD_SCK and DOGLCD_MOSI are overwritten during compilation.

I am not sure why this is happening, maybe it has something to do with https://github.com/MarlinFirmware/Marlin/issues/10400 or https://github.com/MarlinFirmware/Marlin/issues/8358.

[Config.zip](https://github.com/MarlinFirmware/Marlin/files/7747304/Config.zip)

### Version of Marlin Firmware

Marlin bugfix-2.0.x (Dec 20 2021 19:03:20) 

### Printer model

Creality CR-20 Pro

### Electronics

MKS Monster8

thinkyhead, 2021-12-26:
It seems you're a pioneer in making the `MKS_MINI_12864` work with the Monster 8 board. Hopefully it shouldn't be too hard. 

The LCD and the SD should be okay with sharing the same SPI bus, where CS in one state selects the LCD and the other state selects the SD card. That's how the simpler display plus SD card devices work, anyway.

See if this works any better. This is my best guess…
```cpp
    #define DOGLCD_A0                EXP2_10_PIN
    #define DOGLCD_CS                EXP2_07_PIN
    #define DOGLCD_SCK               EXP2_09_PIN
    #define DOGLCD_MOSI              EXP2_05_PIN
```

Klabbob, 2021-12-26:
Hey, Thank you for your input. I just wired it up and flashed Marlin with the updated definitions, but now the display no longer displays anything. 

M43 also does not return any definition of DOGLCD_A0 or DOGLCD_CS anymore. The pins PA6 and PA4 are not existent in that list. Does this mean that they did not get defined?

ellensp, 2021-12-26:
@Klabbob there is a known but in pins-debugging that if the pin can be analaog, it doesn't get listed at all

Klabbob, 2021-12-26:
@ellensp thanks for that information, good to know.

The wiring from @thinkyhead did not work at all. I also tried the following configurations:

```
#define DOGLCD_A0                PD11
#define DOGLCD_CS                EXP1_04_PIN
```

This is the original configuration in the unaltered pin file in marlin. It works without the SD card plugged, but PD11 is not on the EX1 and EX2 headers.

```
#define DOGLCD_A0                EXP1_07_PIN
#define DOGLCD_CS                EXP1_08_PIN
```

This also works without the SD card plugged in. I tested this as it is the default config for the FYSETC_MINI_12864_2_1 screen.

```
#define DOGLCD_A0                EXP1_05_PIN
#define DOGLCD_CS                EXP1_04_PIN
```

This also works without the SD... this is close to the original configuration with just EXP1_05_PIN as A0 so that it is on the header... 

I suspect that my problem is not directly pin related as all these configurations work the same and just stop working once the micro SD is plugged in. I suspect that the CS line is not properly set, no matter which pin it is assigned to.

Maybe another question if someone more knowledgeable is around: The screen is wired to pins corresponding to SPI1 and the SD card to pins of SPI3, shouldn't these buses be independent? Or is this maybe something that would need to be configured?

descipher, 2021-12-26:
Ummm, not going to work on EXP1 ...

EXP2 has the pins you need 
#define EXP2_07_PIN                         PA4
#define EXP2_08_PIN                         PE9
#define EXP2_09_PIN                         PA5
#define EXP2_10_PIN                         PA6

However be careful the pin 1 indicator seems to be incorrect .. 

[image]

descipher, 2021-12-26:
Maybe try it this way.
``` 
    #define DOGLCD_CS                EXP1_04_PIN
    #define DOGLCD_A0                EXP1_05_PIN
//
// SPI / SD Card
//
#define SD_SCK_PIN                    EXP2_09_PIN
#define SD_MISO_PIN                  EXP2_10_PIN
#define SD_MOSI_PIN                  EXP2_05_PIN
#define SDSS                                EXP2_07_PIN
#define SD_DETECT_PIN               EXP2_04_PIN
```

Klabbob, 2021-12-26:
@descipher I am already using the following pins as mentioned in the original post:

```cpp
    #define DOGLCD_A0                EXP1_04_PIN   // PE15
    #define DOGLCD_CS                EXP1_05_PIN   // PD8
    #define DOGLCD_SCK              EXP2_09_PIN   // PA5
    #define DOGLCD_MOSI           EXP2_05_PIN  // PA7
```

The SD card that I am using is the onboard reader of the Monster8 Mainboard and can therefore NOT be changed:

```cpp
#if SD_CONNECTION_IS(ONBOARD)
  #define ENABLE_SPI3
  #define SD_SS_PIN                         -1
  #define SDSS                              PC9
  #define SD_SCK_PIN                        PC10
  #define SD_MISO_PIN                       PC11
  #define SD_MOSI_PIN                       PC12
  #define SD_DETECT_PIN                     PC4   // SD_DETECT_PIN doesn't work with NO_SD_HOST_DRIVE disabled
```

Am I missing something in your suggestion?

descipher, 2021-12-26:
Well which one do you want? ... either the ONBOARD or LCD like ever other board does :/ If you want the LCD one then configure it. If not de-configure it because it will likely hang the board.

descipher, 2021-12-27:
FYI 
1607 configuration_adv.h
```
  /**
   * Set this option to one of the following (or the board's defaults apply):
   *
   *           LCD - Use the SD drive in the external LCD controller.
   *       ONBOARD - Use the SD drive on the control board.
   *  CUSTOM_CABLE - Use a custom cable to access the SD (as defined in a pins file).
   *
   * :[ 'LCD', 'ONBOARD', 'CUSTOM_CABLE' ]
   */
  //#define SDCARD_CONNECTION LCD

  // Enable if SD detect is rendered useless (e.g., by using an SD extender)
  //#define NO_SD_DETECT
```

Klabbob, 2021-12-27:
@descipher I am not sure if you even read my original post. Because again you are talking about something that has nothing to do with the problem that should be solved here. I clearly stated that I have an original Creality screen _**WITHOUT**_ an SD reader. Therefore I want to use the reader of the Monster8, as it is my only option. 

Maybe a small summary so that we are talking about the same things here: I have a a Creality 12864 screen **without** an SD card reader and an EXP1 and an EXP2 headers. I have it connected to the EXP1 and EXP2 headers of a MKS Monster8, which has an onboard micro SD card reader. I got the screen working with the pins stated in my last post, **as long as no SD card is plugged into the board**. The screen freezes as soon as the micro SD card is plugged in, but the card is working as it should, as I confirmed via Pronterface. I have also done some prints from it in the meantime. So the problem that should be solved here, is that I want to have both the screen and the SD card working at the same time, so that I can use the screen to start a print.

Klabbob, 2021-12-27:
I found a working solution! So in the end it is not a problem of the pins that are assigned, but a configuration issue. Maybe someone of the **_Devs can change this in the pins_MKS_MONSTER8.h_** so that it works out of the box in future releases.

```cpp
  // MKS MINI12864 and MKS LCD12864B; If using MKS LCD12864A (Need to remove RPK2 resistor)
  #if ENABLED(MKS_MINI_12864)
    #define ENABLE_SPI1
    #define FORCE_SOFT_SPI
    //#define LCD_BACKLIGHT_PIN             -1
    //#define LCD_RESET_PIN                 -1
    #define DOGLCD_A0              EXP1_04_PIN
    #define DOGLCD_CS              EXP1_05_PIN 
    #define DOGLCD_SCK            EXP2_09_PIN
    #define DOGLCD_MOSI          EXP2_05_PIN
```

The key here is to insert the `#define ENABLE_SPI1` and the `#define FORCE_SOFT_SPI`. I found out that the onboard SD card reader has a `#define ENABLE_SPI3`. This is why I tried to enable the SPI1 bus here, as this is the one that is connected to pins of the display, according to the datasheet of the STM32F4. I now suspect that the SPI3 was somehow used to communicate with the display, as it was the only one enabled, but was then reassigned to the SD card as soon as it was plugged in. Don't quote me on that as I am not sure. But anyway it is working with this configuration.

The `#define DOGLCD_A0` and the `#define DOGLCD_CS` has been altered from the original configuration as well so that the EXP1 and EXP2 ports of the display and the board can be routed through 1 to 1. So _**no changes in the pin order is necessary**_, pin 1 on EXP1 on the display side is also pin 1 on EXP1 on the board side and so forth. 

For those looking for a pinout of the 12864 Creality screen with two expansion headers:

```cpp
/**                      -------                                                           -------
 *   VCC              | 1   2 |   GND                      NC                     | 1   2 |   GND
 *   NC                | 3   4 |   DOGLCD_A0         KILL_PIN             | 3   4 |   SD_DET
 *   DOGLCD_CS   5   6 |   NC                        DOGLCD_MOSI    5   6 |   BTN_EN1
 *   NC                | 7   8 |   NC                        SDSS                   | 7   8 |   BTN_EN2
 *   BTN_ENC      | 9 10 |   Beeper                  DOG_LCD_SCK   | 9  10|   DOGLCD_MISO
 *                        -------                                                            -------
```

I hope this is useful for those that are running in to the same problem and maybe this can be added to Marlin in the definitions of this board as I suspect that it will be a problem with all displays that are hooked up as the SPI1 is not enabled correctly.

descipher, 2021-12-27:
> CR-20 Pro
Oh, sorry, my bad, I did skim the details to fast. Seems you have found the real issue in that there are configuration defines that set config elements which are more bard specific. Now with the explosion of possible CPU's that will have to be gated into the board pin config logic . e.g. use SPI1 or SPI3 or even use soft spi etc.

thisiskeithb, 2021-12-28:
Fixed in https://github.com/MarlinFirmware/Marlin/commit/c160a57397bbd31c44c404f7c56a269f8fd802d6

commit c160a573 (2021-12-27): 📌 Define MKS Monster8 pins for MKS_MINI_12864

Fixes #23324
commit 4cfe812c (2022-01-09): 📌 Define MKS Monster8 pins for MKS_MINI_12864

Fixes #23324
closed 2021-12-28 by thisiskeithb