MarlinFirmware/Marlin #23325: Backlash compensation on independent dual-X-axis printer creates increasing X-offset each layer

Zabex1, 2021-12-20:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

I want to use backlash compensation on a dual-X-axis printer. With backlash compensation set to 0 it works fine. For values greater than zero, each layer starts a bit more towards the home position of the axis. It seems as if the backlash value is subtracted at every "tool-change" (moving X-axis 1 to home position, move X-axis2 to print position).

### Bug Timeline

Old one. Found it already in 2.0.5.3, but ignored it until I was sure it is a Marlin issue.

### Expected behavior

Switching on the backlash compensation with the right value should produce a vertical wall where a vertical wall is designed.

### Actual behavior

Switching on the backlash compensation with a value > 0 produce a sloping wall where a vertical wall is designed.
Here a photo of two waste towers of the same object. Left one without backlash compensation, right one with compensation.
[image]
.

### Steps to Reproduce

1. Create a Configuration with dual X axis and backlash compensation enabled.
2.put M425 X0.08 Y0.05 Z0.01 and M425 F1 S0 in Start-GCode (Set backlash to specific values for all axis and switch it on).
3. Print a two color object with i.e. yellow in left and blue in right extruder
4. The result is a V-shaped waste tower, where blue moves to the right and yellow to the left. 

### Version of Marlin Firmware

2.0.9.2

### Printer model

ZIDEX  (Zabex own IDEX;  constructed by myself)

### Electronics

Bigtree SKR pro V1.1

### Add-ons

none

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Cura

### Host Software

Repetier Host

### Additional information & file uploads

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/7747374/Configuration.zip)
Log files shows no difference between good and bad objects.
Here the Startup-GCode that is executed before the object is printed. In this version 
[StartupGCode.zip](https://github.com/MarlinFirmware/Marlin/files/7747384/StartupGCode.zip)
This code produces good objects. Please remove Line 44 (the last line) to see the erroneous result.

thinkyhead, 2021-12-26:
The backlash code is currently incompatible with IDEX and implementing this support will require some study.

CC: @marciot

jaeryang-sindoh, 2022-01-27:
I think basklash make problems when print several output that separate from each other.

tombrazier, 2022-02-27:
This might be fixed by #23814. Not certain of this but the toolchange code does call sync_plan_position() which is fixed in that PR. So worth testing again.

jaeryang-sindoh, 2022-03-04:
Layer shift still exist when IDEX. 
@Zabex1 Could you try this fix and check layer shift ?

thisiskeithb, 2022-03-09:
Now that #23814 is merged, please download the latest [`bugfix-2.0.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) & report your findings.

Zabex1, 2022-04-03:
Tested using bugfix-2.0.x from 2022-04-03. 
Unfortunately: Layer shift still exist with active backlash compensation on IDEX machine.

xsuper9988, 2022-12-14:
> Now that #23814 is merged, please download the latest [`bugfix-2.0.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) & report your findings.

I have a simple test way
I don't know if this helps with source code modification

My Marlin Ver. "Marlin-2.1.1"

Annotation of pictures
1. X1 X2 overlap hole
2. X1 shift hole
3. X2 shift hole
When in IDEX mode
I turned backlash on
Then the original overlap hole will shift to the left and right after Tool Changed
Operation Process
1. X1 moves to Z0.0
2. Tool Change
3. After Tool Change, X2 is at Z0.0, and the holes overlap at this time
4. Move Z to Z10.0 position, and run 10 times Tool Change
5. Then move X1 and X2 to the position of Z0.0
6. Then X1、X2 hole is Shift
[image]

EYHJ59, 2022-12-14:
My suggestion is... when nozzle change, last direction bits has to be changed forcibly.

void Backlash::add_correction_steps(const int32_t &da, const int32_t &db, const int32_t &dc, const uint8_t dm, block_t * const block) {

  static uint8_t last_direction_bits;
  if(backlash_extruder[0] != active_extruder) {
    backlash_extruder[0] = active_extruder;
    if(active_extruder == 0) last_direction_bits |= 0x01;
    else last_direction_bits &= 0xFE;
  }
  uint8_t changed_dir = last_direction_bits ^ dm;
  ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

And if X1 motor and X2 motor are separated, each motor has different distance_mm value.(but maybe 2 value is almost same)