MarlinFirmware/Marlin #23328: [BUG] G34, Z_STEPPER_ALIGN_XY not updating properly through flashing firmware

Luukver, 2021-12-21:
### Did you test the latest `bugfix-2.0.x` code?

No, but I will test it now!

### Bug Description

The initial setup of the G34 probing coordinates worked fine (Z_STEPPER_ALIGN_XY), except one was a little too close to the edge so I changed it. After re-flashing the firmware on the BTT GTR the coordinates hadn't changed like I expected. instead the initial ones remained causing a lot of confusion as it seemed like Marlin was ignoring the configured values.

In combination with the probing margin configuration for mesh leveling this caused the G34 process to throw a probing error before moving to the changed position because the old position is now out of bounds. But because the new values are still stored somewhere the out of bounds error doesn't show causing further confusion.

What I've tried:

- I've tried saving to EEPROM in case anything wasn't updated yet, didn't work.
- I've tried adjusting Z_STEPPER_ALIGN_KNOWN_STEPPER_POSITIONS positions in case it was using that in any way even though it should be disabled. didn't work
- I've tried removing the Probe margin configuration. didn't work

I did find a workaround:

- Manually updating the probe coordinate in question using M422 S3 X<> Y<> seems to actually apply the updated value properly. Saving this to EEPROM also works.

### Bug Timeline

New bug

### Expected behavior

I expected the probing coordinates of the third axis to change according to the ones configured in Z_STEPPER_ALIGN_XY

### Actual behavior

The old initial values remained regardless of whatever was configured in Configuration_adv.h

### Steps to Reproduce

1. Setup triple z-axis alignment with G34
2. Configure the Z_STEPPER_ALIGN_XY values accordingly
3. Flash the firmware to the board
4. Change a coordinate (in this case the third axis)
5. Re-Flash the firmware to the board
6. Actual behavior of G34 hasn't changed compared to initial setup

### Version of Marlin Firmware

2.0.7.2

### Printer model

Custom

### Electronics

BTT GTR V1.0 + BTT TFT35 V3.0

### Bed Leveling

ABL Bilinear mesh

### Additional information & file uploads

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/7755107/Marlin.zip)

thisiskeithb, 2021-12-21:
> Did you test the latest bugfix-2.0.x code?
> No, but I will test it now!

> Version of Marlin Firmware
> 2.0.7.2

Please download [`bugfix-2.0.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to test with the latest code and let us know if you're still having this issue.

ManuelMcLure, 2021-12-21:
I believe the probe positions are saved in EEPROM which means you need to reinitialize the EEPROM after updating the probe positions in the config. Try `M502` followed by `M500` after updating the firmware and see if that solves the issue.

Luukver, 2021-12-21:
> Please download [`bugfix-2.0.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to test with the latest code and let us know if you're still having this issue.

I will try the bug-fix version as soon as I can, updating the config files is somewhat tedious (any better way to do this?).

> I believe the probe positions are saved in EEPROM which means you need to reinitialize the EEPROM after updating the probe positions in the config. Try `M502` followed by `M500` after updating the firmware and see if that solves the issue.

I just tried this and you're correct! M502 followed by M500 does update the values properly.
Still this shouldn't be a required step however.

thisiskeithb, 2021-12-21:
> Still this shouldn't be a required step however.

It is when changing values stored in EEPROM.

closed 2021-12-21 by thisiskeithb