MarlinFirmware/Marlin #23329: [FR] Adjust HOTEND_IDLE_MIN_TRIGGER at runtime

qwertymodo, 2021-12-21:
### Is your feature request related to a problem? Please describe.

Marlin's hotend idle timout feature uses a hard-coded temperature threshold that doesn't seem to be adjustable after the firmware has been compiled.  Unfortunately, this means that if you set that temperature to something low and reasonable for a filament like PLA, but want to preheat something like ABS which can be preheated at much higher temperatures without charring, it's really easy to end up triggering the hot end timeout because the preheat temperature is above the timeout trigger temperature.  If you set the trigger temperature higher for ABS, then the feature stops working properly for PLA, because now it won't trigger at PLA charring temperatures.

### Are you looking for hardware support?

N/A

### Describe the feature you want

It would be nice if there was some way, either via gcode or through the LCD interface, to adjust the value of HOTEND_IDLE_MIN_TRIGGER at runtime.  It would also be nice if that change was done automatically when selecting one of the preheat options from the LCD menu so that the preheat functions don't trigger an idle timeout (i.e. when you select "Preheat PLA" it would set the hotend temperature to the configured preheat temperature and also adjust HOTEND_IDLE_MIN_TRIGGER to the same temperature).

slowbro, 2022-11-25:
I implemented the first part of your request: an LCD Menu (under Configuration) to modify the HOTEND_IDLE_* variables at runtime. PR submitted.

thisiskeithb, 2022-11-25:
Added in https://github.com/MarlinFirmware/Marlin/pull/25015

closed 2022-11-25 by thisiskeithb