MarlinFirmware/Marlin #23332: [BUG] Enabling CASE_LIGHT_ENABLE (M355) causes build error

rcdaley, 2021-12-22:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

With RGB_LED enabled and functioning, I tried enabling CASE_LIGHT_ENABLE to use the RGB LEDs as a case light.  When CASE_LIGHT_ENABLE is defined, I get the following build error:

Marlin\src\feature\caselight.cpp: In static member function 'static void CaseLight::update(bool)':
Marlin\src\feature\caselight.cpp:68:87: error: 'n10ct' was not declared in this scope
   68 |     leds.set_color(LEDColor(color.r, color.g, color.b OPTARG(HAS_WHITE_LED, color.w), n10ct));
      |                                                                                       ^~~~~
*** [.pio\build\STM32F401VE_STEVAL\src\src\feature\caselight.cpp.o] Error 1

[image]

if I replace line 68 in caselight.ccp,

    leds.set_color(LEDColor(color.r, color.g, color.b OPTARG(HAS_WHITE_LED, color.w), n10ct));

with,

leds.set_color(LEDColor(color.r, color.g, color.b));

everything builds and works correctly.

### Bug Timeline

It started when I defined CASE_LIGHT_ENABLE

### Expected behavior

I expected case lights to work.

### Actual behavior

Build Error

### Steps to Reproduce

define CASE_LIGHT_ENABLE and RGB_LED
build project

### Version of Marlin Firmware

2.0.9.2

### Printer model

Custom/STMEVAL-3DP001

### Electronics

STMEVAL-3DP001

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Cura

### Host Software

OctoPrint

### Additional information & file uploads

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/7760095/Configuration.zip)

ellensp, 2021-12-22:
User did not test bugfix.
This has already been fixed in bugfix https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/Marlin/src/feature/caselight.cpp#L67 
```CPP
  #if CASE_LIGHT_IS_COLOR_LED
    leds.set_color(LEDColor(color.r, color.g, color.b OPTARG(HAS_WHITE_LED, color.w) OPTARG(NEOPIXEL_LED, n10ct)));
  #else // !CASE_LIGHT_IS_COLOR_LED
```
vs the buggy older version https://github.com/MarlinFirmware/Marlin/blob/2.0.x/Marlin/src/feature/caselight.cpp#L68
```CPP
  #if CASE_LIGHT_IS_COLOR_LED
    leds.set_color(LEDColor(color.r, color.g, color.b OPTARG(HAS_WHITE_LED, color.w), n10ct));
  #else // !CASE_LIGHT_IS_COLOR_LED

```

closed 2021-12-22 by ellensp

rcdaley, 2021-12-23:
Sorry, thought I tested bugfix.