MarlinFirmware/Marlin #23342: undefined reference to `mmu2_M600()

Claguee, 2021-12-23:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Using 2.0.9.2 Bug fix, config wont compile with the following error 

C:\Users\clagu\AppData\Local\Temp\ccpsM57X.ltrans1.ltrans.o: In function `pause_print(float const&, XYZval<float> const&, bool, float const&) [clone .part.10]':
<artificial>:(.text+0x5546): undefined reference to `mmu2_M600()'
C:\Users\clagu\AppData\Local\Temp\ccpsM57X.ltrans1.ltrans.o: In function `GcodeSuite::process_parsed_command(bool)':
<artificial>:(.text+0x8154): undefined reference to `mmu2_choose_filament()'
<artificial>:(.text+0x818a): undefined reference to `mmu2_choose_filament()'
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\mega2560\firmware.elf] Error 1

[Configs.zip](https://github.com/MarlinFirmware/Marlin/files/7770360/Configs.zip)

### Version of Marlin Firmware

2.0.9.2

### Printer model

Anycubic Chiron

### Electronics

Stock

### Add-ons

MMU

ellensp, 2021-12-23:
the display type ANYCUBIC_LCD_CHIRON does not support having MMU2_MENUS

Claguee, 2021-12-23:
I have a sk4 1.4T with a tft70, is this something that would be compatible?

thisiskeithb, 2021-12-23:
> I have a sk4 1.4T with a tft70, is this something that would be compatible?

Yes.

closed 2021-12-23 by Claguee