MarlinFirmware/Marlin #23348: [BUG] (bug summary)

mhkjoon, 2021-12-25:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Hi
The problem I have is that when printing layers are printed in the direction of the x-axis forward, but when I change the direction of the shape, this problem does not exist.
That's why I have to put the object in a corner of the table so that it does not reach the middle of the table. I also tested it with different software, but it gives the same answer. I put the photos below. Thanks

[image]
[image]
[image]

### Version of Marlin Firmware

2

### Your Slicer

Simplify3D

ellensp, 2021-12-25:
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- 3D Printing Discord server. Join link: https://discord.gg/qNnSzsug
- Marlin specific Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2021-12-25 by ellensp

mhkjoon, 2021-12-25:
ex bro i think its marlin bug.

thinkyhead, 2022-04-10:
> ex bro i think its marlin bug.

We don't know that yet, so you'll need to work with someone in the recommended forums and do troubleshooting and try to narrow it down first. We prefer to limit this issue queue only to confirmed and actionable bugs, preferably with a good collection of data. Maybe you just have a slipping belt.