MarlinFirmware/Marlin #23352: [BUG] 2.0.9.3 not resulting in working firmware.bin on SKR2.0b  F407 ?

Kaween-prog, 2021-12-25:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

SKR2b with F407.  Previous builds (2.0.9.2 and bugfix version up to 2 about 2 weeks ago) created working firmware.bin files.

Now, with 2.0.9.3, the firmware.bin file will build without errors (using either my own configuration.h and configuration_adv.h files, but also with the default configuration files in place), the resulting firmware.bin will attempt to flash in the SKR2.0b (file gets renamed to firmware.cue, so the firmware was recognized as such and the flash did function), but the board does not operate at all after the flash (no display, no way of communicating over serial, nothing).

I have the odd impression 2.0.9.3 is creating an F429 firmware.bin although "it's not asked to do so".  
Board and enviroment definition are set as : 

configuration.h =  #define MOTHERBOARD BOARD_BTT_SKR_V2_0_REV_B 
platform.ini = default_envs = BIGTREE_SKR_2

Same issue with the latest bugfix.
Problem exists even with the default configuration.h / configuration_adv.h .

### Bug Timeline

Only noticed it now with the launch of 2093. Worked correctly up to bugfix 8/12/21 beginning this month

### Expected behavior

correct functionality of board after firmware.bin file is flashed

### Actual behavior

board flashes firmware and renames it to firmware.cue, but the board itself won't react after that.  Reflashing it with the 8/12/21-based bugfix returns the board back to functionality.

Tested 2 different SKR2.0B F407's.  Do not have a F429 available, so I'm not sure the resulting firmware.bin is "indeed F429".

### Steps to Reproduce

download current 2.0.9.3 (or newest bugfix)
use the default configuration.h file, set the board define to SKR2.0b and compile (no errors nor warnings during compilation)
flash SKR2.0b board with the resulted firmware.bin.  After flash, the  SKR2.0b board will not react until it gets flashed again with a previous version of Marlin (as said, work up to 8/12/21, maybe later too, haven't checked that yet)

### Version of Marlin Firmware

2.0.9.3 

### Printer model

makes no difference, issue exists with the default configuration.h too

### Electronics

stock electronics

### Add-ons

BTT TFT 3.5 V3, but problem also exists without this LCD connected

Graylag-PD, 2021-12-25:
Same behaviour on SKR PRO, also F407 board. But as far as I know, SKR PRO only ever existed as F407, so issue probably goes deeper

Kaween-prog, 2021-12-25:
Thanks for confirming this Graylag-PD

Kaween-prog, 2021-12-25:
@Graylag-PD  Thank you for confirming this ! I was thinking I was seeing ghosts by now.
But yes, if the SKR PRO is showing the same behavior, one must assume something goes wrong with the F407 build dependencies themselves.

thisiskeithb, 2021-12-25:
My BTT002 and SKR 2 builds no longer work with 2.0.9.3. There was also reports of a RAMPS board not working on Discord.

I haven’t tested LPC yet.

Reverting [♻️ Refactor HAL as singleton (#23295)](https://github.com/MarlinFirmware/Marlin/commit/e211ff148c39bf5dace72de7cffbb83f19d3f1bf) allows them to boot again.

ellensp, 2021-12-26:
RAMP is confirmed non working.
Serial Boot log on RAMP

```
> start
>  External Reset
> Marlin 2.0.9.2
> echo: Last Updated: 2021-09-03 | Author: (none, default config)
> echo: Compiled: Dec 26 2021
> echo:

```
it just stops there

mcottrell, 2021-12-26:
This also occurs on my SKR MINI E3 V1.2

ellensp, 2021-12-26:
For RAMPS
If you change Marlin/src/MarlinCore.cpp line 1263 from 
```CPP
  SERIAL_ECHO_MSG(STR_FREE_MEMORY, hal.freeMemory(), STR_PLANNER_BUFFER_BYTES, sizeof(block_t) * (BLOCK_BUFFER_SIZE));
```
back to 
```CPP
  SERIAL_ECHO_MSG(STR_FREE_MEMORY, freeMemory(), STR_PLANNER_BUFFER_BYTES, sizeof(block_t) * (BLOCK_BUFFER_SIZE));
```
It boots
So an issue with hal.freeMemory() perhaps.

ellensp, 2021-12-26:
In Marlin/Marlin/src/HAL/AVR/HAL.h is
```CPP
  // Free SRAM
  static inline int freeMemory() { return freeMemory(); }
```
This seems to cause an infinite loop
I changed it to
```CPP
  // Free SRAM
  static inline int halfreeMemory() { return freeMemory(); }

```
And updated Marlin/src/MarlinCore.cpp to 
```CPP
  SERIAL_ECHO_MSG(STR_FREE_MEMORY, hal.halfreeMemory(), STR_PLANNER_BUFFER_BYTES, sizeof(block_t) * (BLOCK_BUFFER_SIZE));
```
And this works as expected

pilot76103, 2021-12-26:
Same issue here on my mega2560 for Longer LK5 Pro. Sits on main boot screen and does not allow serial communication over USB.

ellensp, 2021-12-26:
Same issue on skr1.4

In Marlin/src/HAL/LPC1768/HAL.h is
```CPP
  // Free SRAM
  static inline int freeMemory() { return freeMemory(); }
```
This seems to cause an infinite loop
I changed it to
```CPP
  // Free SRAM
  static inline int halfreeMemory() { return freeMemory(); }

```
And updated Marlin/src/MarlinCore.cpp to 
```CPP
  SERIAL_ECHO_MSG(STR_FREE_MEMORY, hal.halfreeMemory(), STR_PLANNER_BUFFER_BYTES, sizeof(block_t) * (BLOCK_BUFFER_SIZE));
```
And this works as expected

pilot76103, 2021-12-26:
> In Marlin/Marlin/src/HAL/AVR/HAL.h is
> 
> ```c++
>   // Free SRAM
>   static inline int freeMemory() { return freeMemory(); }
> ```
> 
> This seems to cause an infinite loop I changed it to
> 
> ```c++
>   // Free SRAM
>   static inline int halfreeMemory() { return freeMemory(); }
> ```
> 
> And updated Marlin/src/MarlinCore.cpp to
> 
> ```c++
>   SERIAL_ECHO_MSG(STR_FREE_MEMORY, hal.halfreeMemory(), STR_PLANNER_BUFFER_BYTES, sizeof(block_t) * (BLOCK_BUFFER_SIZE));
> ```
> 
> And this works as expected

Made these changes for my mega2560 on my Longer LK5 Pro and while the serial and USB interfaces work, it reset all EEPROM settings and the bed and thermistor read -15ºC.

thinkyhead, 2021-12-26:
Sorry about that! I just pushed a patch to correct the `freeMemory` loop issue. Keep me posted if you see any other weird things!

thisiskeithb, 2021-12-26:
> Sorry about that! I just pushed a patch to correct the `freeMemory` loop issue. Keep me posted if you see any other weird things!

CI failed: https://github.com/MarlinFirmware/Marlin/actions/runs/1623041383

2.0.9.3 will need to be retagged once this is fixed.

pilot76103, 2021-12-26:
Issue is not fully resolved and may need it's own issue opened. After the freeMemory loop issue was resolved both my bed and thermistor temp show -15ºC on power up.

thinkyhead, 2021-12-26:
@pilot76103 — What is your `MOTHERBOARD` ?

pilot76103, 2021-12-26:
> @pilot76103 — What is your `MOTHERBOARD` ?
> 
> Longer LK5 … ?

#define MOTHERBOARD BOARD_LONGER3D_LKx_PRO

thisiskeithb, 2021-12-26:
Reopening since bugfix/latest 2.0.9.3 still doesn't compile due to `undefined reference to MarlinHAL::adc_enable(short)` errors. Sample build output below, but can also been seen in the [Actions log](https://github.com/MarlinFirmware/Marlin/actions/runs/1623041383):

```prolog
Linking .pio/build/BIGTREE_BTT002_VET6/firmware.elf
/Users/keith/.platformio/packages/toolchain-gccarmnoneeabi@1.90201.191206/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld: .pio/build/BIGTREE_BTT002_VET6/src/src/module/temperature.cpp.o: in function `Temperature::init()':
/Users/keith/Documents/github-mbp.nosync/Marlin/Marlin/src/module/temperature.cpp:2332: undefined reference to `MarlinHAL::adc_enable(short)'
/Users/keith/.platformio/packages/toolchain-gccarmnoneeabi@1.90201.191206/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld: /Users/keith/Documents/github-mbp.nosync/Marlin/Marlin/src/module/temperature.cpp:2368: undefined reference to `MarlinHAL::adc_enable(short)'
/Users/keith/.platformio/packages/toolchain-gccarmnoneeabi@1.90201.191206/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld: /Users/keith/Documents/github-mbp.nosync/Marlin/Marlin/src/module/temperature.cpp:2377: undefined reference to `MarlinHAL::adc_enable(short)'
collect2: error: ld returned 1 exit status
*** [.pio/build/BIGTREE_BTT002_VET6/firmware.elf] Error 1
```

pilot76103, 2021-12-26:
> > @pilot76103 — What is your `MOTHERBOARD` ?
> > Longer LK5 … ?
> 
> #define MOTHERBOARD BOARD_LONGER3D_LKx_PRO
> 
> The issue remains with working config from 2.0.9.2.

One other notable behavior, when I updated from 2.0.9.2 to 2.0.9.3, I get this message.
"2021-12-25 22:09
EEPROM Version Error"

ellensp, 2021-12-26:
@thisiskeithb yes I got this also update Marlin/src/HAL/STM32/HAL.h line 243 to
`  static void adc_enable(const pin_t pin) {} `

ellensp, 2021-12-26:
@pilot76103 yes, this is by design. to force to you refresh your eeprom contents when up update your firmware values.

pilot76103, 2021-12-26:
> @pilot76103 yes, this is by design. to force to you refresh your eeprom contents when you up update your firmware values.

Ok, still need to figure out what causes the bed and thermistor to show -15ºC now instead of the actual temperature. Behavior changed between the 2.0.9.2 release and 2.0.9.3.

thinkyhead, 2021-12-26:
@pilot76103 — A good assortment of typos (and one particularly ugly one) are now fixed. You *might* get better results now.

pilot76103, 2021-12-26:
> @pilot76103 — A good assortment of typos (and one particularly ugly one) are now fixed. You _might_ get better results now.

Compiling now. Will update in just a moment.

pilot76103, 2021-12-26:
> @pilot76103 — A good assortment of typos (and one particularly ugly one) are now fixed. You _might_ get better results now.

@thinkyhead, no joy. Still showing -15ºC for both.

ellensp, 2021-12-26:
@pilot76103 please open your own issue

thinkyhead, 2021-12-26:
I found that I had badly transcribed the code for `adc_start` (AVR). That probably has something to do with it! A patch has been pushed….

thisiskeithb, 2021-12-26:
I can open a new issue, but now I'm having issues with SPI-based TMC drivers.

<details><summary>Here's the output of a build with 5160s on X/Y/Z and a TMC2209 on E (UART-based TMCs seem fine on this build & others):</summary>
<p>

```log
Send: M122
Recv: 		X	Y	Z	E
Recv: Address					0
Recv: Enabled		false	false	false	false
Recv: Set current	0	0	0	0
Recv: RMS current	94	94	94	55
Recv: MAX current	133	133	133	78
Recv: Run current	0/31	0/31	0/31	0/31
Recv: Hold current	0/31	0/31	0/31	0/31
Recv: Global scaler	256/256	256/256	256/256
Recv: CS actual	0/31	0/31	0/31	30/31
Recv: PWM scale	0	0	0
Recv: vsense					0=.325
Recv: stealthChop	false	false	false	true
Recv: msteps		256	256	256	8
Recv: interp		false	false	false	true
Recv: tstep		0	0	0	max
Recv: PWM thresh.
Recv: [mm/s]
Recv: OT prewarn	false	false	false	false
Recv: triggered
Recv:  OTP		false	false	false	false
Recv: pwm scale sum				36
Recv: pwm scale auto				0
Recv: pwm offset auto				36
Recv: pwm grad auto				14
Recv: off time	0	0	0	3
Recv: blank time	16	16	16	36
Recv: hysteresis
Recv:  -end		-3	-3	-3	-3
Recv:  -start		1	1	1	6
Recv: Stallguard thrs	0	0	0	0
Recv: uStep count	0	0	0	16
Recv: DRVSTATUS	X	Y	Z	E
Recv: sg_result	0	0	0	0
Recv: stallguard
Recv: fsactive
Recv: stst		*	*	*
Recv: olb
Recv: ola
Recv: s2gb
Recv: s2ga
Recv: otpw
Recv: ot
Recv: 157C
Recv: 150C
Recv: 143C
Recv: 120C
Recv: s2vsa
Recv: s2vsb
Recv: Driver registers:
Recv: 		X	0x00:00:00:00	 Bad response!
Recv: 		Y	0x00:00:00:00	 Bad response!
Recv: 		Z	0x00:00:00:00	 Bad response!
[...]
Recv: 
Recv: 
Recv: Testing X connection... Error: All LOW
Recv: Testing Y connection... Error: All LOW
Recv: Testing Z connection... Error: All LOW
Recv: Testing E connection... OK
Recv: //action:notification TMC CONNECTION ERROR
```

</p>
</details>

pilot76103, 2021-12-26:
> 

@thisiskeithb, could this be related to my issue?
https://github.com/MarlinFirmware/Marlin/issues/23356

thinkyhead, 2021-12-26:
The HAL refactor has been reverted and postponed (#23357) since it has not gotten enough testing in the two weeks since its creation. It's difficult to get testing on PRs, and I cannot always find the time to do testing, so I will often just push to bugfix to see what shakes out. There have also been recent changes related to PWM and timers that have some degree of uncertainty, but hopefully the main issues are now solved by reverting the HAL.

thisiskeithb, 2021-12-26:
> The HAL refactor has been reverted and postponed (#23357) since it has not gotten enough testing in the two weeks since its creation.

I've confirmed the revert resolved my SPI TMC issue.

With that, I'd say any issues should be posted on the linked PR as users test it.

closed 2021-12-26 by thisiskeithb