MarlinFirmware/Marlin #23355: [BUG] Thermistor and Bed Report -15ºC Instead of Actual Temperature on 2.0.9.3

pilot76103, 2021-12-26:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

On Marlin 2.0.9.3, the thermistor and bed report -15ºC instead of the actual temperature. Worked on 2.0.9.2 without issue.

### Bug Timeline

12/25/2021

### Expected behavior

Thermistor and Bed should report the actual temperature.

### Actual behavior

Thermistor and Bed show -15ºC as the current temperature.

### Steps to Reproduce

Compile 2.0.9.3 with a working configuration from 2.0.9.2.

### Version of Marlin Firmware

2.0.9.3

### Printer model

Longer LK5 Pro

### Electronics

Stock

### Add-ons

N/A

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Cura

### Host Software

OctoPrint

### Additional information & file uploads

https://www.dropbox.com/s/aixnwqw4y05v7d3/Marlin%202.0.9.3%20Bug%20Report.zip?dl=0

Changed the DGUS Reloaded

ellensp, 2021-12-26:
Fixed in bugfix (so far)  https://github.com/MarlinFirmware/Marlin/commit/00e6e90648012ca0b954139f867a9a0201319209

pilot76103, 2021-12-26:
@ellensp and @thinkyhead, confirmed resolved under 516793bd2c8d3755d950a29bf3b74c16100ccc21.

closed 2021-12-26 by pilot76103