MarlinFirmware/Marlin #23360: [BUG] 2.0.9.3 (current version after Hal rollback) : SKR2.0b not reading SD cards from BTT screen in 12864 mode

Kaween-prog, 2021-12-26:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Tried 2 different SKR2.0b's, and 2 different type of BTT screens with different firmware on them, and used several different SD cards for these tests. All come up with the same issue : the SD card content is NOT being read anymore (screen still shows "print from Media" (but does this REGARDLESS of media having been inserted into the SD slot, so the normal "No Media" display when no SD card is inserted does NOT show up anymore)

This still worked flawless in 2.0.9.2 and bugfix versions up to 8/12/2021.

* tested on 2 different SKR2.0B's : same result
* tested on 2 different BTT screens (2.4 and 3.5, both with old and new firmwares) : same result
* tested on different SD cards (which work under the previous Marlin version) : same result
* changed toggling the "#define SD_DETECT_STATE HIGH" option in configuration_adv.h : same result
* tried current bugfix download : board doesn't even start (? same effect as reported yesterday, caused by the HAL-issue, which is resolved in the current 2.0.9.3 Release download, but not in the bugfix download.
* tried with the default configuration.h and configuration_adv.h files 
with #define SDSUPPORT enabled
display set as #define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER
and tested with both  #define SD_DETECT_STATE HIGH enabled and disabled.

Makes no difference : SD card will not read, contents does not show, media change does not get recognized either.
However, it now shows a "release media/attach media" menu option (which doesn't yield any result) which was NOT present when compiling 2.0.9.2 or bugfix version up to 8/12/21

Tested again with a Fysetc Mini 12864 2.1 display : same result : display functions, NO SD card is being recognized nor read.

### Bug Timeline

New bug since release of 2.0.9.3 with the Hal-roll-back

### Expected behavior

* correct media detection when pulling the SD card out of the display (showing "no media")
* correct SD card reading when SD card is inserted in the SD slot of the display

(as said, still works with 2.0.9.2 and with bugfix of 8/12/2021)

### Actual behavior

* media change is not being seen (pulling SD card from display SD slot does not result in "no media" message)
* inserted SD card allows "print from media" to select, but nothing can be read from the card.

### Steps to Reproduce

1) download current 2.0.9.3 and compile (even with the stock configuration.h and configuration_adv.h files, with of course the display of your choice and an active "SDsupport" define in place
2) try to select a file from SD card : not functioning (as explained above)

Note : when the TFT3.5 (or 2.4) is functioning in "touch" mode (serial interface instead of using EXP1/EXP2), the SD cards ARE being recognized and read correctly.

### Version of Marlin Firmware

2.0.9.3

### Printer model

Anycubic Predator with SKR2.0b and TFT3.5 display (but problem also exists when using the default configuration.h and configuration_adv.h files)

### Electronics

stock

### Add-ons

stock

descipher, 2021-12-26:
FYI
  /**
   * Set this option to one of the following (or the board's defaults apply):
   *
   *           LCD - Use the SD drive in the external LCD controller.
   *       ONBOARD - Use the SD drive on the control board.
   *  CUSTOM_CABLE - Use a custom cable to access the SD (as defined in a pins file).
   *
   * :[ 'LCD', 'ONBOARD', 'CUSTOM_CABLE' ]
   */
  //#define SDCARD_CONNECTION LCD

Kaween-prog, 2021-12-26:
@descipher : Hah ! Well, this works, but I'm kinda puzzled as to "why".
Reason : I (clearly wrongly) interpreted this code to be directly connected to the binary file transfer option, so I never un-remarked that line, yet .... it always worked previously even without it on the same SKR boards;

But regardless, thanks for the pointer, it did solve the issue.

closed 2021-12-26 by Kaween-prog

descipher, 2021-12-26:
Your welcome.
Previously the default was LCD in the pins file b4 #23274.