MarlinFirmware/Marlin #23361: [BUG] Bugfix-2.0.x no longer respects `Z_MIN_PROBE_PIN` (MKS Robin nano / STM32F1)

dburr, 2021-12-26:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

For various reasons, I wanted to leave my printer's Z endstop switches connected while at the same time using a BLTouch, so that I have the option of switching between ABL and MBL without having to take the printer apart and swap connections on the motherboard. To this end, I have my BLTouch hooked up to a different pin on my MKS Robin nano board, and have defined `Z_MIN_PROBE_PIN` in `Configuration.h` to point to that pin (`PA2`). With the latest `bugfix-2.0.x` (as of 2021-12-26 12:40 PST) the `Z_MIN_PROBE_PIN` is no longer being honored, and the printer is defaulting back to using `Z_MIN_PIN` for the probe pin. To test this, I homed the printer while manually triggering my Z endstop switch (which is still hooked up to the "real" `Z_MIN` pin) and the printer correctly stopped its Z motion.

### Bug Timeline

bugfix-2.0.x (as of 2021-12-26 12:40 PST)

### Expected behavior

Printer homes using the BLTouch hooked up to a custom pin as defined by `Z_MIN_PROBE_PIN` in `Configuration.h`

### Actual behavior

Printer still homes using `Z_MIN_PIN`

### Steps to Reproduce

1. Hook BLTouch to a custom pin, and set it in `Z_MIN_PROBE_PIN` in `Configuration.h`
2. Try homing the printer

### Version of Marlin Firmware

bugfix-2.0.x

### Printer model

Sainsmart Coreception 300

### Electronics

MKS Robin nano 1.2

### Add-ons

BLTouch

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Prusa Slicer

### Host Software

OctoPrint

### Additional information & file uploads

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/7777549/Configuration.zip)

Sjoerd, 2021-12-27:
Same issue in 2.0.9.2 and 2.0.9.1!

ellensp, 2021-12-30:
You have #define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN set
This says the probe is on the Z_MIN_PIN
Setting Z_MIN_PROBE_PIN will not override this
Disable Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN then it uses Z_MIN_PROBE_PIN

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- 3D Printing Discord server. Join link: https://discord.gg/qNnSzsug
- Marlin specific Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2021-12-30 by ellensp