MarlinFirmware/Marlin #23364: [BUG] possible bug/behavior change : LCD screen stops showing filename.gcode during prints ?

Kaween-prog, 2021-12-27:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Not sure if it's a bug or a "known change in behavior" which escaped me ?

Up to Marlin 2.0.9.1,  whenever printing occured, the bottom line of a 12864 RRDCFG would show the filename of the print in progress on the bottom line, scrolling in case it was longer than the screen width provided.  

However, starting from 2.0.9.2 and onwards, this behavior has changed : while a printer will start the job showing the print file name, whenever something induces a change in for example temperature (say, temperature of bed is lowered after 5 layers), the filename will be replaced with a "Bed Cooling" prompt, which remains static on the display and will continue to show even when the selected temperature was reached.

When the machine is Paused during such a situation, resuming the print will again show the actual print filename to be displayed.

2.0.9.3 shows the same behavior.  The change in behavior does not seem to be connected to the used gcode file, as using the same gcode files on a 2.0.9.1 (and earlier) firmware version does not show this change of behavior.

Also, at the same time, the Mesh Edit screen (the "radar screen" editor) seems to not be "exitable" anymore.  On older firmwarebuilds (2.0.8.2 for sure, not sure about later versions), whenever I select the graphic Mesh edit, the system returns to the main info screen for a flash of a second during manual modifications of the Z-height.  When "left to its own devices" for a couple of seconds, this results in an "exit" from the Mesh Edit screen and the capability to save/store the edited points.

In 2.0.9.3 however, no such effect is seen : the "short flash of the info screen" doesn't happen during editing in Mesh Edit, but there also seems litterally no way to exit the Mesh Edit radar screen anymore ?

I'm not sure I'm missing something in terms of changed functionality or intended behavior here ?  Anyone able to shine some light into this matter in terms of this being a bug or changed behavior-as intended ?

### Bug Timeline

2.0.9.2 for the filename-during-printing, probably true for the mesh edit "can't exit Radar screen" condition too.

### Expected behavior

expected behavior : file name of print remaining on screen/scrolling on the bottom line during print UNLESS the slicer introduces other behavior over an M117 command   (which is not the case)

### Actual behavior

filename of current printjob gets overwritten by heater status messages ("bed cooling" ect). 

### Steps to Reproduce

effect 1 : (the loss of the filename display during prints)
use 2.0.9.2 onwards to compile a firmware
print gcode (it's not slicer related : as said, the same gcode on 2.0.8.2 shows the filename constantly, starting 2.0.9.2 this changed)

effect 2 : (UBL Mesh Edit "can't escape the radar screen")
use 2.0.9.3 onwards to compile a firmware
run a UBL procedure, populate mesh, save to slot 0
start Mesh Edit ("radar"): individual points can be accessed and changed correctly, but it is impossible to exit the radar screen ?

### Version of Marlin Firmware

2.0.9.3

### Printer model

Generic Delta, Kossel clone

### Electronics

SKR2.0b board running Marlin

### Add-ons

no add-ons

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

IdeaMaker

### Host Software

SD Card (headless)

ellensp, 2021-12-27:
The bottom line has always just been the status line. 
Before marlin didn't give out any status information and it was only set with M117, now marlin does provide status information with ui.set_status.

Kaween-prog, 2021-12-27:
Interesting.  So at least that part is by design.

I know about the "bottom line is status line" of course, but is it really logical to "freeze the last status message" like this ? Just wondering.  If the bed for example drops from 60°C to 55°C after the first layer as intended by the gcode running, is it really that useful to see that "bed cooling" status for the next 24 hours of the print ?

Just wondering. :)

thisiskeithb, 2021-12-27:
See https://github.com/MarlinFirmware/Marlin/pull/23135

closed 2021-12-29 by thisiskeithb