MarlinFirmware/Marlin #23370: [BUG] Possible bug; Home positions for x & y set to 155 & 115 after MANUAL_X_HOME_POS

gotuass, 2021-12-27:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

After updating my firmware to marlin bugfix 2.0.x and adjusting my settings for a bltouch my ender 5 pro with board 4.2.2 will no longer home to x220 and y220. Instead it homes to x155 y115 and it deploys the probe in the back right corner of the bed, instead of going to the middle of the bed to make the z axis homing.
Using pronterface and sending g1 x0 y0 will make the print head go just over the left half of the print bed. This means that x and y 220 is beyond the right side of the print bed.

I can get around this issue by not setting MANUAL_X_HOME_POS and MANUAL_Y_HOME_POS in config.h, but then my printer is off center.

my offsets are;
MANUAL_X_HOME_POS 5
MANUAL_Y_HOME_POS -1

### Bug Timeline

New. 24/12/2021

### Expected behavior

Expected to see x220 and y220 when the printer was homed and in the back right corner. Then expected to see that the printer would be centered when sending g1 x110 y110 via pronterface. also expected that x0 y0 would be in the front left corner of the printer.

### Actual behavior

Printerhead homes to the endstops but stays there for the z axis homing. Also the x and y axis have 155 and 115 respectively as grid coordinates while it should have been x220 y220.

### Steps to Reproduce

download marlin bugfix 2.0.x on github.
replace config.h and config_adv.h with the same name folder but from bugfix configuration examples. creality>ender 5 pro
in config.h uncomment;
bltouch
auto_bed_leveling_bilinear
z_safe_homing
use_probe_for_z_homing
z_min_probe_pin PB1
lcd_bed_leveling

in config_adv.h uncomment
babystep_zprobe_offset
probe_offset_wizard

in config.h comment;
z_min_probe_uses_z_min_endstop_pin

in config.h set
MANUAL_X_HOME_POS 5
MANUAL_Y_HOME_POS -1

compile and flash firmware to printer. using pronterface connect and write command m503 then m500.
auto home the printer and check for normal behavior.

### Version of Marlin Firmware

bugfix-2.0.x

### Printer model

Creality ender 5 pro

### Electronics

V4.2.2 silent board (stock)

### Add-ons

bltouch

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Cura

### Host Software

Pronterface

### Additional information & file uploads

[Marlin config.zip](https://github.com/MarlinFirmware/Marlin/files/7781849/Marlin.config.zip)

https://user-images.githubusercontent.com/96749277/147508969-f02a0ff9-6add-4740-9c52-c7dbfeeda2cc.mp4

caractacus, 2021-12-29:
I am experiencing the same issue on Ender 3 Pro SKR1.3 TMC2209.

So far this has cost me 2 days and an SKR1.3 bricked

[Archive.zip](https://github.com/MarlinFirmware/Marlin/files/7790876/Archive.zip)
.

thinkyhead, 2022-01-05:
Whenever there are homing or leveling issues, we now ask everyone to follow a standard procedure to gather more information so that we're not just taking stabs in the dark. Here is the boilerplate:

- Download Marlin `bugfix-2.0.x` to test with the latest code.
- Enable `DEBUG_LEVELING_FEATURE` and `M114_DETAIL` and re-flash the firmware.
- Connect to your printer from host software such as Cura, Printrun or Repetier Host.
- Send `M502` and `M500` to ensure your Configurations are applied.
- Issue the command `M111 S247` to enable maximum logging.
- Perform a `G28` to do your standard homing procedure.
- Copy the log output into a **.TXT file** and attach it to your next reply.

Repeat this procedure, if needed, to demonstrate inconsistencies. From these logs we should hopefully get a better idea of what's going on with your machine.

thisiskeithb, 2022-01-09:
> // Manually set the home position. Leave these undefined for automatic settings.
> // For DELTA this is the top-center of the Cartesian print volume.
> #define MANUAL_X_HOME_POS 5
> #define MANUAL_Y_HOME_POS -1

This is a configuration issue. Comment these out and reflash. You likely meant to set `X_MIN_POS` and `Y_MIN_POS`.

You have an Ender-5 which homes to X & Y max. You also have a BLTouch with `Z_SAFE_HOMING` enabled using default settings to home in the center of the bed.

Be sure to reset your EEPROM with `M502` followed by `M500` after flashing to ensure bad homing positions/offsets are wiped.

closed 2022-01-09 by thisiskeithb