MarlinFirmware/Marlin #23371: [BUG] More than 2 steps per eISR loop executed?

iZz0s, 2021-12-28:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Hi,

i got recently some messages which i don't know what to do with. in my testings, it have to do with acceleration and linear advance. If i go lower with acceleration (currently M204 P800), it seems to happen a lot times more - on the other hand, it dissapears without linear advance or with higher linear advance values - as example, it happens a lot more times with lin_advance 0.04 then as with 0.06 or higher.

are this messages a problem, need i to change something? prints are looking okay but this messages gave me some open questions.

### Bug Timeline

dont know

### Expected behavior

no eISR loop messages

### Actual behavior

eISR loop messages

### Steps to Reproduce

1. Lower Printing Acceleration > more eISR messages
2. small lin_advance value like 0.04 > more eISR messages
3. disabled lin_advance > no eISR messages
4. higher printing acceleration > no or a alot less eISR messages?

### Version of Marlin Firmware

bugfix-2.0.x newest

### Printer model

Sovol SV01

### Electronics

SKR Mini E3 v2.0

### Add-ons

BMG Extruder, LDO 0.9 Pancake, Dragon Hotend, BLTouch

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Prusa Slicer

### Host Software

Repetier Host

sjasonsmith, 2021-12-31:
You must have `LA_DEBUG` to be seeing these messages. Disable that and they will go away.

Do they indicate a real problem? Probably not. They indicate a fundamental issue with step timing when using linear advance, but the firmware is working as designed. As long as your TMC stepper drivers are not disabling themselves while extruding you are probably fine.

Two things you can do:

1. If you are using more then 16 microsteps on your extruder, turn that down. More microsteps means more work for the firmware, and more likelihood that you will encounter this situation.

2. Disable `LA_DEBUG` in your configuration files. This will eliminate these prints, as well as eliminate some unnecessary overhead in the Linear Advance logic. It's really there for debug/development use.

As it is, I don't think there is a real bug here. Yes, there are improvements to be made to linear advance, but those are captured in various other bugs already.

closed 2021-12-31 by sjasonsmith

iZz0s, 2022-01-01:
Thank you for your answer :)

Yes, i had LA_DEBUG activated and don't know why i did it. I recompile now without it.

Im using 16 Microsteps on every Axis, but 0.9° Steppers on X/Y/E. So its X160/Y160 and E830 (BMG) Steps.

The prints are running and looking good, i was just curious about this message. no stepper problems like disabling, stuttering etc.