MarlinFirmware/Marlin #23380: [FR] NOZZLE_TO_PROBE Z offset value Sanity Check and Notes

rongoodman303, 2021-12-28:
### Is your feature request related to a problem? Please describe.

It is not clear that the NOZZLE_TO_PROBE Z offset value must always be a negative number. Compile is successful with no errors or warnings when a non-negative Z offset is specified, and G29 leveling appears to work. But the result after G29 is a "floating" nozzle, above the bed by the incorrect offset, and inability to successfully get a valid Z offset wizard result. 

### Are you looking for hardware support?

NA

### Describe the feature you want

1) Sanity check should error when a non-negative Z offset is provided.
2) In the examples in the NOZZLE_TO_PROBE_OFFSET section, add below the sample grid:

 * 
 * The Z offset must always be a negative number because the probe must always be higher than the nozzle.

### Additional context

In addition to experiencing this myself, I have seen a number of user posts in various forums about the issue of "floating" nozzles, after G29 ABL, and many of those can probably be attributed to a non-negative Z offset being specified.

thisiskeithb, 2021-12-29:
> It is not clear that the NOZZLE_TO_PROBE Z offset value must always be a negative number. 

That is not correct: https://github.com/MarlinFirmware/Marlin/pull/20344

closed 2021-12-29 by thisiskeithb

rongoodman303, 2021-12-29:
This issue needs to be reopened for further and wider discussion. 

The issue you cited was for a very specific printer and highly unusual situation, and it also clearly specifies that NOZZLE_TO_PROBE_OFFSET_Z_POSITIVE must also be enabled as a double check, which it is NOT. Certainly when BLTOUCH is enabled, it cannot be a positive number, as their documentation makes clear. BLTouch and its many clones are the vast majority of ABL devices in use today, so the negative impact of a non-negative value to a larger community of users outweighs the needs of a very small minority, especially when their issue has already been addressed by other means. 

At the very minimum, the sanity check I proposed needs to be implemented when BLTOUCH is enabled. Whether or not there needs to be another sanity check for NOZZLE_TO_PROBE_OFFSET_Z_POSITIVE when it is a positive value is another issue, because it obviously does not get verified as described in the issue you cited. And a comment needs to be added to the Configuration.h as I suggested, with addition of "for BLTouch and similar ABL devices", to minimize issues that a non-positive value could have to the vast majority of users.