MarlinFirmware/Marlin #23381: [BUG] printer stops/resets mid print (Anet ET4 Pro)

jaakkolantero, 2021-12-28:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

The printer stops mid-print. Usually somewhere from 30mins to 5hours into printing.

### Bug Timeline

I have tested 2.0.9.2, 2.0.9.3 and latest bugfix and all of them fail

### Expected behavior

I expected for the print to complete.

### Actual behavior

It looks like that the printer is restarting. 
- Hotend and bed temp are set to 0. 
- All axes are movable. 
- There is no option to resume printing.
- While logging with PC serial communication dropped a few times before print stopped.
- When the serial connection dropped printer made a clonking sound (probably because it stopped for a subsecond)
- All longer prints have failed so far.

### Steps to Reproduce

1. Start printing
2. Printing stops mid print

### Version of Marlin Firmware

FIRMWARE_NAME:Marlin bugfix-2.0.x (Dec 28 2021 17:22:55)

### Printer model

Anet ET4 Pro

### Electronics

Stock

### Add-ons

Printer is behind UPS.

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Cura

### Host Software

SD Card (headless)

### Additional information & file uploads

[configurations.zip](https://github.com/MarlinFirmware/Marlin/files/7786295/configurations.zip)

## Software Fault detected ## on line 18065
[print_log_error.txt](https://github.com/MarlinFirmware/Marlin/files/7786304/print_log_error.txt)

[AET4PRO_z_axis_parkside_mount.zip](https://github.com/MarlinFirmware/Marlin/files/7786346/AET4PRO_z_axis_parkside_mount.zip)

ellensp, 2021-12-29:
Short version for anyone looking

```
## Software Fault detected ##
Cause: Hard
R0   : 0x00000001
R1   : 0x00000003
R2   : 0x40020C00
R3   : 0x7833B99B
R12  : 0x00000000
LR   : 0x080181CF
PC   : 0x0801E4D0
PSR  : 0x61000000
CFSR : 0x00000000
HFSR : 0x40000000
DFSR : 0x00000000
AFSR : 0x00000000
MMAR : 0x7833B99F
BFAR : 0x7833B99F
ExcLR: 0xFFFFFFE9
ExcSP: 0x2001FE60
Backtrace:#1 : unknown@0x0801E4CC+4 PC:0x0801E4D0
#2 : unknown@0x08036F48+402438318 PC:0x200027F6
```

descipher, 2021-12-29:
Questions:
Have you a tried different SD?
What version was it working on prior to this state?
Did you try without the UPS supply?

jaakkolantero, 2021-12-29:
Thank you for looking into this so fast. I really appreciate it.

- Yes, I have tried 3 different SD cards.
- It worked with Anet's stock firmware. Stock firmware was slightly over extruding so that was one of the reasons to change to marlin.
- I got UPS 2 days ago so it was running without it before that.

Forgot to mention these changes:
- I changed hotend with thermistor and heater (old one was really dirty)
- I have tried different PSU.
- I have checked connections (pins stay in place and connectors are fitted properly)

I could test an older version of marlin but "Anet ET4 Pro" config was introduced in 2.0.8.0 (If I remember correctly) so to test older versions I have to reference https://github.com/davidtgbe/Marlin to get the configuration right.
What version should I test?

I have STLink if that is needed to debug the problem.

So if I would have to guess the problem is with mainboard or marlin. I will remove the mainboard tonight and check if something pops up.

descipher, 2021-12-29:
I suspect power loss recovery.
Have you run a print without it?

robbycandra, 2021-12-30:
I have also experienced more printer reset during printing lately. 

In the past, this problem was usually solved using a faster SDCard or reformat the card. Power Loss Recovery tends to make SDCard broken. 

Now, I have experienced 3 Printer resets. in maybe about 50 print jobs. But that time I use USB Flash Drive on SKR2. After changing back to SDCard, the problem seems to disappear.

jaakkolantero, 2021-12-30:
Unfortunately changing M413(Power loss recovery) didn't help.

I removed the mainboard and found some issues with grounding. I just started a new print so fingers crossed. I also changed USB cable to a shorter one. I'll let you know how it went and what I did if the changes worked.

robbycandra, 2021-12-30:
Yes... grounding can also be another problem. The old Anycubic formax has an issue with grounding so users have to make case for the 12868 LCD.

jaakkolantero, 2022-01-02:
And it seems to work now! Believe it or not but it was a grounding issue(I don't know how I missed this the first time) 🙈 Here is what I did if anyone is facing same issue.

I had to remove some paint from case to get connection between AC protective earth and corner of mainboard.
[image]

I also grounded stepper motors to DC negative to get rid of possible static charge based on this thread https://forum.duet3d.com/topic/10879/how-to-ground-steppers/10

I even changed back to 3meter usb cable and there has been zero lost connections after fixing grounding issue.

closed 2022-01-02 by jaakkolantero