MarlinFirmware/Marlin #23385: [BUG] Z-offset not working in real time (while printing)

TalZiv, 2021-12-29:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Hi,

SKR mini e3 v3 + bl-touch.

On my original ender 3 board I could set z-offset in real time while printing and tune it to get the best 1st layer, Now when I turn the dial it does nothing.
if I save it the next print will use this value which defeat the purpose of tuning it on the fly while printing

Please advise, Thank you

### Bug Timeline

new

### Expected behavior

Z offset to change while printing

### Actual behavior

Z offset does not reflect in real time

### Steps to Reproduce

1. print
2. change z-offset
3. 

### Version of Marlin Firmware

2.0.9.3 and tested SKR e3 v3 firmware which is bugfix branch

### Printer model

Creality Ender 3

### Electronics

SKR mini e3 v3

### Add-ons

bl-touch

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Cura

### Host Software

OctoPrint

GMagician, 2021-12-29:
Not sure but I suppose you have to enable BABYSTEP_ZPROBE_OFFSET

thisiskeithb, 2021-12-29:
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- [Marlin Documentation](https://marlinfw.org) hosted on MarlinFW.org
- [Marlin Forum](https://forums.reprap.org/list.php?415) hosted on RepRap.org
- [Marlin Firmware](https://www.facebook.com/groups/1049718498464482/) Facebook Group 
- [Marlin Firmware for 3D Printers](https://www.facebook.com/groups/3Dtechtalk/) Facebook Group
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- [Marlin Firmware](https://discord.gg/n5NJ59y) Discord server

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2021-12-29 by thisiskeithb

TalZiv, 2021-12-30:
For any one that will come here, the issue was that even that BABYSTEPPING was enabled the BABYSTEP_MULTIPLICATOR_Z was set to 1 so any change made to the Z-offset was so minute it was barely noticeable, I changed it to 4 and could see some changes in the Z axis.