MarlinFirmware/Marlin #23386: [BUG] #define PINS_DEBUGGING does not compile for SKR mini e3 V3.0

SinkWhole, 2021-12-29:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

When moving to a new SKR mini V3.0 the VS Code stopped compiling with following errors:

Compiling .pio\build\STM32G0B1RE_btt\src\src\gcode\control\M17_M18_M84.cpp.o
In file included from Marlin\src\gcode\config\../../pins/pinsDebug.h:170,
                 from Marlin\src\gcode\config\M43.cpp:29:
Marlin\src\gcode\config\../../pins/../HAL/STM32/pinsDebug.h: In function 'int8_t digital_pin_to_analog_pin(pin_t)':
Marlin\src\gcode\config\../../pins/../HAL/STM32/pinsDebug.h:164:14: error: 'NUM_ANALOG_FIRST' was not declared in this scope; did you mean 'PNUM_ANALOG_BASE'?
  164 |   Ard_num -= NUM_ANALOG_FIRST;
      |              ^~~~~~~~~~~~~~~~
      |              PNUM_ANALOG_BASE
Compiling .pio\build\STM32G0B1RE_btt\src\src\gcode\control\M211.cpp.o
Compiling .pio\build\STM32G0B1RE_btt\src\src\gcode\control\M400.cpp.o
*** [.pio\build\STM32G0B1RE_btt\src\src\gcode\config\M43.cpp.o] Error 1

I tried the latest marlin and the Bigtreetech fork https://github.com/bigtreetech/Marlin/tree/SKR-mini-E3-V3.0-G0B1
I use the Auto Build marlin Panel 

### Bug Timeline

since using SKR Mini V3.0 

### Expected behavior

Succesfull Firmware Compile

### Actual behavior

FAILED

### Steps to Reproduce

1.having #define PINS_DEBUGGING in configuration_adv.h enabled
2. Compile Marlin using ABM Panel in VS Code in the STM32G0B1RE_BTT environment

### Version of Marlin Firmware

bugfix-2.0.x

### Printer model

Ender 3 Pro

### Electronics

BTT SKR mini e3 V3.0

thisiskeithb, 2021-12-29:
This is being worked on: https://github.com/MarlinFirmware/Marlin/pull/22896

ellensp, 2022-02-21:
This is not something that is being looked into in https://github.com/MarlinFirmware/Marlin/pull/22896

thisiskeithb, 2022-02-21:
Oh! I thought you were patching pins debugging on STM32, which would include this board.

ellensp, 2022-02-21:
only  on boards I have access to, I don't have access to a STM32G0B1RE to be able to check it

InsanityAutomation, 2022-02-21:
I use maple for development on these boards, then once im done with debugging functions test if it still works ok on stm32.

Eddiiie, 2022-03-14:
I have this board (STM32G0B1RE).  Is there anything I can do to help test m43 fix?  :)

thisiskeithb, 2022-03-14:
> I have this board (STM32G0B1RE). Is there anything I can do to help test m43 fix? :)

Can you test/provide feedback to @ellensp in #22896?

Eddiiie, 2022-03-14:
Hi @ellensp (and @thisiskeithb  )
Not sure if I did this right, but I think so... because it compiles!!  M43 would not compile before your patches.

Here is what I did:
Using Marlin `CONFIGURATION_ADV_H_VERSION 02000903`   (is there a better way to get the version?)
Went to the Files view https://github.com/MarlinFirmware/Marlin/pull/22896/files
Found the files on my computer; renamed them to .old and made new one's from your fixes..
That is:  
```
Marlin/src/HAL/STM32/pinsDebug.h
Marlin/src/gcode/config/M43.cpp
```
I also did
```
buildroot/share/PlatformIO/variants/MARLIN_BIGTREE_OCTOPUS_PRO_V1_F429/variant.h
buildroot/share/PlatformIO/variants/MARLIN_BIGTREE_OCTOPUS_V1/variant.h
/```
But those were not the correct files..  
I found the correct file -
`buildroot/share/PlatformIO/variants/**MARLIN_G0B1RE/variant_MARLIN_STM32G0B1RE.h**`

Made the following changes:
```
//#define NUM_DIGITAL_PINS        62
#define NUM_DIGITAL_PINS        112
#define NUM_ANALOG_FIRST		NUM_DIGITAL_PINS
```

As I said, Marlin now compiles with M43 enabled!
Unfortunately, my printer is printing,  I need 10 hours to report back.

Hope this helps.

Eddiiie, 2022-03-14:
Print job finished, firmware updated.   Sent M43.   It looks like it works.   There are a couple of strange characters on the terminal window (using Octoprint) for some of the pins...  Unused pins I guess?

Here is the output...

```
Send: M43
Recv: PIN: PA8         M42 P8           NEOPIXEL_PIN                           Output = 0
Recv: PIN: PA9         M42 P9           BTN_EN1                                Input  = 1
Recv: PIN: PA10        M42 P10          BTN_EN2                                Input  = 1
Recv: PIN: PA11        M42 P11          <unused/unknown>                       Input  = 0
Recv: PIN: PA12        M42 P12          <unused/unknown>                       Input  = 0
Recv: PIN: PA13        M42 P13          <unused/unknown>                       Alt Function: 0  - system (misc. I/O)
Recv: PIN: PA14        M42 P14          <unused/unknown>                       Alt Function: 0  - system (misc. I/O)
Recv: PIN: PA15        M42 P15          BTN_ENC                                Input  = 1
Recv: PIN: PB3         M42 P19          E0_STEP_PIN                            protected
Recv: PIN: PB4         M42 P20          E0_DIR_PIN                             protected
Recv: PIN: PB5         M42 P21          BEEPER_PIN                             Output = 0
Recv: PIN: PB6         M42 P22          I2C_SCL_PIN                            Input  = 1
Recv: PIN: PB7         M42 P23          I2C_SDA_PIN                            Input  = 1
Recv: PIN: PB8         M42 P24          LCD_PINS_RS                            Output = 0
Recv: PIN: PB9         M42 P25          LCD_PINS_D4                            Output = 1
Recv: PIN: PB13        M42 P29          X_STEP_PIN                             protected
Recv: PIN: PB14        M42 P30          X_ENABLE_PIN                           protected
Recv: PIN: PB15        M42 P31          CONTROLLER_FAN_PIN                     protected
Recv: .                                 FAN2_PIN                               protected
Recv: PIN: PC0         M42 P32          X_MIN_PIN                              protected
Recv: .                                 X_STOP_PIN                             protected
Recv: PIN: PC1         M42 P33          Y_MIN_PIN                              protected
Recv: .                                 Y_STOP_PIN                             protected
Recv: PIN: PC2         M42 P34          Z_MIN_PIN                              protected
Recv: .                                 Z_STOP_PIN                             protected
Recv: PIN: PC3         M42 P35          SD_DETECT_PIN                          Input  = 0
Recv: PIN: PC6         M42 P38          FAN_PIN                                protected
Recv: PIN: PC7         M42 P39          E0_AUTO_FAN_PIN                        protected
Recv: .                                 FAN1_PIN                               protected
Recv: PIN: PC8         M42 P40          HEATER_0_PIN                           protected
Recv: PIN: PC9         M42 P41          HEATER_BED_PIN                         protected
Recv: PIN: PC10        M42 P42          <unused/unknown>                       Alt Function: 0  - system (misc. I/O)
Recv: PIN: PC11        M42 P43          <unused/unknown>                       Alt Function: 0  - system (misc. I/O)
Recv: PIN: PC12        M42 P44          POWER_LOSS_PIN                         Input  = 0
Recv: PIN: PC13        M42 P45          <unused/unknown>   Analog in =     0   Input  = 0
Recv: PIN: PC14        M42 P46          Z_MIN_PROBE_PIN                        protected
Recv: PIN: PC15        M42 P47          <unused/unknown>   Analog in =     0   Input  = 0
Recv: PIN: PD0         M42 P48          <unused/unknown>   Analog in =     0   Input  = 0
Recv: PIN: PD1         M42 P49          E0_ENABLE_PIN                          protected
Recv: PIN: PD2         M42 P50          <unused/unknown>   Analog in =     0   Input  = 0
Recv: PIN: PD3         M42 P51          <unused/unknown>   Analog in =     0   Input  = 0
Recv: PIN: PD4         M42 P52          <unused/unknown>   Analog in =     0   Input  = 0
Recv: PIN: PD5         M42 P53          <unused/unknown>   Analog in =     0   Input  = 0
Recv: PIN: PD6         M42 P54          LCD_PINS_ENABLE                        Output = 0
Recv: PIN: PD8         M42 P55          LED_PIN                                Output = 0
Recv: PIN: PD9         M42 P56          <unused/unknown>   Analog in =     0   Input  = 0
Recv: PIN: PF0         M42 P57          <unused/unknown>   Analog in =     0   Input  = 0
Recv: PIN: PF1         M42 P58          <unused/unknown>   Analog in =     0   Input  = 0
Recv: PIN: PF2         M42 P59          <unused/unknown>   Analog in =     0   Input  = 0
Recv: PIN: ÞÓ\x03\x08\x15 (A1)   M42 P1           <unused/unknown>                       Output = 0
Recv: PIN: ÞÓ\x03\x08\x15 (A1)   M42 P1           <unused/unknown>                       Output = 0
Recv: PIN: ÞÓ\x03\x08\x15 (A1)   M42 P1           <unused/unknown>                       Output = 0
Recv: PIN: ÞÓ\x03\x08\x15 (A1)   M42 P1           <unused/unknown>                       Output = 0
Recv: PIN: ÞÓ\x03\x08\x15 (A1)   M42 P1           <unused/unknown>                       Output = 0
Recv: PIN: ÞÓ\x03\x08\x15 (A1)   M42 P1           <unused/unknown>                       Output = 0
Recv: PIN: ÞÓ\x03\x08\x15 (A1)   M42 P1           <unused/unknown>                       Output = 0
[... cut ...]
Recv: PIN: ÞÓ\x03\x08\x15 (A1)   M42 P1           <unused/unknown>                       Output = 0
Recv: PIN: ÞÓ\x03\x08\x15 (A1)   M42 P1           <unused/unknown>                       Output = 0
Recv: PIN: ÞÓ\x03\x08\x15 (A1)   M42 P1           <unused/unknown>                       Output = 0
Recv: PIN: ÞÓ\x03\x08\x15 (A1)   M42 P1           <unused/unknown>                       Output = 0
Recv: PIN: ÞÓ\x03\x08\x15 (A1)   M42 P1           <unused/unknown>                       Output = 0
Recv: ok

```

Triangulix, 2022-07-30:
Same issue for the BTTT SKR 3

dstrohl, 2022-08-07:
For the BTT SKR 3 (which I have also and is also having the same problem using v2.1.1 (Bug fix))  I checked "buildroot\share\PlatformIO\variants\MARLIN_H743Vx\variant_MARLIN_STM32H743VX.h", which I *think* is the correct file that it would be pulling from and the NUM_ANALOG_FIRST is not defined.  I am assuming that the changes in pinsDebug.h are there in this branch and therefore the variant file is the only thing that needs updating.

I dont however know what to put in there... what is that value supposed to be?  in the variant.h file for Marlin BTT E3 RRF,  it is set to 35, 

I'm happy to set this to something and try it.  any suggestions as to how to determine the right value?

dstrohl, 2022-08-07:
In case it helps, here is the schematic:  https://github.com/bigtreetech/SKR-3/blob/master/Hardware%20(SKR%203%20EZ)/BIGTREETECH%20SKR%203%20EZ%20V1.0-SCH.pdf

cbagwell, 2022-08-29:
I just swapped out my Creality board for an SKR Mini E3 V3.0 and ran into this compile issue.  I've mostly gotten it working but could use some input on a submittable version.

It seems obvious that NUM_ANALOG_FIRST should be the pin # of the first analog input.  So I set mine to PIN_A0 but that is set to PNUM_ANALOG_BASE which might be a better value.  BTW, the value is set to 0xc0/192 from platformio.

```
--- a/buildroot/share/PlatformIO/variants/MARLIN_G0B1RE/variant_MARLIN_STM32G0B1RE.h
+++ b/buildroot/share/PlatformIO/variants/MARLIN_G0B1RE/variant_MARLIN_STM32G0B1RE.h
@@ -124,6 +124,7 @@
 #define NUM_DIGITAL_PINS        62
 #define NUM_REMAP_PINS          2
 #define NUM_ANALOG_INPUTS       16
+#define NUM_ANALOG_FIRST       PIN_A0
 
 // SPI definitions
 #ifndef PIN_SPI_SS
```

With that I could compile and issue M43 but the analog pins were missing from the output.  It looks like at least VALID_PIN() was not written expecting there to be a hole between NUM_DIGITAL_PINS and NUM_ANALOG_FIRST.  So I made this change:

```
--- a/Marlin/src/HAL/STM32/pinsDebug.h
+++ b/Marlin/src/HAL/STM32/pinsDebug.h
@@ -112,7 +112,7 @@ const XrefInfo pin_xref[] PROGMEM = {
   #define HAS_HIGH_ANALOG_PINS 1
 #endif
 #define NUMBER_PINS_TOTAL NUM_DIGITAL_PINS + TERN0(HAS_HIGH_ANALOG_PINS, NUM_ANALOG_INPUTS)
-#define VALID_PIN(ANUM) ((ANUM) >= 0 && (ANUM) < NUMBER_PINS_TOTAL)
+#define VALID_PIN(ANUM) (((ANUM) >= 0 && (ANUM) < NUM_DIGITAL_PINS) || ((ANUM) >>= PNUM_ANALOG_BASE && (ANUM) < NUM_ANALOG_INPUTS))
 #define digitalRead_mod(Ard_num) extDigitalRead(Ard_num)  // must use Arduino pin numbers when doing reads
 #define PRINT_PIN(Q)
 #define PRINT_PIN_ANALOG(p) do{ sprintf_P(buffer, PSTR(" (A%2d)  "), DIGITAL_PIN_TO_ANALOG_PIN(pin)); SERIAL_ECHO(buffer); }while(0
```

Are those analog #define's I used valid values for all STM32 variants?

With that, I can see my analog pins now:

```
Send: M43
Recv: PIN: PA0  (A0)   M42 P3           <unused/unknown>   Analog in =     0   Input  = 0
Recv: PIN: PA1  (A1)   M42 P4           SERVO0_PIN                             Output = 0
Recv: PIN: PA2  (A2)   M42 P5           <unused/unknown>                       Alt Function: 1  - TIM1/TIM2 (probably PWM)
Recv: PIN: PA3  (A3)   M42 P6           <unused/unknown>                       Alt Function: 1  - TIM1/TIM2 (probably PWM)
Recv: PIN: PA4  (A4)   M42 P7           SDSS                                   Output = 1
Recv: PIN: PA5  (A5)   M42 P16          DOGLCD_SCK                             Alt Function: 0  - system (misc. I/O)
Recv: PIN: PA6  (A6)   M42 P17          <unused/unknown>                       Alt Function: 0  - system (misc. I/O)
Recv: PIN: PA7  (A7)   M42 P18          DOGLCD_MOSI                            Alt Function: 0  - system (misc. I/O)
Recv: PIN: PA8         M42 P8           NEOPIXEL_PIN                           Output = 0
Recv: PIN: PA9         M42 P9           BTN_EN1                                Input  = 1
Recv: PIN: PA10        M42 P10          BTN_EN2                                Input  = 1
Recv: PIN: PA11        M42 P11          <unused/unknown>                       Input  = 0
Recv: PIN: PA12        M42 P12          <unused/unknown>                       Input  = 1
Recv: PIN: PA13        M42 P13          <unused/unknown>                       Alt Function: 0  - system (misc. I/O)
Recv: PIN: PA14        M42 P14          <unused/unknown>                       Alt Function: 0  - system (misc. I/O)
Recv: PIN: PA15        M42 P15          BTN_ENC                                Input  = 1
Recv: PIN: PB0  (A8)   M42 P19          Z_STEP_PIN                             protected
Recv: PIN: PB1  (A9)   M42 P20          Z_ENABLE_PIN                           protected
Recv: PIN: PB2  (A10)  M42 P21          Y_DIR_PIN                              protected
Recv: PIN: PB3         M42 P19          E0_STEP_PIN                            protected
Recv: PIN: PB4         M42 P20          E0_DIR_PIN                             protected
Recv: PIN: PB5         M42 P21          BEEPER_PIN                             Output = 0
Recv: PIN: PB6         M42 P22          I2C_SCL_PIN                            Input  = 1
Recv: PIN: PB7         M42 P23          I2C_SDA_PIN                            Input  = 1
Recv: PIN: PB8         M42 P24          LCD_PINS_RS                            Output = 0
Recv: PIN: PB9         M42 P25          LCD_PINS_D4                            Output = 1
Recv: PIN: PB10 (A11)  M42 P22          Y_STEP_PIN                             protected
Recv: PIN: PB11 (A12)  M42 P23          Y_ENABLE_PIN                           protected
Recv: PIN: PB12 (A13)  M42 P24          X_DIR_PIN                              protected
Recv: PIN: PB13        M42 P29          X_STEP_PIN                             protected
Recv: PIN: PB14        M42 P30          X_ENABLE_PIN                           protected
[... 15 lines cut ...]
Recv: PIN: PC9         M42 P41          HEATER_BED_PIN                         protected
Recv: PIN: PC10        M42 P42          <unused/unknown>                       Alt Function: 0  - system (misc. I/O)
Recv: PIN: PC11        M42 P43          <unused/unknown>                       Alt Function: 0  - system (misc. I/O)
Recv: PIN: PC12        M42 P44          POWER_LOSS_PIN                         Input  = 0
Recv: PIN: PC13        M42 P45          <unused/unknown>   Analog in =     0   Input  = 0
[... cut ...]
Recv: PIN: PF2         M42 P59          <unused/unknown>   Analog in =     0   Input  = 0
Recv: PIN: 2¼\x03\x08\x15 (A1)   M42 P1           <unused/unknown>                       Output = 0
Recv: PIN: 2¼\x03\x08\x15 (A1)   M42 P1           <unused/unknown>                       Output = 0
Recv: ok P15 B3
```

I'm not sure what those 2 last items are yet.  I only know that pins_xref[] is returning Pin 1 for both which isn't a valid pin.  Perhaps its as simple as digital pins need to be reduced from 62 to 60.

thisiskeithb, 2022-09-16:
Fixed in https://github.com/MarlinFirmware/Marlin/pull/24748

closed 2022-09-16 by thisiskeithb