MarlinFirmware/Marlin #23390: [BUG] BLTouch not working with BIGTREETECH-SKR-mini-E3 V2 when using Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN

LazeMSS, 2021-12-30:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Trying to use the latest 2.0.9.3 with BLTouch on the BIGTREETECH-SKR-mini-E3 V2 - with the BLTouch attached to the z-probe pin (PC14) - printer is an Ender 3 Pro

So basically using the BLtouch as both z-endstop and probe - and connected to the PC14 port - and NOT connected to the Z-stop port (PC2)

I have enabled BLTouch and use the Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN setting - then when doing a G28 it will center to the bed and then deploy and stow the bltouch 3-4 times and the fail with:  Error:!! STOP called because of BLTouch error - restart with M999

The strange this I can make it work by changing the pin assignement in:
https://github.com/MarlinFirmware/Marlin/blob/2.0.9.3/Marlin/src/pins/stm32f1/pins_BTT_SKR_MINI_E3_common.h#L51

If I change that line to:
`#define Z_STOP_PIN                          PC14`

then everything work fine - but I don't like making changes to the pin files.

Will upload config files ASAP

### Bug Timeline

Found the bug in the prev. release and also in the bugfix

### Version of Marlin Firmware

Marlin 2.0.9.3 - Dec 30 2021 11:00:34

### Printer model

Creality Ender 3

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Prusa Slicer

### Host Software

OctoPrint

ellensp, 2021-12-30:
Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN says your probe is plugged into the z-min endstop plug.
Do not set this.
Enable USE_PROBE_FOR_Z_HOMING is you want to use the probe for homing

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- 3D Printing Discord server. Join link: https://discord.gg/qNnSzsug
- Marlin specific Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2021-12-30 by ellensp

LazeMSS, 2021-12-30:
Sorry - It was actually the other way around:
If I use the Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN  and the change the pin assigment everything works as it should. If I use the USE_PROBE_FOR_Z_HOMING I get an PRINTER HALTED always. I will do more digging and write when I have it all sorted out.