MarlinFirmware/Marlin #23392: [BUG] Power outage in Marlin 2.0 starts the print from the beginning

Maxisy, 2021-12-30:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

I have an Ender 3, I'm using Marlin bugfix-2.0.x. Power outage feature starts printing the model from the beginning.

### Bug Timeline

I tried using power outage today, after updating Marlin.

### Expected behavior

I expected the printing to continue from the point that power outage happened at.

### Actual behavior

It started printing from the beginning.

### Steps to Reproduce

1. I start a print (G-code generated with Cura)
2. I unplug the printer from the outlet.
3. I plug it back and select continue printing.
4. The printer warms up and it starts printing from the beginning.

### Version of Marlin Firmware

Marlin bugfix-2.0.x

### Printer model

Ender 3 32bit board

### Electronics

Stock electronics (motherboard v4.2.2)

### Add-ons

none

### Bed Leveling

MBL Manual Bed Leveling

### Your Slicer

Cura

### Host Software

SD Card (headless)

### Additional information & file uploads

My marlin configurations: https://gofile.io/d/FvVNnF (I can't upload here, I don't know why)

thinkyhead, 2022-01-05:
If you enable the `DEBUG_POWER_LOSS_RECOVERY` option you can use some special commands to check the integrity of power-loss recovery.
```cpp
M413 L : recovery.load()
M413 W : recovery.save(true)
M413 P : recovery.purge()
M413 D : recovery.debug(F("M413"))
M413 O : recovery._outage() ... (Requires POWER_LOSS_PIN)
M413 E : SERIAL_ECHOF(recovery.exists() ? F("PLR Exists\n") : F("No PLR\n"))
M413 V : SERIAL_ECHOF(recovery.valid() ? F("Valid\n") : F("Invalid\n"))
```
If the PLR file is created correctly with `M413 W` and is shown to be valid with `M413 V` then we can follow up by checking the startup logic which is supposed to detect the PLR file and resume the print at the proper point in the file.

Maxisy, 2022-01-05:
Thank you. I'm new to 3D printing so could you explain it to me more thoroughly please? Where should I put this code?

thinkyhead, 2022-01-05:
Configuration.h:
```cpp
#define DEBUG_POWER_LOSS_RECOVERY
```

Then you can issue the `M413` commands as described from your host console.

Maxisy, 2022-01-05:
Thank you, I will add that line to my Configuration.h and install new version on my printer and get back to you.

closed 2022-03-24 by github-actions[bot]