MarlinFirmware/Marlin #23393: [BUG] Artillery Genius Pro (Ruby 32bit) Does not Boot

Dviros, 2021-12-30:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

After flashing compiled versions 2.0.9.2 and 2.0.9.3 with the updated Artillery Genius Pro configs, including my own modified ones, the board does not boot, requiring to boot to DFU.
Can confirm that compiling Artillery's 2.0.9.1 firmware and configs are indeed working as expected.

### Expected behavior

Board to boot.

### Actual behavior

Board does not boot.

### Steps to Reproduce

1. Use the latest version.
2. Implement the added configs and compile.
3. Flash on DFU, exit DFU post successful flash.

### Version of Marlin Firmware

2.0.9.1

### Printer model

Artillery Genius Pro

### Electronics

Stock

### Add-ons

/

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Cura

### Host Software

OctoPrint

### Additional information & file uploads

Configs can be found here:

https://github.com/MarlinFirmware/Configurations/tree/06d4c7d2309d11b20a7050933a6abb2d025091d1/config/examples/Artillery/Genius%20Pro

Sonicboom247, 2021-12-30:
tagging along, hopefully I mislabeled my issue #23244 and it is not the display but actually the board not booting an we can get this resolved.

ellensp, 2021-12-31:
those configs arent even part of marlin yet...

thinkyhead, 2022-01-02:
Since the issue manifested between 2.0.9.1 and 2.0.9.2 perhaps you can help us find the exact date in `bugfix-2.0.x` where the problem started. 

Start at the date halfway between 2.0.9.1 and 2.0.9.2, and then keep going halfway to the date of the working/nonworking version as you find that it breaks/works. It should only take 4 or 5 tests to narrow down the exact day.

The board is apparently the MKS Gen. L so maybe I can do some tests with the bare board and compare notes.

Sonicboom247, 2022-01-02:
@thinkyhead the board is the Artillery Ruby, i am out of town until tomorrow night but from what I recall the issue manifested about two months ago when I first reported it

thisiskeithb, 2022-01-02:
> Since the issue manifested between 2.0.9.1 and 2.0.9.2

Artillery Ruby board support wasn’t added to Marlin until November 1 / 2.0.9.3, so I’m not sure how that is possible: https://github.com/MarlinFirmware/Marlin/pull/23029

I suspect MKS’ fork/files are being used here.

Sonicboom247, 2022-01-02:
That's so odd, i have a working 2.0.9.1 and am pretty sure it was working since 2.0.7.1 or 2 as that's what came on the hornet when stock. But you are correct, somewhere between 2.0.9.1 and bugfix it went south for me.

thisiskeithb, 2022-01-02:
> That's so odd, i have a working 2.0.9.1 and am pretty sure it was working since 2.0.7.1 or 2 

Did you backport board support or know where you found those versions? Ruby board support *was not* added to Marlin until 2.0.9.3.

Maybe @gzhuli / Artillery can hop in here and help since I don’t have one of these boards to troubleshoot with a debugger.

Sonicboom247, 2022-01-02:
I'd be happy to send you the working version i have if it helps, it's been a year now and I don't recall all the sources i used to get it to work as i just recently started really adding timestamps to my changes. I do recall working with @ellensp who discovered my working version stopped working when env:Artillery_Ruby switched to env: artillery_ruby, then to env:Marlin_Artillery_Ruby and most recently back to env:Artillery_Ruby

thisiskeithb, 2022-01-02:
@Sonicboom247: You are definitely mixing up your versions. This board was added as `env:Artillery_Ruby` and still is.

Please start over with a completely clean copy of the latest bugfix and manually migrate your old config settings over line by line. Be sure to pay attention to any code changes/parameter renames if your config is really old.

Do not copy in any other folders or files from other copies of Marlin.

Does it compile? Does it boot?

PrEzi3, 2022-01-04:
Maybe I am mixing things up, but I lately tried the 2.9.0.2 bugfix code late october, when support for Ruby was added and faced very similar behavior as described here (mainboard not booting, TFT working fine but no connection on any port any speed) on the Sidewinder X2 (Ruby board - same as Genius Pro or Genius 2). And here is the firmware 2.9.0.1 that was provided by Artillery which boots fine 
https://github.com/artillery3d/sidewinder-x2-firmware

In the next couple of days I will try to flash the firmware based of 2.9.0.3 source and see if it makes any difference, but this described here is definitely not a fluke.

gzhuli, 2022-01-04:
Seems it was my fault that mixing up the code when confirming the PR. The startup code was missing due to .gitignore skipping *.s files and ignorecase=true in config, thus startup.S was ignored when committing.

I've created a PR to correct this. Sorry for the inconvenience.

Dviros, 2022-01-04:
Thanks @gzhuli.
I'm wondering if it's possible to add a generic dependency file check for potential other boards as well, making sure that all required files exists in the board path prior to flashing.

thisiskeithb, 2022-01-04:
This will be fixed in #23446

closed 2022-01-04 by thisiskeithb

PrEzi3, 2022-01-04:
@gzhuli  @thisiskeithb --- confirmed --- #23446  fixed the issue. Thank you!

Dviros, 2022-01-04:
Can confirm as well that this fixes the issue. Thanks a lot!

sjasonsmith, 2022-01-04:
I tend to think bugs should remain open until after the PR is merged.