MarlinFirmware/Marlin #23394: Ender 3 v2 DWIN scroll wheel - dents are out of sync with menu item ticks

RipperDrone, 2021-12-30:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

When I turn the LCD display wheel one notch, the menu sometimes jumps to the next/previous item as supposed, sometimes it doesn't. Feels like the turn wheel resolution of notches does not quite match with the encoder wheel's number of 'clicks' needed to go to the next/previous item. Feels strange, sometimes having to turn the wheel by two notches, sometimes by one notch only...

Is there anything in the firmware taking the encoder singnals into consideration so this de-sync in resolution can be fixed?

### Bug Timeline

Since day 1 of using this firmware (July 2021)

### Expected behavior

Turning the jog dial wheel by one notch should make the menu move to the next/previous item, depending on direction of turning.

### Actual behavior

Sloppy 'sometimes menu item does change, sometimes it needs another notch turn' behavior

### Steps to Reproduce

Load 2.0.9.3 or latest (1hr ago'S) 2.0.x bugfix version of firmware onto Ender 3 v2 printer, update display with matching DWIN directory via SD card, start osing the printer's menu.

### Version of Marlin Firmware

2.0.9.3 or bugfix 2.0.x

### Printer model

Ender 3 v2

### Electronics

SKR Mini E3 2.0

### Add-ons

Hemera direct drive extruder, BLTouch

### Bed Leveling

ABL Linear grid

### Your Slicer

Cura

### Additional information & file uploads

[Configurations.zip](https://github.com/MarlinFirmware/Marlin/files/7794407/Configurations.zip)

tome9111991, 2021-12-30:
I dont have this issue.
Maybe you have issue with you display?
Or check connection of you display cable.

RipperDrone, 2021-12-30:
@tome9111991 - did you carefully turn the knob, and you are getting a precise 'scroll forward/backward one menu item per knob notch'? If so, my hardware seems to behave differently - turning the jog wheel forward and backward, I can provoke a 'no menu item change, have to turn further' event once every 3-4 attempts...

I have checked the cable, however it is a pigtail mod added to the original cable, to make it fit the DWIN LCD wiring visa SKR Mini E3 2.0. Thus, maybe extending length by 100mm makes the cable connection susceptible to noise. However, the display works great and without any blackouts or flickering whatsoever. To me it still looks like the jog wheel notches are just a LITTLE less angle increment than the firmware expects to register it as a 'previous item/next item' command... (?)

RipperDrone, 2021-12-30:
@tome9111991 - wow, thanks for the hint to the wiring, I re-plugged the pigtail pins, routed the cable a littel differently and tried again --> This time everything worked ok. Seems to actually be the cable! 

Closing this issue...

closed 2021-12-30 by RipperDrone