MarlinFirmware/Marlin #23397: [BUG] (Release media doesn't work)

Cysign, 2021-12-31:
### Did you test the latest `bugfix-2.0.x` code?

No, but I will test it now!

### Bug Description

'release media' in the menu doesn't work on Anet A8 Plus, running Marlin 2.0.9.3 .

### Bug Timeline

-

### Expected behavior

I ecspect Marlin to release SD card.

### Actual behavior

Nothing happens.

### Version of Marlin Firmware

2.0.9.3

### Printer model

Anet A8 Plus

### Electronics

stock electronics

### Bed Leveling

MBL Manual Bed Leveling

### Your Slicer

Cura

Cysign, 2021-12-31:
Using the current bugfix-release the fan speed cannot be changed at all.

ellensp, 2022-01-01:
"I ecspect Marlin to release SD card."  

The actual slot is purely mechanical, it cant eject the sdcard
Its not an advanced 32 bit controller. It can't give the sdcard to the operating system...

what exactly do you imagine this should do?

Cysign, 2022-01-01:
It acts like an "Eject USB device" on Windows, I guess. At least my other printer atcs like this - but it's runnint on Marlin 2.0.x Bugfix release...

Won't removing sd card without releasing harm any files?
And after removing it without releasing Marlin needs a RESET to work properly, again (else it'll get a random reset).

descipher, 2022-01-03:
No configuration files included ..

thinkyhead, 2022-01-03:
Does the G-code command `M22` work?

kaoalex, 2022-01-03:
It was expected the marlin software unmount the sd card before we should be able to remove it safely.
To install Marlin 2.0.9.3 to Anet A8 I had to disable the SD writing option (SDCARD_READONLY), I think this should be the reason the option 'release media' is not working.

thinkyhead, 2022-01-03:
I'm not seeing any evidence of that. All the `card.release()` method does is close open files and clear `flag.mounted`, then the user interface should be updating itself to reflect the change in mounted state according to that same flag. You can try adding a serial message to `CardReader::release` and see that it is happening…
```cpp
SERIAL_ECHOLNPGM("card.release() was called!");
```

ignisf, 2022-04-10:
Hello, just wanted to say that I can confirm that the issue seems to be present in today's `bugfix` branch. Tested on Ender 3 Pro:

* Configuration: https://gist.github.com/ignisf/f8e49ab8dd4a6cdae897cdd56a03f33e
* Video: https://user-images.githubusercontent.com/1153188/162631292-639bbf45-a9cf-4905-91de-93d725ba8cb0.mp4

thisiskeithb, 2022-04-10:
> Does the bot re-open issues closed for staleness?

No, but us mortals can.

ignisf, 2022-04-10:
Please let me know if I can test anything. I did not quite understand where @thinkyhead wanted the debug statement inserted.

thisiskeithb, 2022-04-10:
> Please let me know if I can test anything. I did not quite understand where @thinkyhead wanted the debug statement inserted.

Add `SERIAL_ECHOLNPGM("card.release() was called!");` below this line in `cardreader.cpp`: https://github.com/MarlinFirmware/Marlin/blob/7a7485ce7140121b9c0e9b4bddc47d228117bd61/Marlin/src/sd/cardreader.cpp#L525

closed 2022-06-21 by github-actions[bot]