MarlinFirmware/Marlin #23406: [BUG] (Ramps board: bed heater pin is not defined when LASER_FEATURE is defined)

hwmland, 2021-12-31:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

When LASER_FEATURE is defined, HEATER_BED_PIN is not defined for MKS_GEN_L board (even laser pins are defined on different pins):
```cpp
#define LASER_FEATURE
#if EITHER(SPINDLE_FEATURE, LASER_FEATURE)
  #define SPINDLE_LASER_ACTIVE_STATE    LOW    // Set to "HIGH" if SPINDLE_LASER_ENA_PIN is active HIGH

  #define SPINDLE_LASER_USE_PWM                // Enable if your controller supports setting the speed/power
  #if ENABLED(SPINDLE_LASER_USE_PWM)
    #define SPINDLE_LASER_PWM_PIN   44
    #define SPINDLE_LASER_ENA_PIN SERVO2_PIN
    #define SPINDLE_LASER_PWM_INVERT    false  // Set to "true" if the speed/power goes up when you want it to go slower
    #define SPINDLE_LASER_FREQUENCY     5000   // (Hz) Spindle/laser frequency (only on supported HALs: AVR and LPC)
  #endif
```

### Bug Timeline

Pull request #22983

### Expected behavior

bed heater pin should be defined

### Actual behavior

bed heater pin is not defined

### Steps to Reproduce

Use MKS_GEN_L board printer configuration and define additionally LASER_FEATURE (laser using different pin)

### Version of Marlin Firmware

commit b07887687e

### Printer model

Ender 3

### Electronics

MKS_GEN_L

### Add-ons

Laser, bltouch

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Cura

### Host Software

OctoPrint

sjasonsmith, 2021-12-31:
@hwmland can you attach configuration files which reproduce this build failure?

hwmland, 2021-12-31:
[Configuration_files.zip](https://github.com/MarlinFirmware/Marlin/files/7797040/Configuration_files.zip)
Sure, here you are. There is as well (in Configuration.h) my workaround for this problem. Warkaround is simple, but it was somehow difficult (for me) to find core reason for reported error in sanity check.

descipher, 2021-12-31:
I can confirm the issue exists, it appears to be related to the ramp selections.
If the board is RAMPS_xx_SF it fails.
pins.h line 48
#elif MB(RAMPS_13_SF,  RAMPS_14_SF,  RAMPS_PLUS_SF,  RAMPS_14_RE_ARM_SF,  RAMPS_SMART_SF,  RAMPS_DUO_SF,  RAMPS4DUE_SF) || EITHER(SPINDLE_FEATURE, LASER_FEATURE)

Looking deeper ..

hwmland, 2021-12-31:
@descipher : well, I'd say that ``` || EITHER(SPINDLE_FEATURE, LASER_FEATURE)``` makes the problem (this is part added in recent PR).
 ```RAMPS_xx_SF``` means (in my understanding) RAMPS + Spindle/Laser + Fan. - and here heated bed is not expected, everything OK.
I see following solutions (one of):
1. remove again this ``` || EITHER(SPINDLE_FEATURE, LASER_FEATURE)```
2. check in pins_RAMPS.h if laser/spindle is defined on the pin used usually for heater. If yes - fine. If not - define heater_bed on the usual pin

descipher, 2021-12-31:
Yes but there are reasons for it that need to be accounted. This is part of some reorg on the FET vs MOSFET macros. Not sure of all the components yet. Will need to ping @thinkyhead and see what the intent was B4 I go hacking anything in the pins.h defines. 
 

What is your feature set requirement. 3D printer and laser combined? etc.

descipher, 2022-01-01:
Ok, looks like we should only have spindle as a set for #define FET_ORDER_SF 1 since we can have combined 3D & Laser operations.  THERMAL_PROTECTION_BED it would only fail if the temp sensor is defined which would not occur in the case of a spindle. Keep in mind this fault only occurs because no bed temp array defined when there is a sensor enabled so it's not only the missing HEATED_BED_PIN.

hwmland, 2022-01-01:
Yah, my feature set is 3D printer + laser. And I know as well about use cases 3D+laser+spindle. Laser is quite natural (does not require big forces), spindle is questionable, but in the case of usage as an CNC drill - makes sense.

descipher, 2022-01-01:
The PR will address it, really there is no need base the FET pins on the activation of the  LASER or SPINDLE features since RAMPS_XX_SF already does that by definition.

hwmland, 2022-01-01:
I found 2 other issues related to this change:
1) FAN_PIN get moved from pin 9 to pin 8 (this is fixed by proposed PR)
2) FAN1_PIN is not defined anymore (previously on pin 7). Not fixed by current PR
	=> reason: '''IS_RAMPS_xxx''' was not defined at all. Now detected as '''FET_ORDER_EFB'''

Moreover - now there is always some ```FET_ORDER_xxx``` defined (#else falback to ```FET_ORDER_EFF``` in pins.h), hence last part (disable SF) is never reached in pins_RAMPS.h:
```cpp
#if FET_ORDER_EFB                                 // Hotend, Fan, Bed
  #define HEATER_BED_PIN            MOSFET_C_PIN
#elif FET_ORDER_EEF                               // Hotend, Hotend, Fan
  #define HEATER_1_PIN              MOSFET_B_PIN
#elif FET_ORDER_EEB                               // Hotend, Hotend, Bed
  #define HEATER_1_PIN              MOSFET_B_PIN
  #define HEATER_BED_PIN            MOSFET_C_PIN
#elif FET_ORDER_EFF                               // Hotend, Fan, Fan
  #define FAN1_PIN                  MOSFET_C_PIN
#elif DISABLED(FET_ORDER_SF)                      // Not Spindle, Fan (i.e., "EFBF" or "EFBE")
  #define HEATER_BED_PIN            MOSFET_C_PIN
  #if EITHER(HAS_MULTI_HOTEND, HEATERS_PARALLEL)
    #define HEATER_1_PIN            MOSFET_D_PIN
  #else
    #define FAN1_PIN                MOSFET_D_PIN
  #endif
#endif
```

descipher, 2022-01-01:
=> reason: '''IS_RAMPS_xxx''' was not defined at all. Now detected as '''FET_ORDER_EFB'''

IS_RAMPS_ defines are replaced by FET_ORDER_

SanityCheck.h
#elif ANY(IS_RAMPS_EEB, IS_RAMPS_EEF, IS_RAMPS_EFB, IS_RAMPS_EFF, IS_RAMPS_SF)
  #error "The IS_RAMPS_* conditionals (for heater/fan/bed pins) are now called FET_ORDER_*."

hwmland, 2022-01-01:
I know, that they get replaced. But the point is still the same.

descipher, 2022-01-01:
Unless I am missing something for the MKS_GEN_L it will never set RAMPS_XXX, it uses it's own defines or will use the defaults that are set in pins_ramps.h which are the same as b4. FAN1_PIN is set to MOSFET_D_PIN which defaults to 7 in MKS_GEN_L.h 

Are you seeing something different?

~~~
#define BOARD_INFO_NAME "MKS GEN L"
//
// Heaters / Fans
//
// Power outputs EFBF or EFBE
#define MOSFET_D_PIN 7
~~~

hwmland, 2022-01-01:
I think you are missing fact, that based on pins.h
```cpp
#if   MB(RAMPS_13_EFB, RAMPS_14_EFB, RAMPS_PLUS_EFB, RAMPS_14_RE_ARM_EFB, RAMPS_SMART_EFB, RAMPS_DUO_EFB, RAMPS4DUE_EFB)
  #define FET_ORDER_EFB 1
#elif MB(RAMPS_13_EEB, RAMPS_14_EEB, RAMPS_PLUS_EEB, RAMPS_14_RE_ARM_EEB, RAMPS_SMART_EEB, RAMPS_DUO_EEB, RAMPS4DUE_EEB)
  #define FET_ORDER_EEB 1
#elif MB(RAMPS_13_EFF, RAMPS_14_EFF, RAMPS_PLUS_EFF, RAMPS_14_RE_ARM_EFF, RAMPS_SMART_EFF, RAMPS_DUO_EFF, RAMPS4DUE_EFF)
  #define FET_ORDER_EFF 1
#elif MB(RAMPS_13_EEF, RAMPS_14_EEF, RAMPS_PLUS_EEF, RAMPS_14_RE_ARM_EEF, RAMPS_SMART_EEF, RAMPS_DUO_EEF, RAMPS4DUE_EEF)
  #define FET_ORDER_EEF 1
#elif MB(RAMPS_13_SF,  RAMPS_14_SF,  RAMPS_PLUS_SF,  RAMPS_14_RE_ARM_SF,  RAMPS_SMART_SF,  RAMPS_DUO_SF,  RAMPS4DUE_SF)
  #define FET_ORDER_SF 1
#elif HAS_MULTI_HOTEND && TEMP_SENSOR_BED
  #define FET_ORDER_EEB 1
#elif HAS_MULTI_HOTEND
  #define FET_ORDER_EEF 1
#elif TEMP_SENSOR_BED
  #define FET_ORDER_EFB 1
#else
  #define FET_ORDER_EFF 1
#endif
```
we have now defined ```FET_ORDER_EFB``` (nothing was defined before. It's defined now because we have ```TEMP_SENSOR_BED```. So then when you look into pins_ramps.h:
```cpp
#if FET_ORDER_EFB                                 // Hotend, Fan, Bed
  #define HEATER_BED_PIN            MOSFET_C_PIN
#elif FET_ORDER_EEF                               // Hotend, Hotend, Fan
  #define HEATER_1_PIN              MOSFET_B_PIN
#elif FET_ORDER_EEB                               // Hotend, Hotend, Bed
  #define HEATER_1_PIN              MOSFET_B_PIN
  #define HEATER_BED_PIN            MOSFET_C_PIN
#elif FET_ORDER_EFF                               // Hotend, Fan, Fan
  #define FAN1_PIN                  MOSFET_C_PIN
#elif DISABLED(FET_ORDER_SF)                      // Not Spindle, Fan (i.e., "EFBF" or "EFBE")
  #define HEATER_BED_PIN            MOSFET_C_PIN
  #if EITHER(HAS_MULTI_HOTEND, HEATERS_PARALLEL)
    #define HEATER_1_PIN            MOSFET_D_PIN
  #else
    #define FAN1_PIN                MOSFET_D_PIN
  #endif
#endif
```
so ```FAN1_PIN``` is not set at all (we go using 1st ```#if```)

descipher, 2022-01-01:
This is B4 the change which has a problem:

~~~
//
// Augmentation for auto-assigning RAMPS plugs
//
#if NONE(IS_RAMPS_EEB, IS_RAMPS_EEF, IS_RAMPS_EFB, IS_RAMPS_EFF, IS_RAMPS_SF) && !PIN_EXISTS(MOSFET_D)
  #if HAS_MULTI_HOTEND
    #if TEMP_SENSOR_BED
      #define IS_RAMPS_EEB
    #else
      #define IS_RAMPS_EEF
    #endif
  #elif TEMP_SENSOR_BED
    #define IS_RAMPS_EFB
  #else
    #define IS_RAMPS_EFF
  #endif
#endif

//
// Heaters / Fans
//
#ifndef MOSFET_D_PIN
  #define MOSFET_D_PIN                        -1
#endif
#ifndef RAMPS_D8_PIN
  #define RAMPS_D8_PIN                         8
#endif
#ifndef RAMPS_D9_PIN
  #define RAMPS_D9_PIN                         9
#endif
#ifndef RAMPS_D10_PIN
  #define RAMPS_D10_PIN                       10
#endif

#define HEATER_0_PIN               RAMPS_D10_PIN

#if ENABLED(IS_RAMPS_EFB)                         // Hotend, Fan, Bed
  #define HEATER_BED_PIN            RAMPS_D8_PIN
#elif ENABLED(IS_RAMPS_EEF)                       // Hotend, Hotend, Fan
  #define HEATER_1_PIN              RAMPS_D9_PIN
#elif ENABLED(IS_RAMPS_EEB)                       // Hotend, Hotend, Bed
  #define HEATER_1_PIN              RAMPS_D9_PIN
  #define HEATER_BED_PIN            RAMPS_D8_PIN
#elif ENABLED(IS_RAMPS_EFF)                       // Hotend, Fan, Fan
  #define FAN1_PIN                  RAMPS_D8_PIN
#elif DISABLED(IS_RAMPS_SF)                       // Not Spindle, Fan (i.e., "EFBF" or "EFBE")
  #define HEATER_BED_PIN            RAMPS_D8_PIN
  #if EITHER(HAS_MULTI_HOTEND, HEATERS_PARALLEL)
    #define HEATER_1_PIN            MOSFET_D_PIN
  #else
    #define FAN1_PIN                MOSFET_D_PIN
  #endif
#endif
~~~

The result is exactly the same as the current code + the proposed PR. 
I have pulled it and checked.

hwmland, 2022-01-02:
B4: Are you sure your board definition is ```BOARD_MKS_GEN_L``` ?
It defines in pins_MKS_GEN_L.h
```cpp
//
// Heaters / Fans
//
// Power outputs EFBF or EFBE
#define MOSFET_D_PIN 7
``` 
hence 
```cpp
//
// Augmentation for auto-assigning RAMPS plugs
//
#if NONE(IS_RAMPS_EEB, IS_RAMPS_EEF, IS_RAMPS_EFB, IS_RAMPS_EFF, IS_RAMPS_SF) && !PIN_EXISTS(MOSFET_D)
  #if HAS_MULTI_HOTEND
    #if TEMP_SENSOR_BED
      #define IS_RAMPS_EEB
    #else
      #define IS_RAMPS_EEF
    #endif
  #elif TEMP_SENSOR_BED
    #define IS_RAMPS_EFB
  #else
    #define IS_RAMPS_EFF
  #endif
#endif
```
is excluded (last part of the condition - pin exists), hence still none of ```IS_RAMP_``` is defined, hence
```cpp
#if ENABLED(IS_RAMPS_EFB)                         // Hotend, Fan, Bed
  #define HEATER_BED_PIN            RAMPS_D8_PIN
#elif ENABLED(IS_RAMPS_EEF)                       // Hotend, Hotend, Fan
  #define HEATER_1_PIN              RAMPS_D9_PIN
#elif ENABLED(IS_RAMPS_EEB)                       // Hotend, Hotend, Bed
  #define HEATER_1_PIN              RAMPS_D9_PIN
  #define HEATER_BED_PIN            RAMPS_D8_PIN
#elif ENABLED(IS_RAMPS_EFF)                       // Hotend, Fan, Fan
  #define FAN1_PIN                  RAMPS_D8_PIN
#elif DISABLED(IS_RAMPS_SF)                       // Not Spindle, Fan (i.e., "EFBF" or "EFBE")
  #define HEATER_BED_PIN            RAMPS_D8_PIN
  #if EITHER(HAS_MULTI_HOTEND, HEATERS_PARALLEL)
    #define HEATER_1_PIN            MOSFET_D_PIN
  #else
    #define FAN1_PIN                MOSFET_D_PIN
  #endif
#endif
```
goes in branch ```#elif DISABLED(IS_RAMPS_SF)``` => ```#else subbranch``` and defines FAN1_PIN

descipher, 2022-01-02:
The pin is defined and the test is !PIN so it will not exec which was confirmed by the pull and compile. But it is mute if the current PR modification changes are merged.

hwmland, 2022-01-02:
Well I created 4 branches in my fork https://github.com/hwmland/Marlin/branches. Only minimal config changes there (bad temp sensor, board MKS_GEN_L, LASER_FEATURE), I added message to show HEATER_BED_PIN and FAN1_PIN. Here are compilation results

**BEFORE_22983** (before refactoring MOSFET pin layout):
Marlin\src\MarlinCore.cpp:1636:41: note: #pragma message: FAN1_PIN=7
Marlin\src\MarlinCore.cpp:1637:47: note: #pragma message: HEATER_BED_PIN=8

**AFTER_22983** (after refactoring MOSFET pin layout):
build fail (root reason: undefined HEATER_BED_PIN):
Marlin\src\module\temperature.cpp:2107:21: error: 'temp_bed' was not declared in this scope
     if (TP_CMP(BED, temp_bed.raw, maxtemp_raw_BED)) max_temp_error(H_BED);

**BUGFIX_ONLY** (bugfix branch before your latest hotfix)
build fail (root reason: undefined HEATER_BED_PIN):
Marlin\src\HAL\AVR\../../inc/SanityCheck.h:99:4: error: #error "Thermal Runaway Protection for the bed is now enabled with THERMAL_PROTECTION_BED."
   #error "Thermal Runaway Protection for the bed is now enabled with THERMAL_PROTECTION_BED."

**BUGFIX_23409** (your bugfix included)
Marlin\src\MarlinCore.cpp:1666:41: note: #pragma message: FAN1_PIN=-1
Marlin\src\MarlinCore.cpp:1667:47: note: #pragma message: HEATER_BED_PIN=8

FAN1_PIN is not defined as a result. 

How about adding into pin_RAMPS.h:
```cpp
#if !PIN_EXISTS(FAN1) && EXTRUDERS && NONE(HAS_MULTI_HOTEND, HEATERS_PARALLEL)
#define FAN1_PIN MOSFET_D_PIN
#endif
```

hwmland, 2022-01-02:
Sorry, FAN1 is irrelevant, FAN is configured right now

hwmland, 2022-01-03:
Thank you for fixing

closed 2022-01-03 by hwmland

descipher, 2022-01-03:
You’re welcome, one of the things Scott reminded me about was the ability to do tool mode selection using GCode. So I will work on that as well. They are M450,451,452,453. They will be added soonish.