MarlinFirmware/Marlin #23407: [BUG] Command M402 stow and deploy probe randomly

saayalac, 2021-12-31:
### Did you test the latest `bugfix-2.0.x` code?

No, but I will test it now!

### Bug Description

When attempting to do bed leveling sometimes the prover does not behave properly.
The probe deploys well, but sometimes when attempting to stow with M402, the probe stows and deploys again
This it's problematic due the probe will hit the glass or the print and damage something

I measured the servo signal with an oscilloscope and I get the following signal (attachment), the PWM looks wrong
It's worth to mention that the probe it's the following https://www.thingiverse.com/thing:3303618

Tested with multiple SG90 and the same problem, wiring, it's properly soldered

### Bug Timeline

Don't know when it started

### Expected behavior

1. Auto home with G28
2. Run M401
3. Deploy probe
4. Run M402
5. Stow probe
6. Repeat from 2 to 5 N times, all works as expected

### Actual behavior

1. Auto home with G28
2. Run M401
3. Deploy probe
4. Run M402
5. Most cases will stow, but randomly stows and then deploys but M402 does not respond anymore. Must use M401 and then M402 to fix
6. Repeat from 2 to 5 N times, some random fails

### Steps to Reproduce

1. Auto home with G28
2. Run M401
3. Deploy probe
4. Run M402
5. Most cases will stow, but randomly stows and then deploys but M402 does not respond anymore. Must use M401 and then M402 to fix
6. Repeat from 2 to 5 N times, some random fails

At least it works in my setup

### Version of Marlin Firmware

2.0.7.2

### Printer model

Ender 3 pro

### Electronics

SKR 1.4 Turbo

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Cura

### Host Software

SD Card (headless)

### Additional information & file uploads

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/7796889/Marlin.zip)
[image]

sjasonsmith, 2021-12-31:
Try disabling `DEACTIVATE_SERVOS_AFTER_MOVE` and report back.

saayalac, 2022-01-22:
Hello!

Sorry for the big delay.

I tested with your comment and it actually solved the problem, I understand that this it's due that the PWM signal keeps on all the time. This solves the problem, but I don't like the signal to be on all time. Do you have any ideas what could cause the original problem?

Thanks

saayalac, 2022-01-23:
An update:

As I expected, testing with some prints, the probe in some random points started to deploy and stow, the oscilloscope shows PWM signal changing making the servo to change position. So DEACTIVATE_SERVOS_AFTER_MOVE it's not a reliable solution to me.

descipher, 2022-01-23:
That's quite the noise pulse with servo detach on your scope, looks like the call for servo detach leaves the pin floating around. This could be the nature of the probe itself not having a signal clamp. What details can you give us about the probe. I will have a look myself since it's possible to resolve with a pin pull down config when the feature is in use.

saayalac, 2022-01-23:
> That's quite the noise pulse with servo detach on your scope, looks like the call for servo detach leaves the pin floating around. This could be the nature of the probe itself not having a signal [...]

Thanks for your help @descipher

I will try to describe my setup.
I have an Ender 3 pro with a SKR 1.4 Turbo.
The probe I use it's from this project (https://www.thingiverse.com/thing:3303618).
The servo, it's a SG90 and the probe it's an optical sensor which it's triggered when a pin enters into it's sensitivity range. (The trigger works ok, the problem it's the servo).

The whole probe it's connected as shown in the following diagram. It's connected to the board with some jumpers, it's not soldered to the board, but the jumpers are properly connected and I have glued them to keep them fixed.
[image]

The cable from jumpers until the servo and the probe it's properly soldered, there are no floating pins/cables.

I know, it looks a bit disorganized, but it's properly connected
As I mentioned, I have tested it with different servos and all have the same behaviour.
[image]
[image]

Hope this info will help.

saayalac, 2022-01-31:
any idea?

descipher, 2022-01-31:
It's noisy, you may have to eliminate the noise. Proper motherboard, case and PSU mains grounding is where to start and or RC filter, Shielded wire, clamping diode, Inductive choke. Keep high current wires away from it, e.g. heater wires. It's also possible to keep the signal output active LOW instead of floating when the servo is disconnected but that's not an appropriate fix for other configs.

saayalac, 2022-02-01:
Ok, makes sense, will check all wiring and case setup. Thanks

closed 2022-04-14 by github-actions[bot]