MarlinFirmware/Marlin #23411: [BUG] 2.0.9.3 Compile errors in module

brosskgm, 2022-01-01:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

In file included from Marlin/src/module/../inc/../HAL/./LPC1768/HAL.h:39,
                 from Marlin/src/module/../inc/../HAL/HAL.h:30,
                 from Marlin/src/module/../inc/MarlinConfig.h:31,
                 from Marlin/src/module/endstops.h:28,
                 from Marlin/src/module/endstops.cpp:27:
Marlin/src/module/endstops.cpp: In static member function 'static void Endstops::update()':
Marlin/src/module/endstops.cpp:736:32: error: 'Z2_MAX' was not declared in this scope; did you mean 'Z_MAX'?
  736 |         COPY_LIVE_STATE(Z_MAX, Z2_MAX);
      |                                ^~~~~~
Marlin/src/module/../inc/../HAL/./LPC1768/../shared/Marduino.h:42:23: note: in definition of macro '_BV'
   42 | #define _BV(b) (1 << (b))
      |                       ^
Marlin/src/module/../inc/../core/macros.h:104:40: note: in expansion of macro 'SBI'
  104 | #define SET_BIT_TO(N,B,TF) do{ if (TF) SBI(N,B); else CBI(N,B); }while(0)
      |                                        ^~~
Marlin/src/module/endstops.cpp:618:45: note: in expansion of macro 'SET_BIT_TO'
  618 |   #define COPY_LIVE_STATE(SRC_BIT, DST_BIT) SET_BIT_TO(live_state, DST_BIT, TEST(live_state, SRC_BIT))
      |                                             ^~~~~~~~~~
Marlin/src/module/endstops.cpp:736:9: note: in expansion of macro 'COPY_LIVE_STATE'
  736 |         COPY_LIVE_STATE(Z_MAX, Z2_MAX);
      |         ^~~~~~~~~~~~~~~
Marlin/src/module/endstops.cpp:736:32: error: 'Z2_MAX' was not declared in this scope; did you mean 'Z_MAX'?
  736 |         COPY_LIVE_STATE(Z_MAX, Z2_MAX);
      |                                ^~~~~~
Marlin/src/module/../inc/../HAL/./LPC1768/../shared/Marduino.h:42:23: note: in definition of macro '_BV'
   42 | #define _BV(b) (1 << (b))
      |                       ^
Marlin/src/module/../inc/../core/macros.h:104:55: note: in expansion of macro 'CBI'
  104 | #define SET_BIT_TO(N,B,TF) do{ if (TF) SBI(N,B); else CBI(N,B); }while(0)
      |                                                       ^~~
Marlin/src/module/endstops.cpp:618:45: note: in expansion of macro 'SET_BIT_TO'
  618 |   #define COPY_LIVE_STATE(SRC_BIT, DST_BIT) SET_BIT_TO(live_state, DST_BIT, TEST(live_state, SRC_BIT))
      |                                             ^~~~~~~~~~
Marlin/src/module/endstops.cpp:736:9: note: in expansion of macro 'COPY_LIVE_STATE'
  736 |         COPY_LIVE_STATE(Z_MAX, Z2_MAX);
      |         ^~~~~~~~~~~~~~~
Marlin/src/module/endstops.cpp:933:60: error: 'Z2_MAX' was not declared in this scope; did you mean 'Z_MAX'?
  933 |     #define PROCESS_ENDSTOP_Z(MINMAX) PROCESS_DUAL_ENDSTOP(Z, MINMAX)
      |                                                            ^
Marlin/src/module/../inc/../HAL/./LPC1768/../shared/Marduino.h:42:23: note: in definition of macro '_BV'
   42 | #define _BV(b) (1 << (b))
      |                       ^
Marlin/src/module/endstops.cpp:861:34: note: in expansion of macro 'TEST'
  861 |   #define TEST_ENDSTOP(ENDSTOP) (TEST(state(), ENDSTOP))
      |                                  ^~~~
Marlin/src/module/endstops.cpp:885:64: note: in expansion of macro 'TEST_ENDSTOP'
  885 |     const byte dual_hit = TEST_ENDSTOP(_ENDSTOP(A, MINMAX)) | (TEST_ENDSTOP(_ENDSTOP(A##2, MINMAX)) << 1); \
      |                                                                ^~~~~~~~~~~~
Marlin/src/module/endstops.cpp:608:32: note: in expansion of macro '__ENDSTOP'
  608 | #define _ENDSTOP(AXIS, MINMAX) __ENDSTOP(AXIS, MINMAX)
      |                                ^~~~~~~~~
Marlin/src/module/endstops.cpp:885:77: note: in expansion of macro '_ENDSTOP'
  885 |     const byte dual_hit = TEST_ENDSTOP(_ENDSTOP(A, MINMAX)) | (TEST_ENDSTOP(_ENDSTOP(A##2, MINMAX)) << 1); \
      |                                                                             ^~~~~~~~
Marlin/src/module/endstops.cpp:933:39: note: in expansion of macro 'PROCESS_DUAL_ENDSTOP'
  933 |     #define PROCESS_ENDSTOP_Z(MINMAX) PROCESS_DUAL_ENDSTOP(Z, MINMAX)
      |                                       ^~~~~~~~~~~~~~~~~~~~
Marlin/src/module/endstops.cpp:1044:13: note: in expansion of macro 'PROCESS_ENDSTOP_Z'
 1044 |             PROCESS_ENDSTOP_Z(MAX);
      |             ^~~~~~~~~~~~~~~~~
*** [.pio/build/LPC1768/src/src/module/endstops.cpp.o] Error 1

### Bug Timeline

New. 8:00pm MST

### Expected behavior

Compile

### Actual behavior

Would not compile. 

Old version 2.0.7.

### Steps to Reproduce

Download and compile.

### Version of Marlin Firmware

2.0.9.3

### Printer model

i3 Mega

### Electronics

 SKR v1.3 TMC2208, No srtand off boards

### Bed Leveling

MBL Manual Bed Leveling

### Additional information & file uploads

[Configuration files.zip](https://github.com/MarlinFirmware/Marlin/files/7797329/Configuration.files.zip)

ellensp, 2022-01-01:
code seems to be making assumption that is you have dual Z, your second endstop is on z-max if z-max is defined 

If you comment out line 78 of   Marlin/src/pins/lpc1768/pins_BTT_SKR_V1_3.h
ie  #define Z_MAX_PIN                        P1_24  // Z+

It will let you compile, at least get you going, but this is just a work around while issue is looked it

brosskgm, 2022-01-01:
I do have dual z.
I thought I enabled it just like I did in the old config. I have all changes marked //// Changed.
I'll do another compare of my two config files and try again.

Thanks

descipher, 2022-01-01:
Do you have 2 physical end stop switches on the Z axis?
I'm certain you don't have to touch #define COOLER_AUTO_FAN_TEMPERATURE 18
It's a Laser cooler.
I don't see a bug here yet.

descipher, 2022-01-01:
Try commenting out #define Z_MULTI_ENDSTOPS for now.

descipher, 2022-01-01:
Confirmed BUG!
diff --git a/Marlin/src/module/endstops.h b/Marlin/src/module/endstops.h
index a35966a98c..00ad046a97 100644
--- a/Marlin/src/module/endstops.h
+++ b/Marlin/src/module/endstops.h
@@ -46,17 +46,28 @@ enum EndstopEnum : char {
   _ES_ITEM(HAS_K_MIN, K_MIN)
   _ES_ITEM(HAS_K_MAX, K_MAX)
 
+
   // Extra Endstops for XYZ
-  _ES_ITEM(HAS_X2_MIN, X2_MIN)
-  _ES_ITEM(HAS_X2_MAX, X2_MAX)
-  _ES_ITEM(HAS_Y2_MIN, Y2_MIN)
-  _ES_ITEM(HAS_Y2_MAX, Y2_MAX)
-  _ES_ITEM(HAS_Z2_MIN, Z2_MIN)
-  _ES_ITEM(HAS_Z2_MAX, Z2_MAX)
-  _ES_ITEM(HAS_Z3_MIN, Z3_MIN)
-  _ES_ITEM(HAS_Z3_MAX, Z3_MAX)
-  _ES_ITEM(HAS_Z4_MIN, Z4_MIN)
-  _ES_ITEM(HAS_Z4_MAX, Z4_MAX)
+  #if ENABLED(X_DUAL_ENDSTOPS)
+    _ES_ITEM(HAS_X_MIN, X2_MIN)
+    _ES_ITEM(HAS_X_MAX, X2_MAX)
+  #endif
+  #if ENABLED(Y_DUAL_ENDSTOPS)
+    _ES_ITEM(HAS_Y_MIN, Y2_MIN)
+    _ES_ITEM(HAS_Y_MAX, Y2_MAX)
+  #endif
+  #if ENABLED(Z_MULTI_ENDSTOPS)
+    _ES_ITEM(HAS_Z_MIN, Z2_MIN)
+    _ES_ITEM(HAS_Z_MAX, Z2_MAX)
+    #if NUM_Z_STEPPER_DRIVERS >= 3
+      _ES_ITEM(HAS_Z_MIN, Z3_MIN)
+      _ES_ITEM(HAS_Z_MAX, Z3_MAX)
+    #endif
+    #if NUM_Z_STEPPER_DRIVERS >= 4
+      _ES_ITEM(HAS_Z_MIN, Z4_MIN)
+      _ES_ITEM(HAS_Z_MAX, Z4_MAX)
+    #endif
+  #endif
 
   // Bed Probe state is distinct or shared with Z_MIN (i.e., when the probe is the only Z endstop)
   #if !HAS_DELTA_SENSORLESS_PROBING

brosskgm, 2022-01-01:
> Do you have 2 physical end stop switches on the Z axis? I'm certain you don't have to touch #define COOLER_AUTO_FAN_TEMPERATURE 18 It's a Laser cooler. I don't see a bug here yet.

Yes. It's a Mega S that I now have SKR v1.3 and no stand off boards, I re-wired it all to run straight off the SKR.

Ok, I wasn't sure about the cooler fan but only disabled it. It didn't say anything about ot being a laser cooler.

descipher, 2022-01-01:
Ok just wanted to confirm, I have it compiling correctly with the posted diff, but there maybe more to deal with.

brosskgm, 2022-01-01:
Surprisingly I never handled any diffs in Marlin before in 4 years of using it.

Do I replace this with all the above?

// Extra Endstops for XYZ
  _ES_ITEM(HAS_X2_MIN, X2_MIN)
  _ES_ITEM(HAS_X2_MAX, X2_MAX)
  _ES_ITEM(HAS_Y2_MIN, Y2_MIN)
  _ES_ITEM(HAS_Y2_MAX, Y2_MAX)
  _ES_ITEM(HAS_Z2_MIN, Z2_MIN)
  _ES_ITEM(HAS_Z2_MAX, Z2_MAX)
  _ES_ITEM(HAS_Z3_MIN, Z3_MIN)
  _ES_ITEM(HAS_Z3_MAX, Z3_MAX)
  _ES_ITEM(HAS_Z4_MIN, Z4_MIN)
  _ES_ITEM(HAS_Z4_MAX, Z4_MAX)

brosskgm, 2022-01-01:
Only gave couple errors, nothing to prevent from trying it.

descipher, 2022-01-01:
- remove
+ add

descipher, 2022-01-01:
The warnings are not an issue.

brosskgm, 2022-01-01:
Does not boot. "No Printer Attached"

brosskgm, 2022-01-01:
#define MOTHERBOARD BOARD_BTT_SKR_V1_3

Did they change the define for this one? I remeber that happening some time ago.

descipher, 2022-01-01:
You need to pull the latest changes. A fix was applied for that issue very recently.

brosskgm, 2022-01-01:
Ok, I'll get it now.

brosskgm, 2022-01-01:
Most of the time I just keep old until new works, but I'm wanting the latest this time because I'm working to add BLTOUCH and keep dual Z and to get bltouch to run from Y+, but haven't been able to find anyone that has config information on it yet.

brosskgm, 2022-01-01:
Downloaded the nightly, and just compiled. I have to check BTT file. Still no printer attached.

brosskgm, 2022-01-01:
Same define. #define BOARD_BTT_SKR_V1_3 in boards.h

brosskgm, 2022-01-01:
BTW I'm using Visual Studio for Ubuntu 20.4

brosskgm, 2022-01-01:
The printer also shows, System Unknown, but that could be because the printer hasn't attached.

descipher, 2022-01-01:
Is the backlight on?
Does the Marlin logo come up?

brosskgm, 2022-01-01:
When I plug in the TFT24 it does, and also shows eprom version error and will need to reset.

LEt me reset and see what it does.

descipher, 2022-01-01:
TFT24? 
//// Changed Enabled
#define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER

brosskgm, 2022-01-01:
Ok, here is an odd one. With the TFT35 V2.0 it does not connect to printer, the TFT24 after I initialized the eeprom it does.

brosskgm, 2022-01-01:
I have it set up I can use both when I'm doing manual mesh level.

brosskgm, 2022-01-01:
The old setup 2.0.7.3 does fine with either screen.

descipher, 2022-01-01:
Ok, test it out. The EEPROM need to be reset.

brosskgm, 2022-01-01:
I did, still same thing. Does not connect through the TFT35 but does fine with TFT24. Oh, the TFT24 is in Marlin mode sees the printer, but when I switch to touch on the TFT24 it says no printer attached also.

brosskgm, 2022-01-01:
Somethign with the touch mode in this latest version is not connecting with the touch side of things.

descipher, 2022-01-01:
you have set it up for Marlin Mode.. no touch screen etc ..
//// Changed Enabled
#define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER

You need to work on the display config before continuing with the other issues.
You can get help via https://discord.gg/DK6ss3ep  see # support

brosskgm, 2022-01-01:
Ok, old version does fine in either mode. Something different between the two setups.

I'll need to work on it.

Thanks for all your help.

closed 2022-01-01 by brosskgm

brosskgm, 2022-01-01:
Was told by BTT this needed to be inabled in the current version for the touch to work.
In the 2.0.7.3 both Marlin and touch mode work fine, I just now need to re-level all over again. Init eliminated the level.

//// Changed Enabled
#define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER

descipher, 2022-01-01:
Yes I understand but you are not clear on what display you have.  
This may help understand the configuration elements.
https://www.youtube.com/watch?v=DodRRgZf5Lo

brosskgm, 2022-01-01:
I have the TFT35 V2.0 Touch and the TFT24 V1.0 dual mode - I can have either connected and work on Marlin 2.0.7.3

The main one used is the BTT TFT35 V2.0 Touch, has no dial. It uses the TFT plug on the SKR board, the TFT 24 uses the EXP 1 & 2 plugs.

The TFT24 has a dual mode, Touch and Marlin, on current version both modes work.

I can have both connected at the same time if needed, I only connect the TFT24 when I'm needing to do a mesh bed leveling because for some reason when using either touch mode, when you click to begin, touch mode takes it into the bed, the Marliln mode stops nozzle at 5mm and lowers to 3mm when you start turning the dial to lower.

brosskgm, 2022-01-01:
Here is the config files from 2.0.7.3 that everything works.

[Configuration-Marlin.2.0.7.3.zip](https://github.com/MarlinFirmware/Marlin/files/7798264/Configuration-Marlin.2.0.7.3.zip)

brosskgm, 2022-01-01:
> Yes I understand but you are not clear on what display you have. This may help understand the configuration elements. https://www.youtube.com/watch?v=DodRRgZf5Lo

That picture in your link is TFT35V3.0 - It has the turner knob. Mine is V2.0 no turner knob.

I guess if they don't show this type in V2.0 everywhere without the truner knob I can see where someone would think one doesn't know their own equipment.

I took it out of it's housing to send you a photo.

[image]

thinkyhead, 2022-01-02:
This needs to be changed, or the Z Max Endstop will not be defined:
```cpp
#define Z_HOME_DIR -1
```

thinkyhead, 2022-01-02:
Endstop handling needs a general overhaul, so that's happening over at #23425. It should be ready to test at some point today.

ahaddad91, 2022-02-23:
check if the reset pin is oriented correctly on the motherboard for the TFT