MarlinFirmware/Marlin #23415: [BUG] The FSMC TFT driver requires libmaple, no longer supported since 7 years.

rewolff, 2022-01-01:
### Did you test the latest `bugfix-2.0.x` code?

No, but I will test it now!

### Bug Description

When attempting to compile marlin fro my MKS robin nano v1.2, I found that any tft inclusion would attempt to include the libmaple header. When attempting to download it, I found: "don't use this anymore" dated from 7 years ago. 

For now: No LCD support for me. 

### Bug Timeline

I don't know. this is the first time I'm trying to build firmware for my Robin. 

### Expected behavior

Compilation succeeds. 

### Actual behavior

Compilation fails. 

### Steps to Reproduce

Copy over the MKS ROBIN config file from "configuration examples" (separate repo) hit compile ("Verify" in arduino-speak). 

### Version of Marlin Firmware

today's github master. 

### Printer model

This is not for a 3D printer. 

### Electronics

MKS robin V1.2

### Add-ons

None

### Additional information & file uploads

test

ellensp, 2022-01-01:
Cannot replicate
Grabbed Configs from https://github.com/MarlinFirmware/Configurations/tree/bugfix-2.0.x/config/examples/Mks/Robin
Code from https://github.com/MarlinFirmware/Marlin/archive/refs/heads/bugfix-2.0.x.zip
Build environment default_envs = mks_robin

It built without issues mks_robin      SUCCESS   00:00:21.171

Full build log  [robin build log.zip](https://github.com/MarlinFirmware/Marlin/files/7798329/robin.build.log.zip)

sjasonsmith, 2022-01-02:
@rewolff please attach your failing configuration files that can be used to reproduce the problem.

I do vaguely remember that some of the TFT code was coupled to Maple, but I never spent much time with that code.

rewolff, 2022-01-02:
I would've been pissed if "bugfix" had worked. It is anoying that I spent two afternoons trying to get it to compile while it had already been fixed. 

But I just switched to the bugfix branch and... same problems. 

Do you have a "libmaple" installed? line 31 of Marlin/src/HAL/STM32F1/tft/tft_fsmc.h reads

`#include <libmaple/dma.h>
`

I have a linux system here, I have arduino 1.8.15, I googled for arduino on stm32, it tells me to add an "additional boards" url, once I do that it tells me that it has been replaced by a different URL, so I change that and only then do I click "install" on the STM32 support package (version 2.1.0) now, in general the STM32 compilation seems to work. 

I am currently doing this on my ubuntu 20.04.3 LTS release. Yesterday, suspecting build systems problems/interferance (I have the arm-gcc compiler from ARM installed)  I created a virtual machine and did a fresh 21.10 install there. Same procedure, same issue. 

Searching for libmaple, I find a copy in an attempted "arduino-for-STM32" directory that I created jan 24 2021. 

Do you guys use the Version 2.1.0 STM32 support from https://github.com/stm32duino/BoardManagerFiles/raw/main/package_stmicroelectronics_index.json ??

[conf.zip](https://github.com/MarlinFirmware/Marlin/files/7798839/conf.zip)

ellensp, 2022-01-02:
Provided config files still compiles without issues  On platfomio 
Marlin/src/HAL/STM32F1/tft/tft_fsmc.h shows you are ruining maple environment 
should be using Marlin/src/HAL/STM32/tft/tft_fsmc.h

issue is with arduino 1.8.15, 

use platfomio. Yes it run fine on linux.
https://marlinfw.org/docs/basics/install_platformio_vscode.html

rewolff, 2022-01-02:
Why aren't there build instructions that say "important" stuff like: use platformio not, arduino? 

I  compiled Marlin a while ago, and arduino was an option then, and it seems to be an option, now, the ".ino" file is provided. 
I tried "apt install platformio" and it says: `E: Unable to locate package platformio` Arduino seems easier in that respect. There is also a SystemCoreClock vs F_CPU issue that I haven't gotten around to reporting yet. That apparently checks for F_CPU being constant, which it isn't (unless I change something in HAL.h). A "cannot be compiled with arduino, please use platformio" check would have saved me quite some time I'm guessing.

rewolff, 2022-01-02:
Following the documentation to install platformio, I get into a loop. 
`Installing Marlin (CLI)
`
I should have already read `installing Marlin with PlatformIO.` So I click  the link. and then click on the I want to use the CLI link and I get to where I started. If I follow the documentation there is an infinite regression on "please read this first". :-(

ellensp, 2022-01-02:
Stop being a die hard linux geek. Use the GUI   (from a Unix veteran of 30+ years) 

1) install vscode for linux https://code.visualstudio.com/docs/setup/linux
2) start vscode
3) install platformio in vscode https://platformio.org/install/ide?install=vscode
4) open marlin source (with your configs) in vscode. Open the folder with the platfomio.ini file in it.
5) edit platfomio and set default_envs = mks_robin
6) Build marlin. Platfomio will download all required dependencies
7) Copy  Robin.bin firmware file from .pio/build/mks_robin/ in your Marlin directory to a sdcard
8) insert sdcard into controller and reboot/power on

rewolff, 2022-01-02:
Sorry I like CLI better. It seems to compile now. It tells me that it made a firmware.bin in `.pio/build/...` Not a single "firmware.bin" to be found. Turns out the firmware is renamed after the "super handy" script told me it built it. Sigh. 

OK. Found it now. I increased bed size to 1000 1000 which is why I needed this. Have a new binary but the board is 3km away from me now. Will upload asap.

rewolff, 2022-01-03:
OK. I have access to the board today. Copied "Robin.bin" to the SD card: no update, immediate response on serial port. 

I then renamed it to Robin_nano.bin and... no joy. 

I then copied the "official firmware" and when I added the config.txt and subdirs, then it suddenly was queit for a few minutes before finally responding to serial commands again. 

So what triggers the udpate? I still haven't been able to upgrade tio my home-compiled Marlin version. (it still reports marlin 1.0.0.)

thinkyhead, 2022-01-03:
The common bootloader should find `Robin.bin` on the SD card inserted into the board directly. Maybe your board has an unusual bootloader. You might try using an ST-Link connected directly to ICSP with the STM32 Cube Programmer app.

thinkyhead, 2022-01-05:
We can't do anything about bad bootloaders, so this is closed.

closed 2022-01-05 by thinkyhead

rewolff, 2022-01-05:
So convince me my bootloader is bad, and that it isn't that the marlin binary that is produced is failing some reasonable test and thus doesn't get uploaded?

That said: The issue I was reporting turns out to be: "It was not documented that you can't compile this with arduino". 

As I've compiled Marlin with arduino in the past, when given two options, I'll chose the one I'm familiar with. There are millions of things: "You can learn to use software X" that I could learn, but I'm limited and can only master a small percentage of those. So given the choice I'll stick with what I know. In hindsight: I can be "friends" with platformio MUCH easier than with arduino. (that is until it breaks, then too much "should be automatic" that nobody documents where to look when it does break).