MarlinFirmware/Marlin #23416: [BUG] (G34 Mechanical / Sanity Check)

mikemerryguy, 2022-01-01:
### Did you test the latest `bugfix-2.0.x` code?

No, but I will test it now!

### Bug Description

I had enabled G34 mechanical allinlgnment on one machine with dual Z

I copied the configs for other machine with single Z and forgot to disable mechanical alignment 

The firmware still compiled and auto z align shows up in menu
I noticed when calibrating.

Not really an "issue" 
But if ibwere to select that menu I would assume it would cause issues.

I also would like to point out that on my ender 5 pro with bltouch
Mechanical alignment could very well cause the machine to break as there is no real hard stop for the bed to stop raising.

Perhaps another sanity check to ensure the Z has a hard stop.

I know there are many variations of printers out there.

Perhaps just a warning 
//WARNING PLEASE BE ADVISED YOUR MACHIBE NEEDS A HARD STOP POAISTION FOR Z TO HIT  DO NOT ENABLE ON MACHINES THAT THE BED CAN COMOLETLY UNSCREW FROM LEAD SCREW! 

something to that effect !!!

Ok after typing all this I realized some people might have dual Z with a belt

Add an option 
Define dual_z_belt_drive under mechanical 

### Bug Timeline

New

### Expected behavior

Sanity check hard stop on compile if number Z =1 and dual z belt drive = false
Error mechanical aliment requires multiple Z 
Driven by two motors or belts
Enable dual z belt drive or define multiple z motors

### Actual behavior

Single z no belt drive setup
Compiles and enables the option for mechanical alignment  

### Steps to Reproduce

Build marlin
Enable g34 mechanical

### Version of Marlin Firmware

2.0.9.3 official Christmas day evening build

### Printer model

Ender 5 pro

### Electronics

Skr mini e3 1.2

### Add-ons

Bltouch

### Bed Leveling

ABL 3-point

### Your Slicer

Cura

### Host Software

SD Card (headless)

sjasonsmith, 2022-01-02:
I added a sanity check for > 1 motor in the above PR.

I didn't add any more complexity or warnings, since Configuration_adv.h already doesn't describe what the feature even does. Users enable it at their own peril if they don't have a mechanically compatible printer. I don't want to litter warnings all over the place unless this turns out to be a common mistake impacting lots of people. I expect you are an uncommon situation where you moved a config from a compatible printer to an incompatible one.

sjasonsmith, 2022-01-02:
Actually, I abandoned that change.

This is actually not a bug, one of our example compilation tests highlighted the issue.

Many printers drive two separate motors from a single stepper driver. In this case this feature will still work fine, even though Marlin only knows about one Z stepper.

So, I am closing this without any change, as this is expected behavior.

closed 2022-01-02 by sjasonsmith