MarlinFirmware/Marlin #23417: [BUG] Board invalid with "BOARD_CREALITY_V4" when compiling

MarcAntoineLamy, 2022-01-01:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

### My problem
When I'm trying to compile for "BOARD_CREALITY_V4" I got the following error:

python version 3.10.1
ERROR - invalid board
BOARD_CREALITY_V4
[Finished in 130ms]

### What I found out
When looking into it I found what, I think, is the source of the problem. In the file "Marlin\src\pins\pins.h" on the line for "CREALITY_V4" there's 2 environment defined "STM32F103RET6_creality" and "STM32F103RET6_creality_maple". When it try to compile with the python script "autobuild.py" it seeing the 2 of them and try to attach a CPU to each one from the predefined CPU list and fails to do so and abort.

I put together my own solution for that problem and I will let it here in case someone need it or tell me I'm wrong and there's  a better way.

### My solution:
I added those line in the function "get_env(board_name, ver_Marlin)" in "autobuild.py" at end of CPU questioning just before it's throw "invalid_board()" error so people get to choose between environment or abort.
`
    elif env_A != '' and env_B != '':
      
      if env_A != '' and env_B != '' and env_C != '':
        get_answer(board_name, 'Which environment?',"Don't know" , env_A, env_B, env_C)
      else:
        get_answer(board_name, 'Which environment?',"Don't know", env_A, env_B)

      if 1 == get_answer_val:
        invalid_board()
      elif 2 == get_answer_val:
        target_env = env_A
      elif 3 == get_answer_val:
        target_env = env_B
      elif 4 == get_answer_val:
        target_env = env_C
`

### Bug Timeline

I don't know tried to build my own firmware for the first time this week.

### Expected behavior

I expected it to compile.

### Actual behavior

It didn't compile.

### Steps to Reproduce

1. Write "BOARD_CREALITY_V4" on the "MOTHERBOARD" line
2. Setup your setting
3. Try to compile

### Version of Marlin Firmware

2.0.9.3

### Printer model

Ender 3 V2

### Electronics

Stock 4.2.2 board

### Add-ons

CRTouch

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Cura

### Host Software

OctoPrint

thisiskeithb, 2022-01-01:
This file needs a good scrub & updating since it has options for old environments as an example:

https://github.com/MarlinFirmware/Marlin/blob/4de84031d84c9b92a7ce5c8df48b15682e51acdb/buildroot/share/vscode/auto_build.py#L607-L614

sjasonsmith, 2022-01-02:
@thisiskeithb are you working on this? I was poking around and made some changes so it will work with an arbitrary number of environments. I don't want to clobber anything you have in progress, though.

thisiskeithb, 2022-01-02:
> are you working on this? I was poking around and made some changes so it will work with an arbitrary number of environments. I don't want to clobber anything you have in progress, though.

I'm not working on it, so feel free to submit a fix. 🙂

sjasonsmith, 2022-01-02:
@MarcAntoineLamy thanks for the hint, that helped me find the right place to start.

I've made quite a few additional changes to the code to resolve other issues I found. You can find it at Pull Request #23427.
Feel free to try it out or provide any feedback you may have.

thinkyhead, 2022-01-02:
NOTE: The script `mftest` with parameter `-a` is the currently supported method to build from the command-line.
```
$ mftest -a

Auto Build...
Available targets for "RAMPS 1.4 (Power outputs: Hotend, Fan, Bed)" | RAMPS_14_EFB (1020):
 1) mega2560
 2) mega1280
Select a target for 'RAMPS_14_EFB' (1-2) : 1

Selected mega2560
Building environment mega2560 for board RAMPS_14_EFB (1020)...
```

closed 2022-01-02 by thinkyhead