MarlinFirmware/Marlin #23418: [BUG] Marlin 2.0.9.3 cooling fan

Cysign, 2022-01-01:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Always when turning on the machine, part cooling fan runs at 100%. Entering temperature/fan speed it displays 0.

When I change fan speed manually and then back to 0% fan turns off. But as soon as I start a print (cura: no fan on first layers!), fan turns on, suddenly.
Marlin 2.0.9.3.

On Bugfix release it's even worse, I cannot even turn off partcooling fan manually.

I'm printing from SD card, USB disconnected.

### Bug Timeline

new

### Expected behavior

Partcooling fan should be off while machine boots.
Partcooling fan should be on/off like configured in Cura (i.e. off on the first layer).

### Actual behavior

see above

### Steps to Reproduce

Turn on machine (Anet A9 Plus, Atmega1284P).

### Version of Marlin Firmware

2.0.9.3

### Printer model

Anet A8 Plus

### Electronics

stock

### Add-ons

-

### Bed Leveling

MBL Manual Bed Leveling

### Your Slicer

Cura

roel8032, 2022-01-02:
The same problem here, with Anet A6 [2.0.9.3.]

Cysign, 2022-01-02:
Make the size fitting to your board depends on the used microcontroller. As it's a cheap AtMega on the Anet boards there's not much memory/flash available. That's not a problem of Marlin, which is compatible with many different boards/controllers.

But good to know the same behavior occurs on your board - so I's not my fault ;)

kaoalex, 2022-01-03:
Same problem here! My board is the Anet A8 V1.5.
The fan is always on.

thinkyhead, 2022-01-03:
Does `PINS_DEBUGGING` and the `M43` command show any difference comparing Marlin 2.0.9.2 (working) to 2.0.9.3 (non-working) ?

roel8032, 2022-01-03:
> Does `PINS_DEBUGGING` and the `M43` command show any difference comparing Marlin 2.0.9.2 (working) to 2.0.9.3 (non-working) ?

2.0.9.1 was OK. I Skipped 2.0.9.2, so i don't know. Speed control of the fan with host software (Repetier Host) is working normally. You can run the fan at different speeds.

thinkyhead, 2022-01-05:
Would you be able to test some `bugfix-2.0.x` commits to narrow this down?

The strategy is to find a commit from some point in the "distant" past where the feature works. Then, test a commit from halfway between that date and today…. And then you keep going to the commit half-way in between your "known to work" commit and your "known to be broken" commit until you find the exact day where it broke.

If you started from a point 256 commits in the past, it would take no more than 8 tests to find the exact commit that broke it.

roel8032, 2022-01-07:
> Would you be able to test some `bugfix-2.0.x` commits to narrow this down?
>
> The strategy is to find a commit from some point in the "distant" past where the feature works. Then, test a commit from [...]

I don't have that versions.

Ajaif, 2022-01-10:
Anet A6 v1.0 (1.5) [2.0.9.3.]. I confirm the bug.

With the boot the fan starts at full speed. The first use of M107 or M106 S0 doesn't have effect. After setting the speed to any non-zero value e.g. M106 S128 the behavior normalizes - booth M107 and M106 S0 can stop the fan as expected. It seems   the speed variable at startup is initialized with 0 but not the PWM.
After connecting or disconnecting of OctoPrint the fan starts again.

descipher, 2022-01-13:
Can’t be certain but, a very recent merge for AVR PWM to bugfix-2.0.x may resolve these issues.

marlinfwgeek, 2022-01-13:
Tested this morning with the latest available bugfix build and can confirm that it does not fix the issue - fan still runs at 100% but shows as 0%. If the fan is adjusted to 100% and then back to 0%, then fan the turns off.

descipher, 2022-01-13:
> Tested this morning with the latest available bugfix build and can confirm that it does not fix the issue - fan still runs at 100% but shows as 0%. If the fan is adjusted to 100% and then back to 0%, then fan the turns off.

Could you upload your config files.
There isn't much to go by in this report.

Cysign, 2022-01-13:
I'll arrive back home to. orrow, then I can upload my config.
But I'm using the Anet A8 Plus stock config with not many changes (just adjusted/calibrated the axis). So using this stock config would be a good starting point to hunt the bug ;)

marlinfwgeek, 2022-01-13:
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/7865426/Configuration.zip)

Config uploaded - I really don't change much from the Anet A8 examples that would affect this behavior.

marlinfwgeek, 2022-01-13:
Please also see #23459, which has a lot of great information on this problem, but was closed and this issue referenced instead.

descipher, 2022-01-13:
I think I see the issue, on the MELZI board is using pin 4 as the fan, in this case the timer is TCCR0 which is a system protected timer. The PWM code tries to protect timer 0 and timer 1 but it does allow a digitalwrite to that pin. So with the MELZI it's a Mosfet connect to pin 4 and will be active low and thus on immediately on boot.

When the pin is checked with get_pwm_timer it is filtered out and does not get a duty setting via analogWrite or set_pwm_duty.

```break;    // Protect reserved timers (TIMER0 & TIMER1)   ```

We need an exception for some pins on timer 0, here we need to allow a pwm duty config.
I will confer with @thinkyhead on the next steps.

thinkyhead, 2022-01-13:
> I don't have that versions.

As one might expect, Marlin from any point in time can be checked out by commit ID or downloaded from the "Commits" page, linked from the Main page here on GitHub.

roel8032, 2022-01-14:
> > I don't have that versions.
> 
> As one might expect, Marlin from any point in time can be checked out by commit ID or downloaded from the "Commits" page, linked from the Main Marlin page here on GitHub, first selecting the branch whose history you want to browse.

Do you have a link? I can not find that.

descipher, 2022-01-14:
Can you  enable soft fan pwm in configuration.h 
#define FAN_SOFT_PWM
It will give me more info it that changes it, could be your solution as well. I can silence the 1KHz fan noise.

Ajaif, 2022-01-14:
In line 1313 of ./src/module/planner.cpp

`    bool fans_need_update = true;`

works for me.

descipher, 2022-01-14:
Interesting, I saw the init added to that array, but the planner should not be setting the initial state to high or low. It is unaware of the callers requirement. It could be active high or active low depending on the output device.

descipher, 2022-01-14:
It only works if the fan output connection is active low. That not always the case, but is a more common configuration for a MOSFET driven output.

descipher, 2022-01-15:
Update: 
I checked the ANet schematics and they are active HIGH on the outputs so this issue has nothing to do with the 255 initialization in the planner code. 
Setting bool fans_need_update = true; is a problem, setting this flag will break other elements, such as Laser fan control, controller fan control .. etc. 
The correct solution is to properly initialize the fan speed on startup instead of waiting for a movement  block to be created in the planner.
I am working on that but I will not have a test device till Monday.

descipher, 2022-01-15:
PR updated to properly init the fans_need_update once on startup.

thisiskeithb, 2022-01-15:
> PR updated to properly init the fans_need_update once on startup.

This one? https://github.com/MarlinFirmware/Marlin/pull/23520/commits

Probably need to re-push since last commit was yesterday from thinkyhead.

Andreaux, 2022-01-23:
Sorry to be pushy, but has this fix been committed in? I just installed the latest bugfix version (2022-01-23) and the fan still kicks on as soon as the printer regains consciousness....

Andreaux, 2022-01-23:
I've checked a few of the files in /src (/src/HAL/AVR/fast_pwm.cpp for example) and I see the commits by descipher in there so I assume yes. Still, I'm having the issue.

The printer is an Anet A8, stock board flashed with latest Marlin-bugfix-2.0.x.

jimkazmer, 2022-01-23:
I (too) am having this same issue. I found it in the 2.0.x release:
SHORT_BUILD_VERSION "2.0.9.3"
STRING_DISTRIBUTION_DATE "2021-12-25"

default_envs = chitu_f103 
I am converting a (stock) TronXY X5SA-400-PRO to use Marlin. 
The (chitu) board shows: Version CXY-V6-191017

I tried the recent fix (which didn't appear to fix it, but I did not try all the permutations I could have (see questions below): 
SHORT_BUILD_VERSION "bugfix-2.0.x"
STRING_DISTRIBUTION_DATE "2022-01-22"

Problem: Part Cooling Fan starts spinning at max speed after powering on the printer; the UI indicates 0% speed, and changing the setting in the UI does nothing. 

Attached are my "Configuration.h" and "Configuration_adv.h" files.  I have tried the four variations for the settings of:
//#define COOLER_AUTO_FAN_PIN -1
#define COOLER_AUTO_FAN_PIN FAN_PIN
//#define COOLER_FAN_PIN -1
#define COOLER_FAN_PIN FAN_PIN

"FAN_PIN" is defined as "PG13" from "pins_CHITU3D_common.h".

TronXY made it difficult to get Marlin onto the board... I had to first overwrite Tronxt's proprietary firmware version with an older version using a STM32 programmer.  

The extruder fan and the controller fan both behave as configured. I have the controller box fan configured to turn on whenever the stepper motors are active (this works great). The extruder fan turns on at the configured temperature, and turns off when the extruder falls below that temperature (this works great).  I used Marlin many years ago, and really like the improvements in 2.0.

[Configuration-Files_TronXY_X5sa-400-pro.zip](https://github.com/MarlinFirmware/Marlin/files/7921573/Configuration-Files_TronXY_X5sa-400-pro.zip)

Andreaux, 2022-01-23:
These fixes are in the bugfix branch only for now... but I'm having issues with that too.

Andreaux, 2022-01-23:
Are you able to change the speed of the fan and then set 0 as speed to stop it? I can, but it surely is a pain regardless to have to do that after each connect/disconnect, restart, fresh power-on...

jimkazmer, 2022-01-23:
I stared at my PDF file and have come to the conclusion that it is for a different TronXY board than the BOARD_INFO_NAME "Chitu3D V6".
 I'm off to search for my board's correct diagram.

descipher, 2022-01-23:
@jimkazmer Not only is it different, you don't want to setup a laser cooler.
//#define COOLER_AUTO_FAN_PIN -1
//#define COOLER_FAN_PIN -1

jimkazmer, 2022-01-23:
> 

I assumed "COOLER" meant "Part cooler" not "laser cooler".  What defines are for configuring the part cooler?  I thought I was reading through this thoroughly...

descipher, 2022-01-23:
@Andreaux Works on the scope and it was verified using the ATMega2560 ramps pin D4 and pin D9.

descipher, 2022-01-23:
This is not a support area. I would suggest you try Discord. https://discord.gg/

jimkazmer, 2022-01-24:
Okay, thanks... I was just trying to provide information... I will ask my novice questions elsewhere. 

The part cooling fan is stuck on, and nothing I have tried can stop it. I reverted to my prior firmware (painful process with the STM32 programmer) and the fan behaves properly.  Put Marlin back on, and the fan is stuck on. For someone new to building Marlin, this is a time consuming process (learning a new environment, new tools, etc).

descipher, 2022-01-24:
@jimkazmer No worries, thanks for the info.

jimkazmer, 2022-01-24:
@descipher Is there a specific fix I can try, or is the current head of bugfix-2.0.x what I should use?

descipher, 2022-01-24:
You have a different board so this patch will not work for you. This patch is for AVR based controllers. I have noted you're having a similar the issue on an STM32 based board. Don't have confirmation that is broken for the board you have or the STM32 platform.
Will need to compile it on an STM32 and see.

descipher, 2022-01-24:
You could see if 2.0.9.1 works for you https://github.com/MarlinFirmware/Marlin/archive/refs/tags/2.0.9.1.zip

Andreaux, 2022-01-24:
> You could see if 2.0.9.1 works for you https://github.com/MarlinFirmware/Marlin/archive/refs/tags/2.0.9.1.zip

Was that meant for me and the OP?

descipher, 2022-01-24:
> > You could see if 2.0.9.1 works for you https://github.com/MarlinFirmware/Marlin/archive/refs/tags/2.0.9.1.zip
> 
> Was that meant for me and the OP?

For @jimkazmer or you

jimkazmer, 2022-01-24:
@descipher  
I tried 2.0.9.1  
Same behavior on the part cooling fan (began spinning at power on, and could not stop it via the UI).
The extruder fan turned on when temp>50 and back off when temp<50.
The controller fan turned on when I used the stepper motors, and turned off 60 seconds after I disabled the stepper motors.

After a bit of wallowing, I decided to re-build 2.0.9.1 but with this change to my "pins_CHITU3D_V6.h" file:
//#define FAN_PIN                             PG13  // original value
//#define FAN2_PIN                            PG14 //original value
//trying swapping two pins to see what happens
#define FAN_PIN                             PG14
#define FAN2_PIN                            PG13

The problem followed the pin...
The extruder fan began spinning at power on, and could not stop it via the UI.
The part cooling fan turned on when extruder temp>50 and back off when extruder temp<50.

I think this tells us that the hardware is fine (as in, both the PG14 and PG13 pins can function as expected).

I would guess that 2.0.x would behave this same way.

I hate wasting other people's time as much as wasting my own.  My fear is I did some stupid edit in the config files that is causing this... I have scanned so many example (and other Tronxy Marlin conversions), but I just can't see what could be wrong.

descipher, 2022-01-24:
That's valuable info thanks for checking it out. No worries no one's time is waisted , we signed up for it. We just try to keep the information that is related to an issue where it should be. 
That fact that it follows FAN_PIN is important. It looks like it's related to the OPs observe.

descipher, 2022-01-24:
This is always helpful to use https://marlinfw.org/docs/configuration/configuration.html
You will find Cooling FANs in the configuration_adv.h section.

jimkazmer, 2022-01-24:
Thank you.  I will start reading that.

jimkazmer, 2022-01-24:
@descipher 
I found a "work-around" or an "undocumented STM32 requirement" or my mistake...
In "Configuration.h", I uncommented "NUM_M106_FANS " and "FAN_SOFT_PWM".  I am not sure if the first or second or both are required. I left "SOFT_PWM_SCALE" at 0.  

Once I set those, the fan did not start on power-on, and I was able to control it via the UI. I only tested this on 2.0.9.1; tomorrow, I will try it on 2.0.9.3

Uhg... I just checked my reference configurations (from other Tronxy->Marlin migrations), and all had "FAN_SOFT_PWM" enabled... so that is likely the required piece I was missing.  By default, both of those are undefined when you grab the latest code.

// Set number of user-controlled fans. Disable to use all board-defined fans.
// :[1,2,3,4,5,6,7,8]
#define NUM_M106_FANS 1
// Use software PWM to drive the fan, as for the heaters. This uses a very low frequency
// which is not as annoying as with the hardware PWM. On the other hand, if this frequency
// is too low, you should also increment SOFT_PWM_SCALE.
#define FAN_SOFT_PWM

descipher, 2022-01-24:
@jimkazmer Based on the datasheet for the STM32F103x PG13 and PG14 are not PWM pins so you do need:
#define FAN_SOFT_PWM
You probably don't need this defined:
#define NUM_M106_FANS 1

descipher, 2022-01-24:
For all, I just tested STM32F103x:
I don't see an issue with FAST_FAN_PWM on my STM32 and 2.0.9.3.

jimkazmer, 2022-01-24:
I had read the "Configurations.h" file and it was not obvious to me that I needed to define FAN_SOFT_PWM.  In this case, some additional comment may have helped me avoid this.  Thank you for your help.
Jim

Andreaux, 2022-01-24:
I'll be a good test subject as I have not tinkered with the fan settings, just used the example Anet A8 config that came with Marlin 2.0.x and flashed the latest bugfix release. Enabled the BLTouch, enabled Linear Advance, set the probe offsets and Z offset... and that's about it. No change to fans whatsoever in the config, it's vanilla.

marlinfwgeek, 2022-01-24:
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/7929193/Marlin.zip)
Tested the latest bugfix and can confirm that the issue still exists. I'm running the stock Anet A8 board. I've attached my config files.

Andreaux, 2022-01-26:
Sorry for not getting back sooner with info on 2.0.9.1, but I wasn't able to compile it. It had all sorts of issues with my configuration being too recent. Even forcing the version number to read xxx2091, then the declarations had issues here and there. I didn't have time in the immediate to see what is going on. If nobody else comes up with results, I'll try later this week with a vanilla config just to see if the fan kicks in or not...

descipher, 2022-01-26:
I will look at it again, the issue is I can't test it on the 1284 chip. The 2560 and the same PWM registers and pin maps. 
Question: on 2.0.9.3 what is the behavior in the menu at 25 % and 75 % I need to know if its inverted in speed. e.g. 75% is slower than 25%. 100% is off.

ellensp, 2022-01-26:
I just tried this on a 1284p (a BOARD_SANGUINOLOLU_12)
But I used the provided Config files with BOARD_ANET_10, as they are close enough
Updated configs to latest Bugfix as there are a few incompatibilities, but this made no difference.

I confirmed the issue on pin 4.  

I also noticed in M43
```
PIN:   4   Port: B4 (A27)  AVR_SS_PIN                             protected 
.                          FAN_PIN                                protected 
PIN:   5   Port: B5 (A26)  AVR_MOSI_PIN                           Output = 1  
```

When I take steps to eliminate this conflict 
```
PIN:   4   Port: B4 (A27)  FAN_PIN                                protected 
PIN:   5   Port: B5 (A26)  AVR_MOSI_PIN                           Output = 1  
```
Pin 4 behaves as expected.

So can someone else confirm this fixes it.

To replicate this open Marlin/src/HAL/AVR/spi_pins.h
Find and replace     #define AVR_SS_PIN   4 with #define AVR_SS_PIN   -1
 
This is not a final fix, just a test, needs a cleaner solution

ellensp, 2022-01-26:
@descipher definitely  not inverted. M106 S25 is duller (im using a led, not a real fan) than M106 S75

descipher, 2022-01-26:
Perfect, you rock!
We actually had two issues then. Pin conflict was the second issue.

descipher, 2022-01-26:
This would be safer:
149 pins_ANET_10.h 
+#undef AVR_SS_PIN                                 // AVR HAL SPI pin for some boards, conflicts with FAN_PIN 4

ellensp, 2022-01-26:
@descipher I tried that. It does not work.

descipher, 2022-01-26:
Darn, it will need to be set to -1 b4 the HAL get to it.

descipher, 2022-01-26:
I can emulate that pin conflict state on the 2560 and work through some options.

[45 more replies]

closed 2022-01-30 by ellensp