MarlinFirmware/Marlin #23419: [BUG] Z Axis moving upwards

ndk-ender3, 2022-01-01:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

after installing the newest version of bugfix, my Z axis (when probing and homing) moves upwrds.

If I enable reversing Z axis in the code, I cant complie.

regards

### Bug Timeline

new

### Expected behavior

Z axis moving downwards

### Actual behavior

Z axis moving upwards

### Version of Marlin Firmware

bugfix 2.0.x

### Printer model

ender 3 

### Electronics

4.2.2

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Cura

### Host Software

OctoPrint

ellensp, 2022-01-01:
no Configs, no error messages, needs more information

LuxInTenebr1s, 2022-01-09:
I have the same issue. I can provide any info you need per request.
- SKR 2.0 \w F429
- TMC2209 sensorless homing and probing enabled
- homing works just fine
- probing moves my carriage upwards instead of downwards to the actual probing 
- 2.0.9.3 Marlin

Sensorless probing does not have any direction settings. I've also tried to hack move command within G34 source code to invert moving direction but that didn't work.

sajmonSK, 2022-01-14:
I have the same problem (as a nice gouge in my PEI sheet can tell). 
SKR 1.4
TMC2209 UART
Piezo sensor (nozzle as probe)
Marlin 2.0.9.3 stable

Z homing works as it should, but during G34 the Z axis moves to the measurement position, moves upward first, then moves downward and actually probes.
It does the same on the other side and the value keeps rising by about 2mm each time, and the X axis gantry ends up skewed. It looks like it is measuring from the position after the extra upward move

BigTGaming, 2022-01-22:
i am having the same issue with my V2 v4.2.2 and the probe keeps triggering without touching the bed

Seelenkind, 2022-02-02:
either ... or ,

if black and white cable is plugged into Z endstop
#define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN
//#define USE_PROBE_FOR_Z_HOMING

if the black and white cable is connected to the probe
//#define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN
#define USE_PROBE_FOR_Z_HOMING

ndk-ender3, 2022-04-03:
still ongoing

thisiskeithb, 2022-04-03:
> still ongoing

Even with today’s bugfix? Zip up and attach your configs.

ellensp, 2022-04-03:
And tell us where you probe is attached to z_min or probe port

closed 2022-06-13 by github-actions[bot]